bigtreetech/CB2 #1: User Manual is for Pi2

MackinsVII, 2024-05-01:
We need the manual for CB2.

johnnycoyne, 2024-05-02:
> We need the manual for CB2.

flash emmmc?

Travis90x, 2024-05-06:
> We need the manual for CB2.

https://github.com/bigtreetech/CB2/blob/master/BIGTREETECH%20Pi%202%20User%20Manual.pdf

CorriasFrancesco, 2024-05-11:
> > We need the manual for CB2.
> 
> https://github.com/bigtreetech/CB2/blob/master/BIGTREETECH%20Pi%202%20User%20Manual.pdf

that manual is for PI2 which is different from CB2

Travis90x, 2024-05-11:
> > > We need the manual for CB2.
> > 
> > 
> > https://github.com/bigtreetech/CB2/blob/master/BIGTREETECH%20Pi%202%20User%20Manual.pdf
> 
> that manual is for PI2 which is different from CB2

It has the same Soc, gpio scheme and O.S., same procudure to flash etc...
what do you need?

Tina-Builds, 2024-05-14:
> > > > We need the manual for CB2.
> > > 
> > > 
> > > https://github.com/bigtreetech/CB2/blob/master/BIGTREETECH%20Pi%202%20User%20Manual.pdf
> > 
> > 
> > that manual is for PI2 which is different from CB2
> 
> It has the same Soc, gpio scheme and O.S., same procedure to flash etc... what do you need? https://travis90x.altervista.org/bigtreetech-btt-pi-v2-for-klipper/

For example the CB2 has a PCIe 2.1 listed but on the top of the CB2 module is no PCIe connector. So how would one connect anything to the PCIe of the CB2?

Travis90x, 2024-05-14:
You need the motherboard manual as BTT PB4 (CM4 adapter board), if you need pcie take a look to other boards.
[image]
[image]
[image]

themadpants, 2024-05-14:
> It has the same Soc, gpio scheme and O.S., same procedure to flash etc... what do you need? https://travis90x.altervista.org/bigtreetech-btt-pi-v2-for-klipper/

I am trying to install a CB2 on a BTT e3ez board, but the instructions listed for the e3ez (For a CB1) or the PI2 dont work. It wont work with a SD Card or via USB. So no, it's not the same procedure clearly.

Travis90x, 2024-05-14:
When you install an Intel CPU, an SSD, wifi adapter etc on an ASUS motherboard, do you refer to the manual from Intel or ASUS?

themadpants, 2024-05-14:
> When you install an Intel CPU, an SSD, wifi adapter etc on an ASUS motherboard, do you refer to the manual from Intel or ASUS?

Are you naturally this obtuse, or does it take work?

Travis90x, 2024-05-14:
You are right, in my country we say: "washing a donkey is a waste of water and soap."
So next time ask directly to BTT staff how they are "obtuse" to not release the guide, bye.

themadpants, 2024-05-14:
> You are right, in my country we say: "to wash a donkey’s tail is a waste of water and soap." So next time ask directly to BTT staff how they are "obtuse" to not release the guide, bye.

Why do you think people are posting here? Good grief.

Travis90x, 2024-05-14:
Not "everyone", just a fake profile of a small man who doesn't have the courage to show himself.
My CB2 works well with two boards with ssd nvme on pcie (not BTT) so... cheers!

themadpants, 2024-05-14:
> Not "everyone", just a fake profile of a small man who doesn't have the courage to show himself. My CB2 works well with two boards with ssd nvme on pcie (not BTT) so... cheers!

unsubscribe. Get a life dude.

Tina-Builds, 2024-05-14:
> Not "everyone", just a fake profile of a small man who doesn't have the courage to show himself. My CB2 works well with two boards with ssd nvme on pcie (not BTT) so... cheers!

A little bit I can understand him, but his reaction went to far. 
   BTT builds the CB2 mainly for the use on their mainboards or BTT PI4B adapter, but with those combinations you cannot use the PCIe port of the CB2 which is clearly advertised on the product page.
   I know you posted 3rd party adapter boards which make it possible to use the CB2 in its full functionality, but BTT is a manufacture who focuses on 3D printing. So it is not to understand why they would advertise the PCIe feature and have no HW application ready for it.

CorriasFrancesco, 2024-05-14:
And i cannot get it work well even with a manta M8P from BTT which worked perfectly with a CB1, so a specific manual might help to understand if there are particular steps to be taken on the CB2, even just a statement like if something works the same as the PI2 and where are the differences

themadpants, 2024-05-14:
Yes, all of us attempting to use it on a Manta board are just asking for some documentation or support from BTT to get it to work. The instructions are Pi specific and refer to switches or jumpers that are not apparent on the printer control boards. If anyone has successfully installed and configured one on a e3ez or M8P, please share the knowledge. It would really be appreciated.

I agree @Tina-Builds , I might be better off returning this and spend the extra on a CM4. I took a chance on this board, as it advertised CSI etc which would be useful.

CorriasFrancesco, 2024-05-14:
> > And i cannot get it work well even with a manta M8P from BTT which worked perfectly with a CB1, so a specific manual might help to understand if there are particular steps to be taken on the CB2, even just a statement like if something works the same as the PI2 and where are the differences
> 
> Agreed, a manual would be helpful. But it is sadly practice of the Raspberry "clones" that a working OS takes time. IMHO you are better of to buy a CM4 with 2GB/16GB and WiFi rather than a CB2

Yeah i know it is not simple at all to build a working OS but i had to replace a CB1 and the CB2 seemed a natural choice. I am currently trying to solve my problem with the assistance but i appreciate your suggestion, and probably follow it if i cannot solve it in a reasonable time

Tina-Builds, 2024-05-14:
> Yes, all of us attempting to use it on a Manta board are just asking for some documentation or support from BTT to get it to work. The instructions are Pi specific and refer to switches or jumpers [...]

I have a Manta E3EZ and it is at the moment with a CB-1 and that does it for me. The CB-1 is something I can support to buy, if you don't need a CSI/DSI interface, but to buy the CB2 is far to early IMHO as there is not even a Manual for it yet.

CorriasFrancesco, 2024-05-14:
> I have a Manta E3EZ and it is at the moment with a CB-1 and that does it for me. The CB-1 is something I can support to buy, if you don't need a CSI/DSI interface, but to buy the CB2 is far to early [...]

Yes the CB1 worked really great but unfortunately it died and had to find a replacement. For now i completely agree with you sticking to the CB1 is better for the moment.

Tina-Builds, 2024-05-14:
> Yes the CB1 worked really great but unfortunately it died and had to find a replacement. For now i completely agree with you sticking to the CB1 is better for the moment.

Or a CM4 ;)

themadpants, 2024-05-14:
> I have a Manta E3EZ and it is at the moment with a CB-1 and that does it for me. The CB-1 is something I can support to buy, if you don't need a CSI/DSI interface, but to buy the CB2 is far to early [...]

Yes. I already have a E3EZ with a CB1 and it works fine, but no support for DSI or CSI. I am upgrading an identical printer, and figured I would give this a shot and then upgrade the other if it worked for a pi cam and pi screen. I will troubleshoot a little more to see if I can figure it out, but it might be a case of splurging on a CM4.

Tina-Builds, 2024-05-14:
> Yes. I already have a E3EZ with a CB1 and it works fine, but no support for DSI or CSI. I am upgrading an identical printer, and figured I would give this a shot and then upgrade the other if it [...]

I have only the CB-1 and no CB2 at hand. I remember when I had to get the OS image to the eMMC of the CB-1 I was lost at first, but as far as I recall I solved it by following step bay step the mainboard manual, E3EZ in my case. It was something very simple I overlooked at first, with setting the jumpers.

Travis90x, 2024-05-14:
> BTT builds the CB2 mainly for the use on their mainboards or BTT PI4B adapter,

So you are claiming that BTT stated the CB2 should be used exclusively with their adapter boards and >NOT< with other "CM4 socket" boards? Where?

> but BTT is a manufacture who focuses on 3D printing. So it is not to understand why they would advertise the PCIe feature and have no HW application ready for it.

And what if in the future they release an RB4 v2 adapter with PCIe? If the CB2 supports a PCIe signal for the CM4 socket, why not list it among the features?

You are wrong in thinking that the CB2 should be considered only as a device for 3D printers. It competes with all other devices too like CM4, OrangePi, BananaPi etc. 

And just so you know, I use Android TV boxes with 3D printers (Armbian OS), from those with SoCs like Allwinner H6 to Amlogic S905X3, Rockchip RK3229 RK3318 etc, and a USB RP2040 when I need gpio.

https://travis90x.altervista.org/

Tina-Builds, 2024-05-14:
> So you are claiming that BTT stated the CB2 should be used exclusively with their adapter boards and >NOT< with other "CM4 socket" boards? Where?
>
>
> And what if in the future they release an RB4 v2 [...]

To quote an other person here "get a live"

themadpants, 2024-05-15:
> You are wrong in thinking that the CB2 should be considered only as a device for 3D printers. It competes with all other devices too like CM4, OrangePi, BananaPi etc.

The funniest thing is after successfully installing the official BTT image from this repository on the board tonight, it has Klipper preinstalled. Oops. Lmao. Go away.

themadpants, 2024-05-15:
> Yes the CB1 worked really great but unfortunately it died and had to find a replacement. For now i completely agree with you sticking to the CB1 is better for the moment.

I was finally able to get this imaged on the same setup you are attempting. I used Balena Etcher to install the image on an SD card, and before removing from my laptop, I added my wifi info to the config file. Inserted it in to the SD card slot below the spi connector and powered it up with a 24v power supply. It succesfully booted running off the SD card, and I was able to SSH in and perform the install to eMMC described on page 19 of the Pi2 user manual. Once it completed, I powered off the board and removed the SD card. Powered back up and it is running. I had no jumpers installed for this process. This did not work last night powered just off USB, fyi.

CorriasFrancesco, 2024-05-15:
> The other method described using the RKDevTool would not work at all with the Manta. I just got failures on everything.

I managed to flash it too on the eMMC, but i was able to do it using the raspi imager to write the image on the SD (i already had it in my PC) and then flash like you said. Now i am missing only the communication with the board itself which is not seen from the CB2

MackinsVII, 2024-05-15:
[BIGTREETECH CB2 User Manual.pdf](https://github.com/bigtreetech/CB2/files/15320794/BIGTREETECH.CB2.User.Manual.pdf)

BTT have sent me the manual.

themadpants, 2024-05-15:
> > The other method described using the RKDevTool would not work at all with the Manta. I just got failures on everything.
> 
> I managed to flash it too on the eMMC, but i was able to do it using the raspi imager to write the image on the SD (i already had it in my PC) and then flash like you said. Now i am missing only the communication with the board itself which is not seen from the CB2 edit: also usb ports on the manta not working yet for me

Once I flashed the EMMC, I followed the E3EZ instructions from page 30 onwards to flash the board firmware and link them.

CorriasFrancesco, 2024-05-15:
> Once I flashed the EMMC, I followed the E3EZ instructions from page 30 onwards to flash the board firmware and link them.

I'm happy that you made it. My M8P is not flashing the klipper firmware so i had no luck but i will keep trying following directions from the assitance

themadpants, 2024-05-15:
> I'm happy that you made it. My M8P is not flashing the klipper firmware so i had no luck but i will keep trying following directions from the assitance

Are you attempting via SD card or using the DFU method? If you are using an SD card, make sure the card is formatted to FAT32 and you are inserting the card in to the correct SD card slot (MCU)

CorriasFrancesco, 2024-05-16:
> Are you attempting via SD card or using the DFU method? If you are using an SD card, make sure the card is formatted to FAT32 and you are inserting the card in to the correct SD card slot (MCU)

I tried both methods, the DFU is not being seen by the CB2, with the SD cards the firmware is not changing to .CUR and nothing happens

Tina-Builds, 2024-05-17:
The official manual is now out, which covers all the here raised issues as far as I can tell from reading it (NB I have no CB2).
https://github.com/bigtreetech/CB2/blob/master/BIGTREETECH%20CB2%20User%20Manual.pdf
With that it should be possible to get the CB2 working on any Manta board or on the BTT PI4B adapter.

closed 2024-05-20 by bigtreetech