bigtreetech/CB2 #3: BTT-HDMI7 resolution not correct

Garag, 2024-05-03:
I have attached a BTT-HDMI7 via HDMI to the board. The board runs from SD Card the  current image (CB2_Debian11_Klipper_kernel4.19_20240422). As described in the user manual i have configured HDMI display.

Linux starts and  i can see boot messages on the screen and later KlipperScreen. But the resolution is not correct.

It seems that Linux uses a resolution of 1280x720 but the display has only 1024x600.

How can i set the correct resolution?

Kernel Message:
rockchip-vop2 fe040000.vop: [drm:vop2_crtc_atomic_enable] Update mode to 1280x720p0, type: 11 for vp0

zhangyumingsh, 2024-05-29:
> I have attached a BTT-HDMI7 via HDMI to the board. The board runs from SD Card the current image (CB2_Debian11_Klipper_kernel4.19_20240422). As described in the user manual i have configured HDMI [...]

Try flashing the image again

Garag, 2024-06-05:
The problem is with the V1.0.4 image solved. The display just works.
Thank you.

closed 2024-06-05 by Garag