bigtreetech/CB2 #5: 汉字乱码，触控失效，联网不稳定，klipper重启后加载不了。

cavalligao, 2024-05-04:
汉字乱码，触控失效，联网不稳定，klipper重启后加载不了。

Hentrich7274, 2024-05-20:
Kann ich bestätigen "Internetverbindung ist instabil und Klipper kann nach dem Neustart nicht geladen werden."

BTTJUSTIN, 2024-05-23:
Thanks for reaching out. Our support team does not operate using the GitHub
platform but we would be happy to help you via email based ticketing system. Please visit
https://biqu.equipment/pages/technical-support and submit a ticket and we will be in touch with you soon.

closed 2024-05-23 by BTTJUSTIN

looxonline, 2024-05-23:
> 汉字乱码，触控失效，联网不稳定，klipper重启后加载不了。

This appears to be more of an issue with your UI and klipper installation than it is an issue with the hardware or the OS image. If you are able to provide more details we will be able to establish whether or not it is an OS image or hardware issue but for the moment it seems to be a standard support request.