bigtreetech/CB2 #6: No usbserial or CH341 drivers in  OS image V1.0.3

RK392, 2024-05-07:
The usbserial and CH341 drivers are not included with V1.0.3 and I cannot install them to Armbian on V1.0.3 due to a lack of source code. These drivers were included for [CB1](https://github.com/bigtreetech/CB1-Kernel/tree/kernel-5.16/kernel/drivers/usb/serial) and BTT Pi V1.2 but not for the CB2. Since there is still no repo for CB2-Kernel, I am unable to compile and install any drivers successfully.

I bought the BTT Pi 2 to connect to a Creality 4.2.2 board running Klipper. The board uses a GD32F303 which identifies as "QinHeng Electronics CH340 serial converter" using the "lsusb" command and it works with Klipper running in Ubuntu. Unfortunately, due to the drivers not being included with the CB2 V1.0.3 OS image, the BTT Pi 2 cannot communicate with it.

AlvinF01, 2024-05-08:
Try connect usbserial with speed "1500000".
I asked their customer service they said it need to use this baud rate.
I can success  connect it with putty using FT232 TTL USB adapter.
Then you should able to update the klipper and flash your device

AlvinF01, 2024-05-08:
The baud rate is use only connect to PC by usbserial. If you can access to PI2 you should able to build the driver directly
Try follow this guide?
https://learn.sparkfun.com/tutorials/how-to-install-ch340-drivers/linux
Well, I am not sure about the driver, sometimes the klipper firmware version different or misconfig can also caused the mcu not being recognized

RK392, 2024-05-08:
Yes, I already tried this guide. It didn't work since there are no linux kernel headers so I am unable to compile the driver.

> Since there is still no repo for CB2-Kernel, I am unable to compile and install any drivers successfully.

Here is the output when trying to make the CH341 driver:
biqu@bigtreetech-cb2:~/CH341SER_LINUX$ make
make -C /lib/modules/4.19.232-bigtree-cb2/build  M=/home/biqu/CH341SER_LINUX
make[1]: *** /lib/modules/4.19.232-bigtree-cb2/build: No such file or directory.  Stop.
make: *** [Makefile:5: default] Error 2

The build folder doesn't exist at /lib/modules/4.19.232-bigtree-cb2/ which is where linux kernel headers are usually installed. I attempted to compile anyway by installing armbian linux headers for another kernel with a similar version of linux but that didn't work either. 

I assume this issue can be fixed when BTT release the CB2-Kernel repo like they did with the CB1-Kernel.

TheFeralEngineer, 2024-05-08:
I'm having this issue as well. lsusb works fine and shows the ch340 devices connected but it's not creating the /dev/serial paths

RK392, 2024-05-08:
I also tried using a UART connection from the TX and RX pins. I hijacked the rx/tx lines to the CH340 as shown [here](https://www.reddit.com/r/klippers/comments/q5cu20/comment/j02fi3u/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button). Then I edited /boot/armbianEnv.txt by changing console to serial instead of display (console=serial) and enabled overlays=uart0. I tried changing the mcu in printer.cfg to /dev/ttyS0 and S1 and S2. Unfortunately, none of this worked.

I'm not sure which UART overlay to enable and which pins to connect to. The documentation included in the manual for this board needs to be more detailed.

TheFeralEngineer, 2024-05-08:
> I also tried using a UART connection from the TX and RX pins. I hijacked the rx/tx lines to the CH340 as shown [...]

tried this also. I unmasked the uart0, which is usually what's used for pins 6 8 and 10 on other devices and it didn't work.

TheFeralEngineer, 2024-05-08:
hey, FWIW, you can use the LCD port UART pins (PB11/PB10) on v4.2.2 and not have to solder anything to the ch340 to use UART

https://klipper.discourse.group/uploads/short-url/24qZWRwsinz16qJQVAsU7FPNi87.pdf

RK392, 2024-05-08:
Thanks, I'm aware. I opted to solder the wires to the ch340 pins because I'm still using the stock display as works with klipper.

Please let me know if you figure out a solution. I've been trying everything I can think of for 3 days. Maybe you can successfully compile/install the driver somehow or find a different OS image that boots on this board or figure out how to make UART work.

TheFeralEngineer, 2024-05-08:
> Thanks, I'm aware. I opted to solder the wires to the ch340 pins because I'm still using the stock display as works with klipper.
>
> Please let me know if you figure out a solution. I've been trying [...]

the real shitty part is that all of my machines use the CH340 driver. The not so shitty part is they sent me a replacement, thinking the board was bad, so now I have two that don't work 😂

RK392, 2024-05-08:
@TheFeralEngineer So I managed to get a UART port working (/dev/ttyS3). 

I looked at the RK3566 datasheet and compared the pin names with those in the BTT Pi 2 manual. I added a loopback to uart0 pins 8 and 10 which didn't work. Then I tried uart3 at pin 13 (UART3_RX/GPIO1_A0) and pin 15 (UART3_TX/GPIO1A1). It worked! So far I was able to get a terminal to work using the command "picocom -b 250000 /dev/ttyS3".

Still having trouble getting Klipper to work through UART. I'll keep testing and report back. Could you please give it a shot?

Make sure that console=serial overlays=uart3 are enabled in /boot/armbianEnv.txt

Right now, I'm trying the following for [mcu] in klipper:
[mcu]
serial: /dev/ttyS3
baud: 250000
restart_method: command

TheFeralEngineer, 2024-05-08:
I'm flashing a new binary right now, so I'll make these changes and see if it works.

TheFeralEngineer, 2024-05-08:
No dice

Phil T.
The Feral Engineer

www.theferalengineer.com

Catnip contributions 🐱🐱
https://paypal.me/timpsonmusicllc

RK392, 2024-05-08:
SUCCESS!
Couldn't get Klipper to work with uart3, but I finally got it to work with uart5.
[image]

/boot/armbianEnv.txt
console=serial
overlays=uart5

printer.cfg
[mcu]
serial: /dev/ttyS5
baud: 250000
restart_method: command

For now, I can at least use my board through UART. I will keep this issue open since the CH341 drivers are still missing which prevents us from using CH34X devices through USB such as an Arduino Nano.

TheFeralEngineer, 2024-05-08:
i'm too lazy to look at the data sheet. what pins are uart5?

image.png (view on web)<https://github.com/bigtreetech/CB2/assets/37779545/a7dbc6e2-103e-4b81-9ff8-eff165eaf08c>

RK392, 2024-05-08:
Oh sorry, I forgot to mention which pins I used. Came back here to let you know. 

Pin 21 (GPIO3_C2)= UART5 TX
Pin 23 (GPIO3_C3)= UART5 RX

Good luck!

TheFeralEngineer, 2024-05-09:
Yup that works but that's not what I'm aiming for either

Phil T.
The Feral Engineer

www.theferalengineer.com

Catnip contributions 🐱🐱
https://paypal.me/timpsonmusicllc

JohnTheCoolingFan, 2024-05-09:
> I assume this issue can be fixed when BTT release the CB2-Kernel repo like they did with the CB1-Kernel.

I think it's already released and used by the armbian build config:
1. CB2 Kernel 6.6.y: https://github.com/bigtreetech/linux/tree/linux-6.6.y-cb2
2. CB2 Kernel 4.19: https://github.com/bigtreetech/linux/tree/linux-4.19-cb2
3. Armbian build CB2 fork: https://github.com/bigtreetech/build/tree/CB2

TheFeralEngineer, 2024-05-09:
I'll check these out. Thanks. BTT sent me a Rockchip utility to try

Phil T.
The Feral Engineer

www.theferalengineer.com

Catnip contributions 🐱🐱
https://paypal.me/timpsonmusicllc

RK392, 2024-05-10:
> I think it's already released and used by the armbian build config:
>
> 1. CB2 Kernel 6.6.y: https://github.com/bigtreetech/linux/tree/linux-6.6.y-cb2
> 2. CB2 Kernel 4.19: [...]

Thanks. But, I'm having a hard time trying to compile an image under Ubuntu 24.04. Made a clean install of Ubuntu, had to install docker. I tried the CB2 and CB2-2024, and CB2-legacy branches. Usually, I aim for the legacy kernel with Bookworm to match the V1.0.3 OS image. None of them are compiling successfully. Any advice would be appreciated.

MackinsVII, 2024-05-10:
I'm also having problems with serials. I have a STM32 toolhead board connected via RS232 and it sees it, but won't give it a serial.

Is this problem related?

TheFeralEngineer, 2024-05-10:
UART5 seems to be the only one that works so far. Pins 21 (tx) and 23 (Rx)

Phil T.
The Feral Engineer

www.theferalengineer.com

Catnip contributions 🐱🐱
https://paypal.me/timpsonmusicllc

RK392, 2024-05-10:
> > > I assume this issue can be fixed when BTT release the CB2-Kernel repo like they did with the CB1-Kernel.
> > 
> > 
> > I think it's already released and used by the armbian build config:
> > 
> > 1. CB2 Kernel 6.6.y: https://github.com/bigtreetech/linux/tree/linux-6.6.y-cb2
> > 2. CB2 Kernel 4.19: https://github.com/bigtreetech/linux/tree/linux-4.19-cb2
> > 3. Armbian build CB2 fork: https://github.com/bigtreetech/build/tree/CB2
> 
> Thanks. But, I'm having a hard time trying to compile an image under Ubuntu 24.04. Made a clean install of Ubuntu, had to install docker. I tried the CB2 and CB2-2024, and CB2-legacy branches. Usually, I aim for the legacy kernel with Bullseye to match the V1.0.3 OS image. None of them are compiling successfully. Any advice would be appreciated.

So far, I've managed to compile BTT-CB2_1.0.3_Armbian_23.05.0-trunk_Bigtreetech-cb2_bullseye_legacy_4.19.232.img and BTT-CB2_1.0.3_Armbian_23.05.0-trunk_Bigtreetech-cb2_bullseye_legacy_4.19.232_minimal.img on Ubuntu 21.04 using the Armbian build CB2-2024 fork: [https://github.com/bigtreetech/build/tree/CB2](https://github.com/bigtreetech/build/tree/CB2-2024). The CB2 repo refused to compile. And only bullseye, legacy kernel, console interface, (full/minimal) compiles.

Unfortunately, both images don't contain drivers for usbserial or CH341. Which is weird because the 6.6 and 4.19 kernels in the first two links above have these drivers. I guess they are excluded in a config. I'm not even sure if these images will boot and I've spent too much time trying to get this to work. I hope BTT will upload another OS version which includes the drivers and improve the user manual regarding the GPIO. For now, I will continue to use UART5.

P.S. While tinkering with the compiler script, I noticed an option to compile for a CB3 with an RK3588S so I guess a CB3 board is possibly in the works.

TheFeralEngineer, 2024-05-13:
the utility they sent me was just the windows driver, not a patch for the linux system 🙄

RK392, 2024-05-13:
> the utility they sent me was just the windows driver, not a patch for the linux system 🙄

Did you try the patch I posted above?

TheFeralEngineer, 2024-05-13:
Just patched it. Going to fire it up in a minute

Phil T.
The Feral Engineer

www.theferalengineer.com

Catnip contributions 🐱🐱
https://paypal.me/timpsonmusicllc

TheFeralEngineer, 2024-05-13:
worked perfectly. I requested that they enable all of the USB drivers like they did with the CB1, but this is good for now. Thanks!

lusanagi, 2024-05-14:
Thanks @RK392! I just fot my first out of 3 pi2s and flashed the os and was starting to go out of my mind because I wasn't able to to list the id from my "perfectly working in another board" klipper board 
I'll apply the patch tomorrow and get back, although it's confirmed already working

TheFeralEngineer, 2024-05-14:
The patch is pretty easy to apply. I actually opened up all of the USB
drivers and built my own this afternoon. The CB1 didn't have this issue.
I'm glad it's been addressed and RK392 did a hell of a job getting it done
quickly 👊👊

lusanagi, 2024-05-14:
Just updated the kernel on my Pi2 and got my printer up and running with it, thanks a lot @RK392!
Now hopefully @bigtreetech will update the image soon to avoid having to do this

RK392, 2024-05-18:
[V1.0.4](https://github.com/bigtreetech/CB2/releases/tag/V1.0.4) has been released with the CH341 driver in the kernel.

closed 2024-05-18 by RK392