bigtreetech/CB2 #7: USB webcam video capture and mjpg format not support

AlvinF01, 2024-05-08:
Tested on both BTT PI2 and Raspberry 3B+.
I can confirm that my usb webcam can run fine in rpi 3B+ with 15fps.
But cannot get any images in BTT PI2.

It can find my device
```
FHD Camera: FHD Camera (usb-fd800000.usb-1):
        /dev/video9
        /dev/video10
        /dev/media1

```

Here is webcamd.log:
```
 o: www-folder-path......: ./www-mjpgstreamer/
 o: HTTP TCP port........: 8081
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: disabled
Unable to dequeue buffer: No such device
 i: Error grabbing frames
 i: cleaning up resources allocated by input thread
Unable to stop capture: No such device
Done bring up all configured video device
```

There is no 'MJPG' find in 
v4l2-ctl --list-format-ext
```
ioctl: VIDIOC_ENUM_FMT
    Type: Video Capture Multiplanar
    [0]: 'UYVY' (UYVY 4:2:2)
          Size: Stepwise 32x16 - 800x600 with step 8/8
......
```
RK3566 should able to support uvc webcam but I think the kernel is either too old or missing somethings

TheFeralEngineer, 2024-05-13:
there's a ton of usb drivers that aren't installed on the current release of armbian. I sent service an email requesting a new image with the same drivers enabled as the cb1 (which was practically every one as opposed to the pi2, which only has 3 currently enabled)

AlvinF01, 2024-05-14:
I found that RK3566 does not have JPEG driver in mainline of linux.
https://wiki.pine64.org/wiki/Mainline_Hardware_Decoding
So, I don't think armbian will support it in any time soon.
Lucky, I buy this board mainly for hosting 7/24 website....