bigtreetech/CB2 #10: CSI/DSI attachments too dark

igl-x2, 2024-05-16:
So following the manual pdf, I setup my Picam and Waveshare 4.3 DSI display correctly.
They both work, but they are dark as the night.

The Picam picture is barely visible, even when I light the area externally, the image brigthness does not improve.
Same with the TFT. Its working. Its just seems the background lighting is not working/powered properly.
But when I tap the tft, I can see Mainsail throwing the errors out like too low temp for extrusion.

I already checked and replaced the cables, of course.
both devices work as expected when connected to an original Pi 3B

So both devices CSI/DSI seem to have some kind of power issue, like missing a power line or not providing enough current on the existing lines.

Anyone got an idea?

TheFeralEngineer, 2024-05-17:
I noticed this with a cam as well, but mine isn't horrible, just really dark. Thought maybe there was a setting control for it using uv4l, but I haven't figured it out yet.

moom-knight, 2024-05-26:
[image]

Enter the above two commands, and you'll get a lot better

moom-knight, 2024-05-26:
> So following the manual pdf, I setup my Picam and Waveshare 4.3 DSI display correctly. They both work, but they are dark as the night.
>
> The Picam picture is barely visible, even when I light the [...]

ㅁㅁ

moom-knight, 2024-05-26:
알겠지?

igl-x2, 2024-05-26:
ok webcam has improved, dsi tft still dark as the night

looxonline, 2024-05-26:
> I noticed this with a cam as well, but mine isn't horrible, just really dark. Thought maybe there was a setting control for it using uv4l, but I haven't figured it out yet.

I'll get this flagged for the engineers to look at this week. @BTTJUSTIN please take note that this is a documentation problem that needs the required commands to be added to the manual to adjust camera and display exposure/brightness for CSI and DSI.

zhangyumingsh, 2024-05-30:
> And what about DSI brightness? Currently Waveshare DSI display is also dark with 0 brightness.

> And what about DSI brightness? Currently Waveshare DSI display is also dark with 0 brightness.

The Raspberry Pi DSI screen backlight will be updated in future versions

Elwin232, 2024-06-04:
> ![스크린샷 2024-05-26 165231](https://private-user-images.githubusercontent.com/131533923/333861283-7aef7a83-5c3b-42e0-bc89-f6ec119230c3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc0ODk0NTYsIm5iZiI6MTcxNzQ4OTE1NiwicGF0aCI6Ii8xMzE1MzM5MjMvMzMzODYxMjgzLTdhZWY3YTgzLTVjM2ItNDJlMC1iYzg5LWY2ZWMxMTkyMzBjMy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYwNFQwODE5MTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hOTI0YzI2ZDcxMTQzNjQ3YzFhNTE4NjBlN2FiYmZhNjUzMTgyMGNmODk2NmM2NmZiYmRkNTc3YjdiZDBmY2IyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.BTPVBNKZMtSJ9BYRZJvyoyyOYT5JCR87nRO8U15mW9Y)
> 
> Enter the above two commands, and you'll get a lot better

HI， I enter the command you mentioned , but the output  not good
v4l2-ctl -c exposure=1964
v4l2-ctl -c gain_automatic=1
[image]

Could you please post before and after pictures?@moom-knight

igl-x2, 2024-06-28:
CSI still with bad images.

zhangyumingsh, 2024-06-28:
> CSI still with bad images.

At present, there is no work plan for the ISP, and future version updates will include ISP image processing.

decker9000, 2024-10-25:
I'm running 1.0.5 Klipper version, and my camera is dark too. I don't seem to have a control for automatic_gain to enable with v4l2-ctl
I was able to manually adjust gain, exposure, and analogue_gain controls using v4l2-ctl; but would prefer automatic gain and exposure to I don't need to constantly tweak it for lighting conditions.

Pneumanifest, 2024-11-22:
Any update on the progress with the fix for CSI camera being too dark? changing exposure and gain has no effect.

ploucandco, 2025-01-12:
When will the ISP be implemented? It's a bit a shame to advertise that CSI is supported when there is no ISP!

bderancourt, 2025-01-25:
I have the same problem. Please bigtreetech, update your image to provide a recent kernel and with ISP working.