bigtreetech/CB2 #11: login error

timido, 2024-05-17:
Good evening, the latest version of the login error
with login: biqu password: biqu

timido, 2024-05-23:
> mesmo aquiHave you managed to solve it?

zentaur1ger, 2024-05-26:
> > mesmo aquiHave you managed to solve it?

It seems, no one solved this at the moment.
BUT We were on TV at least. This thread is shown in ModBot's review on that CB2 thing ;o)

looxonline, 2024-05-26:
> > > mesmo aquiHave you managed to solve it?
> 
> It seems, no one solved this at the moment. BUT We were on TV at least. This thread is shown in ModBot's review on that CB2 thing ;o)

Will get this escalated as a top priority. @BTTJUSTIN please bring this to the team before the weekly meeting so that they can fix it ASAP.

Pneumanifest, 2024-05-26:
If its the minimal image, use user: root and password: 1234 
Once in it should guide you through setting up a fresh account.

zentaur1ger, 2024-05-26:
> If its the minimal image, use user: root and password: 1234 
> Once in it should guide you through setting up a fresh account.

I tried that today, bc. I've seen in a video, that the OS seems to be armbian based. 
Didn't work either. 
But I tried to login via ssh. From what I've read in the armbian docs, that should work. 
Or might the first login only be possible locally?

timido, 2024-05-26:
Good evening, in the minimal image with the mountain bike pi2 connected to the screen and with the keyboard it asks to create a password for root and asks to create a new account

Pneumanifest, 2024-05-27:
> I tried that today, bc. I've seen in a video, that the OS seems to be armbian based. Didn't work either. But I tried to login via ssh. From what I've read in the armbian docs, that should work. Or might the first login only be possible locally?

It should work over SSH unless SSH was turned off.

timido, 2024-05-27:
For me it only works with the first login locally without using SSH. Then it worked with the login I created with SSH

zhangyumingsh, 2024-05-29:
> If its the minimal image, use user: `root` and password: `1234` Once in it should guide you through setting up a fresh account.
The password for root permission is root

closed 2024-06-06 by BTTJUSTIN