bigtreetech/CB2 #12: Linux kernel 6.6 for cb2 does not work

TheFeralEngineer, 2024-05-18:
When I tried to compile and update to the cb2 6.6 kernel, the Pi2 will not boot. Is there a procedure to make it work?

zhangyumingsh, 2024-06-04:
> When I tried to compile and update to the cb2 6.6 kernel, the Pi2 will not boot. Is there a procedure to make it work?

Change the device tree configuration in the configuration file armbianEnv.txt to rk3566-bigtreetech-cb2.dtb

mateusdemboski, 2024-07-05:
Hey guys, there is any plan to release a new image with the 6.6 kernel?

zhangyumingsh, 2024-07-06:
> Hey guys, there is any plan to release a new image with the 6.6 kernel?

It will take some more time

KittenRaee, 2024-08-20:
> > Hey guys, there is any plan to release a new image with the 6.6 kernel?
> 
> It will take some more time

Any ETA on Kernel 6.6?

neonman63, 2024-09-15:
> > Hey guys, there is any plan to release a new image with the 6.6 kernel?
> 
> It will take some more time

can you upload sources for current 4.19 build?

artem-dudarev, 2024-10-14:
Even CB1 has kernel 5.16. Why did you release CB2 with ancient 4.19?

zhangyumingsh, 2024-10-14:
> Even CB1 has kernel 5.16. Why did you release CB2 with ancient 4.19?

Cb2 will soon release kernel 6.1 version

PepinhoDD, 2024-11-06:
any ETA

bigtreetech, 2024-11-06:
It's getting close. Maybe we will do it in beta first so that users can give it a test run and let us know about their experience.

RK392, 2024-11-17:
I'm sometimes facing CAN bus communication timeout errors while homing using a BTT Pi2 with an Octopus Pro V1.1 and an EBB 36 V1.2.

It's documented [here](https://klipper.discourse.group/t/older-topic-addendum-solved-canbus-communication-timeout-while-homing-z/16099) that the older bullseye kernel is the main culprit.

I would gladly beta test the 6.1 kernel and let you know if it solves the issue.

githubniko, 2024-11-25:
It is unlikely that they will release it. I tried the firmware from Orange Pi 3B, the system started, Ethernet works. It remains to figure out GPIO, Wifi, etc.

looxonline, 2024-11-25:
> It is unlikely that they will release it. I tried the firmware from Orange Pi 3B, the system started, Ethernet works. It remains to figure out GPIO, Wifi, etc.

CB1 on 6.6 is coming in about a week. CB2 on 6.1 is coming in three weeks.

artem-dudarev, 2024-11-25:
> CB1 on 6.6 is coming in about a week. CB2 on 6.1 is coming in three weeks.

And what about CB2 on 6.6?

looxonline, 2024-11-25:
No scheduled date for 6.6 on CB2. Many kernel modules were reworked requiring extensive rework of certain drivers.

RK392, 2024-11-25:
As long as we can get the 6.1 kernel on the Pi 2, then that should be enough to iron out any issues caused by the older kernels.

maxekb, 2025-01-04:
Hello in 6.6 btt hdmi5 working ?

Anarhyst266, 2025-01-10:
@looxonline Are there any updates about 6.1 for CB2?

KittenRaee, 2025-01-13:
https://github.com/bigtreetech/linux-rockchip/tree/bpi-rk-6.1-rkr4.1

You can keep checking progress for yourselves here. Let them cook, or get to work building your own image.