bigtreetech/CB2 #15: Bootvorgang teilweise garnicht möglich

Hentrich7274, 2024-05-26:
Bootvorgang teilweise garnicht möglich bleibt Hängen.
Wenn es mal gebootet hat kommt es wehrent der laufzeit zu Absturz.
Bin enteucht

zhangyumingsh, 2024-05-29:
> Bootvorgang teilweise garnicht möglich bleibt Hängen. Wenn es mal gebootet hat kommt es wehrent der laufzeit zu Absturz. Bin enteucht

Welchen Spiegel benutzen Sie? Das Image auf GitHub wurde getestet und es sollte jetzt kein solches Problem geben

Hentrich7274, 2024-05-30:
ich verwende das image 1.0.4. könnte sein das es an dem Pi adapter liegt.
[image]

zhangyumingsh, 2024-05-30:
> Bootvorgang teilweise garnicht möglich bleibt Hängen.
> Wenn es mal gebootet hat kommt es wehrent der laufzeit zu Absturz.
> Bin enteucht

Wenn es mit einem Bildschirm kommt, können Sie zu einer höheren Stromquelle für die Stromversorgung wechseln

looxonline, 2024-05-30:
> I'm using image 1.0.4. It could be because of the Pi adapter. ![IMG_1180](https://private-user-images.githubusercontent.com/170310089/335060772-a2d6d47f-b8e3-4526-b1d9-908eadda9b25.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcwNTA5NDQsIm5iZiI6MTcxNzA1MDY0NCwicGF0aCI6Ii8xNzAzMTAwODkvMzM1MDYwNzcyLWEyZDZkNDdmLWI4ZTMtNDUyNi1iMWQ5LTkwOGVhZGRhOWIyNS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUzMFQwNjMwNDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02NDc5OTQ0MGI2MmU2ODVlZDlkMWRjODA3ZWI0MjAzZjU0ZTk0ZjI3ZjRlZWVjOTJkYTdkNGJjYTdhNjUwZjZlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.aENPmyq8I_PAAjXYRC5sFwHca6KEGfR2qztesHabhG0)

If it is the RPi 5 adapter then this is indeed the problem. That adapter does not work well with anything other than the RPi.

Hentrich7274, 2024-05-30:
btt pi4b adapter v1.0 ohne display

looxonline, 2024-05-30:
> btt pi4b adapter v1.0 ohne display

Ah OK. I meant the power adapter. Are you sure that it is rated for the current you need?

Hentrich7274, 2024-05-30:
ich betreibe es über usb3 vom laptop im netzteil betrieb das solte genügen keine weitere peripheriegeräte nur zum einrichten.

looxonline, 2024-05-30:
> I run it via USB3 from the laptop using the power supply, that should be enough, no additional peripherals are needed, just for setting it up.

I don't think it will be enough. There is no PD negotiation so it need a PSU that will deliver more than the standard 500mA by default. Try a proper PSU.

Hentrich7274, 2024-05-30:
ich probiere das mal aus.

Hentrich7274, 2024-05-30:
ok alles super cb2 auf m8p gebaut und leuft stabil. Also eine Konstante Stromquelle. Danke für eure Hilfe

looxonline, 2024-05-30:
> ok everything is great cb2 built on m8p and runs stable. So a constant power source. Thanks for your help

Glad that all is OK. I will close the issue now.

closed 2024-05-30 by looxonline