bigtreetech/CB2 #16: gpio pinout issue

combass, 2024-05-29:
GPIO4-C3 dont respond when called GPIO147 (method for calculating GPIO pins on user manual)
work when GPIOCHIP4/GPIO19 is used 

if GPIO147 is used in .cfg, klipper repport : "MCU 'host' error during config : Unable to open GPIO ship line"

zhangyumingsh, 2024-05-30:
> GPIO4_C3 dont respond when called GPIO147 (method for calculating GPIO pins on user manual) work when GPIOCHIP4/GPIO19 is used
> 
> if GPIO147 is used in .cfg, klipper repport : "MCU 'host' error during config : Unable to open GPIO ship line"

When setting up, an error was reported as IO being occupied. The published image will not occupy gpio4_c3 by default

BTTJUSTIN, 2024-05-30:
Thanks for reporting. We will get the documentation updated and close this issue out once the updates are uploaded.

Fisrt calculate method

GPIO4_C3=（GPIO4-GPIO0）*32+(C-A)*8+3
4*32+2*8+3=128+16+3=147

code setting
pin: host:gpio147

Second calculate method

GPIO4_C3 =(C-A)*8+3
(3-1)*8+3=16+3=19

code setting
pin:host:gpiochip4/gpio19

Pneumanifest, 2024-05-30:
You know, the best way to solve these confusions with the pin names is to provide a table that shows the calculated pin number next to the pin. Having the formula is nice but can be confusing especially to those that are used to using straight gpio numbers from a raspberry pie for instance.

looxonline, 2024-05-30:
> You know, the best way to solve these confusions with the pin names is to provide a table that shows the calculated pin number next to the pin. Having the formula is nice but can be confusing especially to those that are used to using straight gpio numbers from a raspberry pie for instance.

This is a good suggestion. @BTTJUSTIN Please take note of this suggestion as a documentation feature request.

BTTJUSTIN, 2024-05-31:
Thanks for making this suggestion. We will discuss it internally and update the documentation if it is approved.

Travis90x, 2024-05-31:
> Thanks for making this suggestion. We will discuss it internally and update the documentation if it is approved."

You have to exclude gpio pin not mapped like gpio138 (wrongly suggested in the manual)

[image]

| Method 2 | Method 1 | PI2/CB2 | PI2/CB2 | Method 1 | Method 2 |
| --- | --- | --- | --- | --- | --- |
| $\color{yellow}{\textsf{3,3v}}$ | $\color{yellow}{\textsf{3,3v}}$ | $\color{yellow}{\textsf{3,3v}}$ | $\color{red}{\textsf{5V}}$ | $\color{red}{\textsf{5V}}$ | $\color{red}{\textsf{5V}}$ |
| $\color{lightgreen}{\textsf{gpio138}}$ | $\color{lightgreen}{\textsf{gpiochip4/gpio10}}$ | $\color{lightgreen}{\textsf{GPIO4-B2}}$ | $\color{red}{\textsf{5V}}$ | $\color{red}{\textsf{5V}}$ | $\color{red}{\textsf{5V}}$ |
| $\color{lightgreen}{\textsf{gpio139}}$ | $\color{lightgreen}{\textsf{gpiochip4/gpio11}}$ | $\color{lightgreen}{\textsf{GPIO4-B3}}$ | $\color{white}{\textsf{GND}}$ | $\color{white}{\textsf{GND}}$ | $\color{white}{\textsf{GND}}$ |
| $\color{lightgreen}{\textsf{gpio97}}$ | $\color{lightgreen}{\textsf{gpiochip3/gpio1}}$ | $\color{lightgreen}{\textsf{GPIO3-A1}}$ | $\color{lightgreen}{\textsf{GPIO0-D1}}$ | $\color{lightgreen}{\textsf{gpio25}}$ | $\color{lightgreen}{\textsf{gpiochip0/gpio25}}$ |
| $\color{white}{\textsf{GND}}$ | $\color{white}{\textsf{GND}}$ | $\color{white}{\textsf{GND}}$ | $\color{lightgreen}{\textsf{GPIO0-D0}}$ | $\color{lightgreen}{\textsf{gpio24}}$ | $\color{lightgreen}{\textsf{gpiochip0/gpio24}}$ |
| $\color{lightgreen}{\textsf{gpio23}}$ | $\color{lightgreen}{\textsf{gpiochip0/gpio23}}$ | $\color{lightgreen}{\textsf{GPIO0-C7}}$ | $\color{lightgreen}{\textsf{GPIO0-B0}}$ | $\color{lightgreen}{\textsf{gpio8}}$ | $\color{lightgreen}{\textsf{gpiochip0/gpio8}}$ |
| $\color{lightgreen}{\textsf{gpio32}}$ | $\color{lightgreen}{\textsf{gpiochip1/gpio0}}$ | $\color{lightgreen}{\textsf{GPIO1-A0}}$ | $\color{white}{\textsf{GND}}$ | $\color{white}{\textsf{GND}}$ | $\color{white}{\textsf{GND}}$ |
| $\color{lightgreen}{\textsf{gpio33}}$ | $\color{lightgreen}{\textsf{gpiochip1/gpio1}}$ | $\color{lightgreen}{\textsf{GPIO1-A1}}$ | $\color{lightgreen}{\textsf{GPIO4-C6}}$ | $\color{lightgreen}{\textsf{gpio150}}$ | $\color{lightgreen}{\textsf{gpiochip4/gpio22}}$ |
| $\color{yellow}{\textsf{3,3v}}$ | $\color{yellow}{\textsf{3,3v}}$ | $\color{yellow}{\textsf{3,3v}}$ | $\color{lightgreen}{\textsf{GPIO4-A3}}$ | $\color{lightgreen}{\textsf{gpio131}}$ | $\color{lightgreen}{\textsf{gpiochip4/gpio3}}$ |
| $\color{lightgreen}{\textsf{gpio113}}$ | $\color{lightgreen}{\textsf{gpiochip3/gpio17}}$ | $\color{lightgreen}{\textsf{GPIO3-C1}}$ | $\color{white}{\textsf{GND}}$ | $\color{white}{\textsf{GND}}$ | $\color{white}{\textsf{GND}}$ |
| $\color{lightgreen}{\textsf{gpio114}}$ | $\color{lightgreen}{\textsf{gpiochip3/gpio18}}$ | $\color{lightgreen}{\textsf{GPIO3-C2}}$ | $\color{lightgreen}{\textsf{GPIO0-C4}}$ | $\color{lightgreen}{\textsf{gpio20}}$ | $\color{lightgreen}{\textsf{gpiochip0/gpio20}}$ |
| $\color{lightgreen}{\textsf{gpio115}}$ | $\color{lightgreen}{\textsf{gpiochip3/gpio19}}$ | $\color{lightgreen}{\textsf{GPIO3-C3}}$ | $\color{lightgreen}{\textsf{GPIO4-A2}}$ | $\color{lightgreen}{\textsf{gpio130}}$ | $\color{lightgreen}{\textsf{gpiochip4/gpio2}}$ |
| $\color{white}{\textsf{GND}}$ | $\color{white}{\textsf{GND}}$ | $\color{white}{\textsf{GND}}$ | $\color{lightgreen}{\textsf{GPIO0-A6}}$ | $\color{lightgreen}{\textsf{gpio6}}$ | $\color{lightgreen}{\textsf{gpiochip0/gpio6}}$ |
| $\color{lightgreen}{\textsf{gpio12}}$ | $\color{lightgreen}{\textsf{gpiochip0/gpio12}}$ | $\color{lightgreen}{\textsf{GPIO0-B4}}$ | $\color{cyan}{\textsf{GPIO0-B3}}$ | $\color{cyan}{\textsf{gpio11}}$ | $\color{cyan}{\textsf{gpiochip0/gpio11}}$ |
| $\color{lightgreen}{\textsf{gpio126}}$ | $\color{lightgreen}{\textsf{gpiochip3/gpio30}}$ | $\color{lightgreen}{\textsf{GPIO3-D6}}$ | $\color{white}{\textsf{GND}}$ | $\color{white}{\textsf{GND}}$ | $\color{white}{\textsf{GND}}$ |
| $\color{lightgreen}{\textsf{gpio127}}$ | $\color{lightgreen}{\textsf{gpiochip3/gpio31}}$ | $\color{lightgreen}{\textsf{GPIO3-D7}}$ | $\color{lightgreen}{\textsf{GPIO0-C1}}$ | $\color{lightgreen}{\textsf{gpio17}}$ | $\color{lightgreen}{\textsf{gpiochip0/gpio17}}$ |
| $\color{lightgreen}{\textsf{gpio16}}$ | $\color{lightgreen}{\textsf{gpiochip0/gpio16}}$ | $\color{lightgreen}{\textsf{GPIO0-C0}}$ | $\color{white}{\textsf{GND}}$ | $\color{white}{\textsf{GND}}$ | $\color{white}{\textsf{GND}}$ |
| $\color{lightgreen}{\textsf{gpio149}}$ | $\color{lightgreen}{\textsf{gpiochip4/gpio21}}$ | $\color{lightgreen}{\textsf{GPIO4-C5}}$ | $\color{lightgreen}{\textsf{GPIO0-A0}}$ | $\color{lightgreen}{\textsf{gpio0}}$ | $\color{lightgreen}{\textsf{gpiochip0/gpio0}}$ |
| $\color{lightgreen}{\textsf{gpio19}}$ | $\color{lightgreen}{\textsf{gpiochip0/gpio19}}$ | $\color{lightgreen}{\textsf{GPIO0-C3}}$ | $\color{lightgreen}{\textsf{GPIO4-C3}}$ | $\color{lightgreen}{\textsf{gpio147}}$ | $\color{lightgreen}{\textsf{gpiochip4/gpio19}}$ |
| $\color{white}{\textsf{GND}}$ | $\color{white}{\textsf{GND}}$ | $\color{white}{\textsf{GND}}$ | $\color{lightgreen}{\textsf{GPIO4-C2}}$ | $\color{lightgreen}{\textsf{gpio146}}$ | $\color{lightgreen}{\textsf{gpiochip4/gpio18}}$ |

looxonline, 2024-06-03:
> > Thanks for making this suggestion. We will discuss it internally and update the documentation if it is approved."
> 
> You have to exclude gpio pin not mapped like gpio138 (wrongly suggested in the manual)
> 
> ![immagine](https://private-user-images.githubusercontent.com/23300077/335722019-20f0427f-d1b2-4d8f-b1e2-38f2c62366b4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc0MDMwOTUsIm5iZiI6MTcxNzQwMjc5NSwicGF0aCI6Ii8yMzMwMDA3Ny8zMzU3MjIwMTktMjBmMDQyN2YtZDFiMi00ZDhmLWIxZTItMzhmMmM2MjM2NmI0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjAzVDA4MTk1NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkyMWZmNzNjODAyMGU0YmQ5Y2MwMjI0YjgwZmQ1NDMyZWMwYjI0NTVlZmEwM2NmYzg3M2Y4NzQyNzVmNTljYTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.EZO9lMBJbW13IaYSP4WUXTKzFybtSwDBpIPsUy0i3xY)
> 
> Method 1	Method 2	PI2/CB2	PI2/CB2	Method 1	Method 2
> 3,3v
> 3,3v
> 3,3v
> 5V
> 5V
> 5V
> gpio138
> gpiochip4/gpio10
> GPIO4-B2
> 5V
> 5V
> 5V
> gpio139
> gpiochip4/gpio11
> GPIO4-B3
> GND
> GND
> GND
> gpio97
> gpiochip3/gpio1
> GPIO3-A1
> GPIO0-D1
> gpio25
> gpiochip0/gpio25
> GND
> GND
> GND
> GPIO0-D0
> gpio24
> gpiochip0/gpio24
> gpio23
> gpiochip0/gpio23
> GPIO0-C7
> GPIO0-B0
> gpio8
> gpiochip0/gpio8
> gpio32
> gpiochip1/gpio0
> GPIO1-A0
> GND
> GND
> GND
> gpio33
> gpiochip1/gpio1
> GPIO1-A1
> GPIO4-C6
> gpio150
> gpiochip4/gpio22
> 3,3v
> 3,3v
> 3,3v
> GPIO4-A3
> gpio131
> gpiochip4/gpio3
> gpio113
> gpiochip3/gpio17
> GPIO3-C1
> GND
> GND
> GND
> gpio114
> gpiochip3/gpio18
> GPIO3-C2
> GPIO0-C4
> gpio20
> gpiochip0/gpio20
> gpio115
> gpiochip3/gpio19
> GPIO3-C3
> GPIO4-A2
> gpio130
> gpiochip4/gpio2
> GND
> GND
> GND
> GPIO0-A6
> gpio6
> gpiochip0/gpio6
> gpio12
> gpiochip0/gpio12
> GPIO0-B4
> GPIO0-B3
> gpio11
> gpiochip0/gpio11
> gpio126
> gpiochip3/gpio30
> GPIO3-D6
> GND
> GND
> GND
> gpio127
> gpiochip3/gpio31
> GPIO3-D7
> GPIO0-C1
> gpio17
> gpiochip0/gpio17
> gpio16
> gpiochip0/gpio16
> GPIO0-C0
> GND
> GND
> GND
> gpio149
> gpiochip4/gpio21
> GPIO4-C5
> GPIO0-A0
> gpio0
> gpiochip0/gpio0
> gpio19
> gpiochip0/gpio19
> GPIO0-C3
> GPIO4-C3
> gpio147
> gpiochip4/gpio19
> GND
> GND
> GND
> GPIO4-C2
> gpio146
> gpiochip4/gpio18
> Method 1	Method 2	PI2/CB2	PI2/CB2	Method 1	Method 2
> 3,3v	3,3v	3,3v	5V	5V	5V
> gpio138	gpiochip4/gpio10	GPIO4_B2	5V	5V	5V
> gpio139	gpiochip4/gpio11	GPIO4_B3	GND	GND	GND
> gpio97	gpiochip3/gpio1	GPIO3_A1	GPIO0_D1	gpio25	gpiochip0/gpio25
> GND	GND	GND	GPIO0_D0	gpio24	gpiochip0/gpio24
> gpio23	gpiochip0/gpio23	GPIO0_C7	GPIO0_B0	gpio8	gpiochip0/gpio8
> gpio32	gpiochip1/gpio0	GPIO1_A0	GND	GND	GND
> gpio33	gpiochip1/gpio1	GPIO1_A1	GPIO4_C6	gpio150	gpiochip4/gpio22
> 3,3v	3,3v	3,3v	GPIO4_A3	gpio131	gpiochip4/gpio3
> gpio113	gpiochip3/gpio17	GPIO3_C1	GND	GND	GND
> gpio114	gpiochip3/gpio18	GPIO3_C2	GPIO0_C4	gpio20	gpiochip0/gpio20
> gpio115	gpiochip3/gpio19	GPIO3_C3	GPIO4_A2	gpio130	gpiochip4/gpio2
> GND	GND	GND	GPIO0_A6	gpio6	gpiochip0/gpio6
> gpio12	gpiochip0/gpio12	GPIO0_B4	GPIO0_B3	gpio11	gpiochip0/gpio11
> gpio126	gpiochip3/gpio30	GPIO3_D6	GND	GND	GND
> gpio127	gpiochip3/gpio31	GPIO3_D7	GPIO0_C1	gpio17	gpiochip0/gpio17
> gpio16	gpiochip0/gpio16	GPIO0_C0	GND	GND	GND
> gpio149	gpiochip4/gpio21	GPIO4_C5	GPIO0_A0	gpio0	gpiochip0/gpio0
> gpio19	gpiochip0/gpio19	GPIO0_C3	GPIO4_C3	gpio147	gpiochip4/gpio19
> GND	GND	GND	GPIO4_C2	gpio146	gpiochip4/gpio18

Thanks for producing this table.

BTTJUSTIN, 2024-06-06:
The documentation has been updated. Thanks again for your suggestion and pointing out the issue

closed 2024-06-06 by BTTJUSTIN

EdenNelson, 2025-03-22:
@BTTJUSTIN This issue should be reopened. The documentation change process is flawed and doesn't include your docs.  
The PDF was updated, but bigtreetech/docs have not, and that's the source for your BTT WIKI.

https://github.com/bigtreetech/docs/blob/master/docs/Pi2.md -- https://bttwiki.com/Pi2.html