bigtreetech/CB2 #17: No devices

EVPilot, 2024-06-02:
I installed the os and went through the installation process. I could not get the cb2 to connect to the MantaE3EZ board. I was about to return the cb2 until I found bigtreetech-pi2-cb2-usb-kernel-patch. This patch fixed the no devices issue. I extracted two .deb files from the download and put them into a file on the cb2. I then followed the instructions. After reboot I was up and running.

combass, 2024-06-02:
where did u find this patch please ? i'm unable to connect my stock mainboard on my sv06 since 3 days
 thanks you

EVPilot, 2024-06-02:
https://github.com/TheFeralEngineer/bigtreetech-pi2-cb2-usb-kernel-patch

combass, 2024-06-02:
thank you, what you mean when you say : no devices issue ? what happen ?

looxonline, 2024-06-02:
@BTTJUSTIN Please report this to the engineering team ASAP on Monday.

EVPilot, 2024-06-02:
> thank you, what you mean when you say : no devices issue ? what happen ? do you know about CH340 ? 

The CB2 would not connect to the Mantae3ez or my computer. I went through many tutorials and tried different methods. I was never able to get the MCU id to put into my config files for kipper. It was fixed after this patch. I saw something about CH340 in the latest image notes but nothing else about a fix or work around.

combass, 2024-06-02:
> The CB2 would not connect to the Mantae3ez or my computer. I went through many tutorials and tried different methods. I was never able to get the MCU id to put into my config files for kipper. It [...]

i will take a look thanks. unfortunatly, this patch dont solve my issue

Elwin232, 2024-06-03:
Hi, did you flash the latest CB2 image 1.31GB in this link?  https://github.com/bigtreetech/CB2/releases/tag/V1.0.4

EVPilot, 2024-06-03:
I did use the latest image. The board is now connecting after using that patch. I am however not seeing my pi cam connected.

igl-x2, 2024-06-03:
Picam is currently broken.
See my other issue

zhangyumingsh, 2024-06-03:
Which camera are you using?

zhangyumingsh, 2024-06-03:
> I installed the os and went through the installation process. I could not get the cb2 to connect to the MantaE3EZ board. I was about to return the cb2 until I found [...]

V1.0.4 has been resolved

zhangyumingsh, 2024-06-03:
> I installed the os and went through the installation process. I could not get the cb2 to connect to the MantaE3EZ board. I was about to return the cb2 until I found [...]

Refer to this document:https://github.com/bigtreetech/Manta-E3EZ/blob/master/BIGTREETECH%20Manta%20E3EZ%20V1.0%20User%20Manual.pdf

EVPilot, 2024-06-03:
> 
> 
> 
> Which camera are you using?

Ok I bought a new usb camera and it works fine. I tossed the pi cam in the trash.

closed 2024-06-03 by EVPilot