bigtreetech/CB2 #18: Random  connection problem with EBB

Sprocyon, 2024-06-04:
At random moments in the middle of printing, the connection with EBB is lost. On forms there is an idea that the matter is in 64-bit OS which is not very compatible with 32-bit mainsail. I don't understand anything about it, so I decided to ask if it can be so, and where can I get a 32-bit version of the image?

Sprocyon, 2024-06-04:
[image]

bigtreetech, 2024-06-04:
Mainsail has no impact at all on the connection to the EBB. Mainsail is an API layer that sits on top of the Klipper API and simplifies it.

bigtreetech, 2024-06-04:
> At random moments in the middle of printing, the connection with EBB is lost. On forms there is an idea that the matter is in 64-bit OS which is not very compatible with 32-bit mainsail. I don't [...]

If anything, you may have a loose connection to your EBB. I recommend you check your connectors and wiring. Also ensure that the chassis of your machine is well Earthed to prevent static from discharging into the EBB which may be causing disconnects. For a better understanding, you would do well to check the klipper logs to see why it is disconnecting.

Sprocyon, 2024-06-04:
https://forum.vorondesign.com/threads/2209-rp2040-missed-scheduling-of-next-digital-out-event.1732/

Sprocyon, 2024-06-04:
Here's a log if it helps.
[klippy (1).zip](https://github.com/user-attachments/files/15567954/klippy.1.zip)

Sprocyon, 2024-06-04:
The grounding is a great idea, we don't have any.... so it's quite possible... I don't know how it can be solved, maybe with plastic mounts... I have a CNC aluminum one now... thanks. I'll look into it.