bigtreetech/CB2 #19: Raspicam v1.3 + crownest

mojamuto, 2024-06-16:
Good afternoon, I am trying to connect a raspicam v1.3 ov5647, I have followed the instructions in the PDF but I get this error:
[image]
Searching on the internet, everything refers to libcamera-hello, but it is not in the Armbian image.

igl-x2, 2024-06-16:
Your config says dev/video9.
Try video0

Aergernis, 2024-06-16:
Hi,
having the same problem. Tested the camera with a Pi and it's working. 3,3V is present when connected to CB2 with M8P V2.

```
[06/16/24 20:53:36] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[06/16/24 20:53:36] crowsnest: Version: v4.1.9-1-gd75a3ae
[06/16/24 20:53:36] crowsnest: Prepare Startup ...
[06/16/24 20:53:36] crowsnest: INFO: Host information:
[06/16/24 20:53:36] crowsnest: Host Info: Distribution: BTT-CB2_1.0.4_Armbian 23.05.0-trunk Bullseye
[06/16/24 20:53:36] crowsnest: Host Info: Kernel: Linux 4.19.232-bigtree-cb2 aarch64
[06/16/24 20:53:36] crowsnest: Host Info: Available CPU Cores: 4
[06/16/24 20:53:36] crowsnest: Host Info: Available Memory: 2029676 kB
[06/16/24 20:53:36] crowsnest: Host Info: Diskspace (avail. / total): 24G / 29G
[06/16/24 20:53:36] crowsnest: INFO: Checking Dependencies
[06/16/24 20:53:36] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[06/16/24 20:53:36] crowsnest: Dependency: 'find' found in /usr/bin/find.
[06/16/24 20:53:36] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[06/16/24 20:53:36] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/src/ustreamer.bin.
[06/16/24 20:53:36] crowsnest: Version Control: ustreamer is up to date. (v6.10)
[06/16/24 20:53:36] crowsnest: INFO: Print Configfile: '/home/philipp/printer_data/config/crowsnest.conf'
[06/16/24 20:53:36] crowsnest:		[crowsnest]
[06/16/24 20:53:36] crowsnest:		log_path: /home/philipp/printer_data/logs/crowsnest.log
[06/16/24 20:53:36] crowsnest:		log_level: verbose
[06/16/24 20:53:36] crowsnest:		delete_log: false
[06/16/24 20:53:36] crowsnest:		no_proxy: false
[06/16/24 20:53:36] crowsnest:		
[06/16/24 20:53:36] crowsnest:		[cam 1]
[06/16/24 20:53:36] crowsnest:		mode: ustreamer
[06/16/24 20:53:36] crowsnest:		
[06/16/24 20:53:36] crowsnest:		enable_rtsp: false
[06/16/24 20:53:36] crowsnest:		rtsp_port: 8554
[06/16/24 20:53:36] crowsnest:		port: 8080
[06/16/24 20:53:36] crowsnest:		device: /dev/video0
[06/16/24 20:53:36] crowsnest:		resolution: 1920x1080
[06/16/24 20:53:36] crowsnest:		max_fps: 30
[06/16/24 20:53:36] crowsnest:		custom_flags: --format=UYVY
[06/16/24 20:53:36] crowsnest: INFO: Detect available Devices
[06/16/24 20:53:36] crowsnest: ERROR: No usable Devices Found. Stopping crowsnest.

```

mojamuto, 2024-06-16:
> Your config says dev/video9. Try video0

I have checked from video0 to video9 and none of them work.

zhangyumingsh, 2024-06-17:
> Hi, having the same problem. Tested the camera with a Pi and it's working. 3,3V is present when connected to CB2 with M8P V2.
>
> ```
> [06/16/24 20:53:36] crowsnest: crowsnest - A webcam Service for [...]

Check if the hardware IO is reversed.Then use i2ctools to see if the device can be scanned,using command :sudo i2cdetect -y 2 
，Finally, read the image with the following command:v4l2-ctl -d /dev/video0 --set-fmt-video=width=640,height=480,pixelformat=NV12 --stream-mmap=3 --stream-skip=3 --stream-to=/usr/nv12.bin --stream-count=1 --stream-poll.If these steps are all correct, according to the document settings, there will definitely be a video stream produced
[image]
[image]
[image]

zhangyumingsh, 2024-06-17:
> Hi, having the same problem. Tested the camera with a Pi and it's working. 3,3V is present when connected to CB2 with M8P V2.
>
> ```
> [06/16/24 20:53:36] crowsnest: crowsnest - A webcam Service for [...]

The rest will be updated in v1.0.5

zhangyumingsh, 2024-06-17:
> Good afternoon, I am trying to connect a raspicam v1.3 ov5647, I have followed the instructions in the PDF but I get this error: ![fallo crownest](https://private-user-images.githubusercontent.com/36079583/340125175-c704ec9a-f08b-45e2-9a99-cd996fe05f9e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg1OTUxMjIsIm5iZiI6MTcxODU5NDgyMiwicGF0aCI6Ii8zNjA3OTU4My8zNDAxMjUxNzUtYzcwNGVjOWEtZjA4Yi00NWUyLTlhOTktY2Q5OTZmZTA1ZjllLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjE3VDAzMjcwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQzMWFjYWM5NzdlNDc5NjhlY2UyYzdjMDUwYjZhZjY1NjUwMTUwNjdlMTM3OWUxODIwOTQxZWQ4ZjU2NDAzMDEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.aPH7vzD8oL4mRY7ygo-QUAt0CTDNMQB4JqqjEwKsjnc) Searching on the internet, everything refers to libcamera-hello, but it is not in the Armbian image.

This issue will be updated in v1.0.5

igl-x2, 2024-06-17:
Will it be possible to Update without losing data / reflash emmc?

zhangyumingsh, 2024-06-26:
> > Your config says dev/video9. Try video0
> 
> I have checked from video0 to video9 and none of them work.

[image]
[image]
You can try the settings shown in the picture.( custom_flags: --format=UYVY )

looxonline, 2024-06-26:
> > > Your config says dev/video9. Try video0
> > 
> > 
> > I have checked from video0 to video9 and none of them work.
> 
> ![set](https://private-user-images.githubusercontent.com/26036538/343090443-21109d2b-1854-4ea9-a967-c718fc3be2da.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTkzOTM2MzcsIm5iZiI6MTcxOTM5MzMzNywicGF0aCI6Ii8yNjAzNjUzOC8zNDMwOTA0NDMtMjExMDlkMmItMTg1NC00ZWE5LWE5NjctYzcxOGZjM2JlMmRhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjI2VDA5MTUzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkzNmVjNDM2N2M2Y2M0ZTA2ZDEyZGU4NWY3NmZjMjQ5NDI1OTU0NzU2ZDZjODU3NjhhMzgyYjc2YWU4ZTFiYzUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Le90PokqirYhCh4vhBGfR38XjczqNvwRVwtvc3xoPus) ![showstream](https://private-user-images.githubusercontent.com/26036538/343090457-ecf14360-4f0a-4b9d-9b93-e30e2d5b60a5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTkzOTM2MzcsIm5iZiI6MTcxOTM5MzMzNywicGF0aCI6Ii8yNjAzNjUzOC8zNDMwOTA0NTctZWNmMTQzNjAtNGYwYS00YjlkLTliOTMtZTMwZTJkNWI2MGE1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjI2VDA5MTUzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ1NTI4YTBlMjk0ZTE1MDk5MGRiMTI3ODY2MjI3NzNiMDVkMDhmNTQzMmI0NGM1ODhmOTY0YTgwZjRkMDY1ZjImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.9tWIO6Db1u0h7kYS2rJ9O_OdpI9GvTZ9i9yiE6RIACM) You can try the settings shown in the picture.( custom_flags: --format=UYVY )

I experience the same problem on my M8P V2.0

zhangyumingsh, 2024-06-26:
> > Your config says dev/video9. Try video0
> 
> I have checked from video0 to video9 and none of them work.

If there is still no stream according to the previous settings. Scan the OV5647 camera to see if there are any devices for the OV5647. As shown in the figure. If there is no device, there is a communication issue between the motherboard and the m8p.
[image]

zhangyumingsh, 2024-06-27:
> Hi, having the same problem. Tested the camera with a Pi and it's working. 3,3V is present when connected to CB2 with M8P V2.
>
> ```
> [06/16/24 20:53:36] crowsnest: crowsnest - A webcam Service for [...]

[image]

Try using this configuration

zhangyumingsh, 2024-06-27:
> Good afternoon, I am trying to connect a raspicam v1.3 ov5647, I have followed the instructions in the PDF but I get this error: ![fallo [...]

Using v1.0.5 version of image

mojamuto, 2024-06-27:
1.0.5 fix the problem

closed 2024-06-27 by mojamuto