bigtreetech/CB2 #22: IMX219 found, but no stream

Aergernis, 2024-06-26:
Hi,
upgraded from OV5647 to IMX219 and can't get to work.

crownset.conf
```
#### crowsnest.conf
#### This is a typical default config.
#### Also used as default in mainsail / MainsailOS
#### See:
#### https://github.com/mainsail-crew/crowsnest/blob/master/README.md
#### for details to configure to your needs.

#####################################################################
####                                                            #####
####      Information about ports and according URL's           #####
####                                                            #####
#####################################################################
####                                                            #####
####    Port 8080 equals /webcam/?action=[stream/snapshot]      #####
####    Port 8081 equals /webcam2/?action=[stream/snapshot]     #####
####    Port 8082 equals /webcam3/?action=[stream/snapshot]     #####
####    Port 8083 equals /webcam4/?action=[stream/snapshot]     #####
####                                                            #####
####    Note: These ports are default for most Mainsail         #####
####    installations. To use any other port would involve      #####
####    changing the proxy configuration or using directly      #####
####    http://<ip>:<port>/?action=[stream/snapshot]            #####
####                                                            #####
#####################################################################
####    RTSP Stream URL: ( if enabled and supported )           #####
####    rtsp://<ip>:<rtsp_port>/stream.h264                     #####
#####################################################################

[crowsnest]
log_path: /home/philipp/printer_data/logs/crowsnest.log
log_level: verbose                      # Valid Options are quiet/verbose/debug
delete_log: false                       # Deletes log on every restart, if set to true
no_proxy: false

[cam 1]
mode: ustreamer                         # ustreamer - Provides mjpg and snapshots. (All devices)
                                        # camera-streamer - Provides webrtc, mjpg and snapshots. (rpi + Raspi OS based only)
enable_rtsp: false                      # If camera-streamer is used, this enables also usage of an rtsp server
rtsp_port: 8554                         # Set different ports for each device!
port: 8080                              # HTTP/MJPG Stream/Snapshot Port
device: /dev/video0                     # See Log for available ...
resolution: 1920x1080                   # widthxheight format
max_fps: 30                             # If Hardware Supports this it will be forced, otherwise ignored/coerced.
custom_flags: --format=UYVY             # You can run the Stream Services with custom flags.
#v4l2ctl:                               # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.

```

crownset.log
```
[06/26/24 18:49:26] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[06/26/24 18:49:26] crowsnest: Version: v4.1.9-1-gd75a3ae
[06/26/24 18:49:26] crowsnest: Prepare Startup ...
[06/26/24 18:49:26] crowsnest: INFO: Host information:
[06/26/24 18:49:27] crowsnest: Host Info: Distribution: BTT-CB2_1.0.5_Armbian 23.05.0-trunk Bullseye
[06/26/24 18:49:27] crowsnest: Host Info: Kernel: Linux 4.19.232-bigtree-cb2 aarch64
[06/26/24 18:49:27] crowsnest: Host Info: Available CPU Cores: 4
[06/26/24 18:49:27] crowsnest: Host Info: Available Memory: 2029724 kB
[06/26/24 18:49:27] crowsnest: Host Info: Diskspace (avail. / total): 24G / 29G
[06/26/24 18:49:27] crowsnest: INFO: Checking Dependencies
[06/26/24 18:49:25] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[06/26/24 18:49:25] crowsnest: Dependency: 'find' found in /usr/bin/find.
[06/26/24 18:49:25] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[06/26/24 18:49:25] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/src/ustreamer.bin.
[06/26/24 18:49:25] crowsnest: Version Control: ustreamer is up to date. (v6.10)
[... 111 lines cut ...]
systemd       1     root   33u  IPv4  17044      0t0  TCP *:111 (LISTEN)
systemd       1     root   35u  IPv6  17048      0t0  TCP *:111 (LISTEN)
rpcbind     592     _rpc    4u  IPv4  17044      0t0  TCP *:111 (LISTEN)
rpcbind     592     _rpc    6u  IPv6  17048      0t0  TCP *:111 (LISTEN)
sshd        850     root    3u  IPv4  32276      0t0  TCP *:22 (LISTEN)
sshd        850     root    4u  IPv6  32278      0t0  TCP *:22 (LISTEN)
python     1265  philipp   11u  IPv4  35745      0t0  TCP *:7125 (LISTEN)
nginx      1297     root    9u  IPv4  35945      0t0  TCP *:80 (LISTEN)
nginx      1297     root   10u  IPv4  35946      0t0  TCP *:7136 (LISTEN)
nginx      1297     root   11u  IPv6  35947      0t0  TCP *:7136 (LISTEN)
nginx      1299 www-data    9u  IPv4  35945      0t0  TCP *:80 (LISTEN)
nginx      1299 www-data   10u  IPv4  35946      0t0  TCP *:7136 (LISTEN)
nginx      1299 www-data   11u  IPv6  35947      0t0  TCP *:7136 (LISTEN)
nginx      1300 www-data    9u  IPv4  35945      0t0  TCP *:80 (LISTEN)
nginx      1300 www-data   10u  IPv4  35946      0t0  TCP *:7136 (LISTEN)
nginx      1300 www-data   11u  IPv6  35947      0t0  TCP *:7136 (LISTEN)
nginx      1302 www-data    9u  IPv4  35945      0t0  TCP *:80 (LISTEN)
nginx      1302 www-data   10u  IPv4  35946      0t0  TCP *:7136 (LISTEN)
nginx      1302 www-data   11u  IPv6  35947      0t0  TCP *:7136 (LISTEN)
nginx      1303 www-data    9u  IPv4  35945      0t0  TCP *:80 (LISTEN)
nginx      1303 www-data   10u  IPv4  35946      0t0  TCP *:7136 (LISTEN)
nginx      1303 www-data   11u  IPv6  35947      0t0  TCP *:7136 (LISTEN)
main       1689  philipp    7u  IPv4  33716      0t0  TCP 127.0.0.1:8080 (LISTEN)

```

looxonline, 2024-06-26:
Hey, we are working on this at the moment. Thanks for logging it.

zhangyumingsh, 2024-06-27:
> Hi, upgraded from OV5647 to IMX219 and can't get to work.
>
> crownset.conf
>
> ```
> #### crowsnest.conf
> #### This is a typical default config.
> #### Also used as default in mainsail / MainsailOS
> #### See: [...]

Using v1.0.5 version of image

Aergernis, 2024-06-27:
> Using v1.0.5 version of image 

Forgot to write it. I'm on 1.0.5. Older versions didn't work at all.

BTTJUSTIN, 2024-07-04:
Hi, does the  V1.0.5 version image fix your IMX219 no stream issue ?@Aergernis

Aergernis, 2024-07-04:
No. OV5647 is working fine, but IMX219 is not streaming ( OV5647 was connected for testing if it's working in general)

zhangyumingsh, 2024-07-05:
> No. OV5647 is working fine, but IMX219 is not streaming ( OV5647 was connected for testing if it's working in general)

Follow the steps in the link to check if your imx219 is working properly. The address of imx219 is 0x10. 
https://github.com/bigtreetech/CB2/issues/23

zhangyumingsh, 2024-07-06:
> > > No. OV5647 is working fine, but IMX219 is not streaming ( OV5647 was connected for testing if it's working in general)
> > 
> > 
> > Follow the steps in the link to check if your imx219 is working properly. The address of imx219 is 0x10. #23
> 
> It is detected, but still `<NO STREAM>` in Webinterface
> 
> ```
>      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00:                         -- -- -- -- -- -- -- --
> 10: UU -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 30: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
> 40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 70: -- -- -- -- -- -- -- --
> ```

Have you tested the imx219 and produced the image? Can you see the image output by the camera? If the image cannot be produced, it means that your IMx219 module is broken

Aergernis, 2024-07-06:
I've tested it on Pi4B and it's working.
`v4l2-ctl -d /dev/video0 --set-fmt-video=width=640,height=480,pixelformat=NV12 --stream-mmap=3 --stream-skip=3 --stream-to=nv12.bin --stream-count=1 --stream-poll` is also working withouzt error when crownset is uninstalled.
When crownset ist running, it errors out with device bussy.

zhangyumingsh, 2024-07-06:
> I've tested it on Pi4B and it's working. `v4l2-ctl -d /dev/video0 --set-fmt-video=width=640,height=480,pixelformat=NV12 --stream-mmap=3 --stream-skip=3 --stream-to=nv12.bin --stream-count=1 [...]

The camera is occupied by the printer's output. First, change the configuration of the printer's video0 to video9

Aergernis, 2024-07-06:
Still the same.
I don't think the output file looks correct.
Here is the output file from `v4l2-ctl -d /dev/video0 --set-fmt-video=width=640,height=480,pixelformat=NV12 --stream-mmap=3 --stream-skip=3 --stream-to=nv12.bin --stream-count=1 --stream-poll` [nv12.zip](https://github.com/user-attachments/files/16114783/nv12.zip)

zhangyumingsh, 2024-08-22:
> Still the same. I don't think the output file looks correct. Here is the output file from `v4l2-ctl -d /dev/video0 --set-fmt-video=width=640,height=480,pixelformat=NV12 --stream-mmap=3 [...]

After executing sudo apt get upgrade or sudo apt get update on the subsequent image version, restart the machine

zhangyumingsh, 2024-08-22:
> Still the same. I don't think the output file looks correct. Here is the output file from `v4l2-ctl -d /dev/video0 --set-fmt-video=width=640,height=480,pixelformat=NV12 --stream-mmap=3 [...]

After executing sudo apt-get upgrade or sudo apt-get update on the subsequent image version, restart the machine

closed 2024-09-29 by BTTJUSTIN