bigtreetech/CB2 #23: IMX219 Camera not detected

DriHut, 2024-06-26:
Hi I just acquired the brand new cb2 which is very cool but when I connect my Camera module 2 to the port on the PI4B adapter I can't get v4l2 to display that the device is there and I still tried to get an image out using ffmpeg or fswebcam but the camera isn't to be found.
when running `sudo v4l2-ctl --list-devices` I get:
```
rkisp-statistics (platform: rkisp):
        /dev/video7
        /dev/video8

rkisp_mainpath (platform:rkisp-vir0):
        /dev/video0
        /dev/video1
        /dev/video2
        /dev/video3
        /dev/video4
        /dev/video5
        /dev/video6
        /dev/media0
``` 

For more precision I installed `BTT-CB2_1.0.5_Armbian 23.05.0-trunk Bullseye with Linux 4.19.232-bigtree-cb2` which is the minimal version so that I can run an octoprint instance on it

looxonline, 2024-06-26:
Hey there. We hear you and we are working on this at the moment. It may take us a week or two to get the drivers sorted out and the image rebuilt.

DriHut, 2024-06-26:
Oh thank you for Your quick reply! 
Wish you luck with that. (Keep us updated)

zhangyumingsh, 2024-06-27:
> Hi I just acquired the brand new cb2 which is very cool but when I connect my Camera module 2 to the port on the PI4B adapter I can't get v4l2 to display that the device is there and I still tried [...]

Scan the CSI camera ov5647 or imx219 device using i2cdetect - y 2, and there will be a device as shown in the picture. The node connected to the device by ov5647 and imx219 is not videoX, but v4l subdev3. The output of the CSI camera is the node video0 output by the ISP, and the stream is obtained from video0.
[image]

zhangyumingsh, 2024-06-27:
> Hi I just acquired the brand new cb2 which is very cool but when I connect my Camera module 2 to the port on the PI4B adapter I can't get v4l2 to display that the device is there and I still tried [...]

Scan the CSI camera ov5647 or imx219 device using i2cdetect - y 2, and there will be a device as shown in the picture. The node connected to the device by ov5647 and imx219 is not videoX, but v4l subdev3. The output of the CSI camera is the node video0 output by the ISP, and the stream is obtained from video0. For now, without opening the code stream of cronnest, use the following command to obtain the photo v4l2 ctl - d/dev/video0-- set fmt video=width=640, height=480,pixelformat=NV12 --stream-mmap=3 --stream-skip=3 --stream-to=/usr/nv12.bin --stream-count=1 --stream-poll， The photo is stored in the/usr directory in nv12. bin.
[image]
[image]
[image]

zhangyumingsh, 2024-06-27:
> Hi I just acquired the brand new cb2 which is very cool but when I connect my Camera module 2 to the port on the PI4B adapter I can't get v4l2 to display that the device is there and I still tried [...]

This is an image obtained by cb2 using v1.05 image during the startup of the m8p board. Read the photos of video0 and store them in the usr directory
[image]
[image]

zhangyumingsh, 2024-06-27:
> Hi I just acquired the brand new cb2 which is very cool but when I connect my Camera module 2 to the port on the PI4B adapter I can't get v4l2 to display that the device is there and I still tried [...]

Try using this configuration
[image]

zhangyumingsh, 2024-06-27:
> Hi I just acquired the brand new cb2 which is very cool but when I connect my Camera module 2 to the port on the PI4B adapter I can't get v4l2 to display that the device is there and I still tried [...]

Using v1.0.5 version of image

DriHut, 2024-06-27:
> > Hi I just acquired the brand new cb2 which is very cool but when I connect my Camera module 2 to the port on the PI4B adapter I can't get v4l2 to display that the device is there and I still tried to get an image out using ffmpeg or fswebcam but the camera isn't to be found. when running `sudo v4l2-ctl --list-devices` I get:
> >...
> 
> Scan the CSI camera ov5647 or imx219 device using i2cdetect - y 2, and there will be a device as shown in the picture. The node connected to the device by ov5647 and imx219 is not videoX, but v4l subdev3. The output of the CSI camera is the node video0 output by the ISP, and the stream is obtained from video0. For now, without opening the code stream of cronnest, use the following command to obtain the photo v4l2 ctl - d/dev/video0-- set fmt video=width=640, height=480,pixelformat=NV12 --stream-mmap=3 --stream-skip=3 --stream-to=/usr/nv12.bin --stream-count=1 --stream-poll， The photo is stored in the/usr directory in nv12. bin. 
> ...

Hi @zhangyumingsh  thanks for the feedback! Tried running `i2cdetect -y 2` which returns:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- 64 -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
when the camera is connected and 
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
when not connected.

Though when running `v4l2-ctl -d /dev/video0 --set-fmt-video=width=640,height=480,pixelformat=NV12 --stream-mmap=3 --stream-skip=3 --stream-to=/usr/nv12.bin --stream-count=1 --stream-poll` it doesn't find the device: `VIDIOC_STREAMON returned -1 (No such device)`

zhangyumingsh, 2024-06-28:
> Hi @zhangyumingsh thanks for the feedback! Tried running `i2cdetect -y 2` which returns:
>
> ```
>      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00:                         -- -- -- -- -- -- -- -- [...]

Your current board does not recognize ov5647 or imx219. The address of imx219 is 0x10, and the address of ov5647 is 0x36. You can use Klipper's image to view the stream

zhangyumingsh, 2024-06-28:
> Hi @zhangyumingsh thanks for the feedback! Tried running `i2cdetect -y 2` which returns:
>
> ```
>      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00:                         -- -- -- -- -- -- -- -- [...]

Use version v1.0.5 of the image.

closed 2024-07-04 by BTTJUSTIN