bigtreetech/CB2 #25: How to use DS18B20 on CB2

oMTEo, 2024-06-29:
As the title says...

I used many DS18B20s on my CB1 followed this guide -> https://gist.github.com/mdvorak/2fccca3fa9f76f5a2e11e567797c76d6

Now Im switched to CB2 and little lost because I get some errors while compiling.

BTTJUSTIN, 2024-07-16:
Thanks for reaching out. Our support team does not operate using the GitHub platform but we would be happy to help you via email based ticketing system. Please visit https://biqu.equipment/pages/technical-support and submit a ticket and we will be in touch with you soon

closed 2024-07-16 by BTTJUSTIN

oMTEo, 2024-08-18:
I want the help here so other people can profit from solutions if available...

HolyFrank, 2025-03-01:
I  want learning too. I have same problem!

35C0lDFuSi0n, 2025-04-15:
Did somebody found a solution ?

oMTEo, 2025-04-15:
No, and thats why I switched to a CM4...

bigtreetech, 2025-04-15:
We should have some feedback on it in the next day or two.

35C0lDFuSi0n, 2025-04-15:
> We should have some feedback on it in the next day or two.

Great, I will patiently wait. 
Thank for your supports.

HolyFrank, 2025-05-18:
> We should have some feedback on it in the next day or two.

Update?

35C0lDFuSi0n, 2025-07-07:
Update?