bigtreetech/CB2 #26: Telegram Bot Installation Error in Version 1.0.5

moom-knight, 2024-07-06:
[image]

Moonraker Telegram bots, which were installed normally in version 1.0.4, experience installation errors in version 1.0.5.

zhangyumingsh, 2024-07-08:
> ![스크린샷 2024-07-06 193553](https://private-user-images.githubusercontent.com/131533923/346268462-2c7a4fee-1c05-4ea1-8187-60de95aa0a84.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjA0MjA3MjgsIm5iZiI6MTcyMDQyMDQyOCwicGF0aCI6Ii8xMzE1MzM5MjMvMzQ2MjY4NDYyLTJjN2E0ZmVlLTFjMDUtNGVhMS04MTg3LTYwZGU5NWFhMGE4NC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcwOFQwNjMzNDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jODFjOWMyNTJjMDlmN2U4ZTIwMWZlYTFlMWIwYWNkNDE0NTBhMDFkZDE4Y2ZkMjcxNjI1ZjcxY2QzYjA4YTdkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.kcBh0d0I40NPArlNlTlUi1TUs3YFMheOoSmcUwAuZ8s)
> 
> Moonraker telegram bot, which were installed normally in version 1.0.4, experience installation errors in version 1.0.5.

First, update the usage commands: sudo apt-get upgrade, sudo apt-get update

moom-knight, 2024-07-08:
> First, update the usage commands: sudo apt-get upgrade, sudo apt-get update

The operation has already been performed.

bigtreetech, 2024-07-08:
> > First, update the usage commands: sudo apt-get upgrade, sudo apt-get update
> 
> The operation has already been performed.

You need to do this not just once at the start of using the module but just before actually running KIAUH.

klipper-vp, 2024-07-11:
The system still lacks many dependencies, which is why it won't install. I also have the same problem and I also couldn't install Telegram or Spoolman

zhangyumingsh, 2024-07-19:
> The system still lacks many dependencies, which is why it won't install. I also have the same problem and I also couldn't install Telegram or Spoolman

Try updating and upgrading the system with root privileges

zhangyumingsh, 2024-07-19:
> ![스크린샷 2024-07-06 [...]

First, enter sudo su to enter root mode, and then use sudo apt get upgrade and sudo apt get update to update and upgrade the system

BTTJUSTIN, 2024-07-26:
Thanks for reaching out. 
1. It could be a problem of the mirror source. Please find the download mirror source corresponding to your country.
2. You can try to update and upgrade with root privileges check if it fix your problem
3.Our support team does not operate using the GitHub platform but we would be happy to help you via
email based ticketing system. Please visit https://biqu.equipment/pages/technical-support and submit a ticket and we will be in touch with you soon.