bigtreetech/CB2 #27: Moonraker Telegram Bot can't install

Lellep90, 2024-07-14:
Installing telegram bot both from kiauh and from commands, it always returns this error and it is not possible to proceed with the installation.
Which currently doesn't happen with the CB1 which works perfectly. I think there are some missing libraries

[image]

zhangyumingsh, 2024-07-19:
> Installing telegram bot both from kiauh and from commands, it always returns this error and it is not possible to proceed with the installation. Which currently doesn't happen with the CB1 which works perfectly. I think there are some missing libraries
> 
> ![image_2024-07-14_10-32-10](https://private-user-images.githubusercontent.com/3503688/348527019-e9f9e036-930f-4222-8bb1-4ee32f2d6a2c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEzNTgyODksIm5iZiI6MTcyMTM1Nzk4OSwicGF0aCI6Ii8zNTAzNjg4LzM0ODUyNzAxOS1lOWY5ZTAzNi05MzBmLTQyMjItOGJiMS00ZWUzMmYyZDZhMmMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcxOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MTlUMDI1OTQ5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9M2Y1NDUyYjUxZGRmMzVjMWY3NDlkNzc5ODUyNGFkMTkzMTgzNTgzMGU0M2IzN2QwNGJjNTUwYzllYTBiNzdiMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.nheHSdSTkud2UVe7GM19nh3lR-Y7ApGJYEswPU1PB9M)

Try updating and upgrading the system with root privileges, and then install the applications you want to install in user mode.

zhangyumingsh, 2024-07-19:
> Installing telegram bot both from kiauh and from commands, it always returns this error and it is not possible to proceed with the installation. Which currently doesn't happen with the CB1 which [...]

Try updating and upgrading the system with root privileges, and then install the applications you want to install in user mode. First, enter sudo su to enter root mode, and then use sudo apt get upgrade and sudo apt get update to update and upgrade the system

iamLellep, 2024-07-19:
I have already tried to update with the sudo apt update and sudo apt upgrade commands but nothing changes.  it doesn't install it.

zhangyumingsh, 2024-07-19:
> I have already tried to update with the sudo apt update and sudo apt upgrade commands but nothing changes. it doesn't install it.

Did you execute sudo su before this?

iamLellep, 2024-07-19:
> > I have already tried to update with the sudo apt update and sudo apt upgrade commands but nothing changes. it doesn't install it.
> 
> Did you execute sudo su before this?

Yes!!

klipper-vp, 2024-07-31:
in sudo nano /etc/apt/sources.list

uncomment the lines that are commented out

in sudo nano /etc/apt/sources.list

uncomment the lines that are commented out

iamLellep, 2024-09-27:
I was able to run some tests. Indeed, by enabling those lines it seems to install correctly, too bad it doesn't work. It doesn't send notifications, while the exact same configuration with build 1.0.4 works perfectly.

iamLellep, 2024-09-27:
> in sudo nano /etc/apt/sources.list
> 
> uncomment the lines that are commented out
> 
> deb http://deb.debian.org/debian bullseye main contrib non-free deb-src http://deb.debian.org/debian bullseye main contrib non-free
> 
> deb http://deb.debian.org/debian bullseye-updates main contrib non-free deb-src http://deb.debian.org/debian bullseye-updates main contrib non-free
> 
> deb http://deb.debian.org/debian bullseye-backports main contrib non-free deb-src http://deb.debian.org/debian bullseye-backports main contrib non-free
> 
> deb http://security.debian.org/ bullseye-security main contrib non-free deb-src http://security.debian.org/ bullseye-security main contrib non-free
> 
> sudo apt update; sudo apt full-upgrade

I was able to run some tests. Indeed, by enabling those lines it seems to install correctly, too bad it doesn't work. It doesn't send notifications, while the exact same configuration with build 1.0.4 works perfectly.

s1ngul0r, 2024-10-01:
i have the same problem on my btt pi 2 :(

bigtreetech, 2024-10-01:
@BTTJUSTIN Please ensure that this gets priority once the national festival is over.

klipper-vp, 2024-10-03:
aqui ta funcionando normal notificções e tudo mais
[image]

iamLellep, 2024-10-19:
> aqui ta funcionando normal notificções e tudo mais ![telegran](https://private-user-images.githubusercontent.com/115836686/373359685-34420bea-1ee5-4b3d-9b06-a632e0040421.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjkzNjQ3ODYsIm5iZiI6MTcyOTM2NDQ4NiwicGF0aCI6Ii8xMTU4MzY2ODYvMzczMzU5Njg1LTM0NDIwYmVhLTFlZTUtNGIzZC05YjA2LWE2MzJlMDA0MDQyMS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAxOVQxOTAxMjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02MDk0NWZkMzExNTBhNTNlNDRiMzI5NWZkYTQzNzQ1NGZmODA0NDY2MzczZTM3ZDg1YjA4ODM5NmRiYjc4Yzg5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.8wX0nkLKyIairCRce0Y9I3nZuOSrnyluq11ecbkgSzw)

I'm happy that it works for you. but it still doesn't work for me. the same configuration transported to 1.0.4 works perfectly. Although I have now managed to install telegram bot on 1.0.5 without error, the notifications do not arrive while with the same telegram configuration file they arrive perfectly both with version 1.0.4 and with other machines. I also ask for a prompt because the holidays have been over for 15 days. I am honestly thinking of selling the cb2 to switch to the cm4, much more reliable and tested. Too bad because the cb1 has always worked perfectly

zhangyumingsh, 2024-10-21:
> I'm happy that it works for you. but it still doesn't work for me. the same configuration transported to 1.0.4 works perfectly. Although I have now managed to install telegram bot on 1.0.5 without [...]

The bad experience we brought you will gradually improve. We are currently updating the new kernel, and we will use the 6.1 kernel for future versions of the Image

iamLellep, 2024-10-31:
> > > aqui ta funcionando normal notificções e tudo mais ![telegran](https://private-user-images.githubusercontent.com/115836686/373359685-34420bea-1ee5-4b3d-9b06-a632e0040421.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjkzNjQ3ODYsIm5iZiI6MTcyOTM2NDQ4NiwicGF0aCI6Ii8xMTU4MzY2ODYvMzczMzU5Njg1LTM0NDIwYmVhLTFlZTUtNGIzZC05YjA2LWE2MzJlMDA0MDQyMS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDE5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAxOVQxOTAxMjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02MDk0NWZkMzExNTBhNTNlNDRiMzI5NWZkYTQzNzQ1NGZmODA0NDY2MzczZTM3ZDg1YjA4ODM5NmRiYjc4Yzg5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.8wX0nkLKyIairCRce0Y9I3nZuOSrnyluq11ecbkgSzw)
> > 
> > 
> > I'm happy that it works for you. but it still doesn't work for me. the same configuration transported to 1.0.4 works perfectly. Although I have now managed to install telegram bot on 1.0.5 without error, the notifications do not arrive while with the same telegram configuration file they arrive perfectly both with version 1.0.4 and with other machines. I also ask for a prompt because the holidays have been over for 15 days. I am honestly thinking of selling the cb2 to switch to the cm4, much more reliable and tested. Too bad because the cb1 has always worked perfectly
> 
> We are working on getting this fixed. We are currently updating the new kernel, and we will use the 6.1 kernel for future versions of the Image

any news on when it will be released? Today I tried again to create a new bot but the problem still persists.
Honestly, I feel a little sad about having bought this CB2 given the software problems that are popping up!

iamLellep, 2024-12-04:
> @zhangyumingsh Please provide an update on this.

Any news?