bigtreetech/CB2 #28: klipperScreen not boot on TFT35

iamLellep, 2024-07-14:
Although klipperscreen is correctly installed by default in the CB2 image, the service does not start even when forced from the command line.
Even when updating the version the problem remains.
The screen remains on waiting for login, so it is connected and working properly.
The problem could come from some missing libraries

Pneumanifest, 2024-07-14:
Ssh in and send this command,
`sudo bash -c "echo needs_root_rights=yes>>/etc/X11/Xwrapper.config"`

iamLellep, 2024-07-14:
after the commands and the restart it still remains stuck on this screen with the error that the driver is missing and it doesn't show the klipperscreen screen

[image]

klipper-vp, 2024-07-14:
Estava com mesmo problema corrigi dessa forma
[image]
[image]
[image]

klipper-vp, 2024-07-14:
me da um FeedBack se foi.

iamLellep, 2024-07-14:
> me da um FeedBack se foi.

I confirm that it works.  I had not enabled display control in the system.cfg file while I had correctly enabled the tft35 overlay from boardenv.cfg

thank you very much!!

closed 2024-07-14 by iamLellep

klipper-vp, 2024-07-14:
Anyway, I'm glad it worked out!!