bigtreetech/CB2 #29: Bluetooth USB dongle fails to function moving from CB1 to CB2 on M8P

CMahk, 2024-07-28:
Hello, I am using the following Bluetooth 5.3 USB dongle on my M8P + CB2:
https://www.amazon.com/gp/product/B0BL3BTVFJ

It didn't work out of the box for the CB1, requiring that rtl8761bu_fw.bin be downloaded elsewhere and provided in /usr/lib/firmware/rtl_bt/ as running dmesg showed that the firmware was missing.

I am experiencing the same problem when using the CB2 now, however it instead calls for rtl8761a_fw.bin and its subsequent config. Once again, I grabbed the firmware from Realtek's driver repo, however the USB dongle still fails to load with the following output from dmesg:
`
biqu@bigtreetech-cb2:~$ dmesg | grep -i blue
[    1.183458] Bluetooth: Core ver 2.22
[    1.183514] Bluetooth: HCI device and connection manager initialized
[    1.183527] Bluetooth: HCI socket layer initialized
[    1.183535] Bluetooth: L2CAP socket layer initialized
[    1.183573] Bluetooth: SCO socket layer initialized
[    3.332951] Bluetooth: RFCOMM socket layer initialized
[    3.332982] Bluetooth: RFCOMM ver 1.11
[    3.332993] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[    3.333000] Bluetooth: HIDP socket layer initialized
[    3.334675] [BT_RFKILL]: bluetooth_platdata_parse_dt: get property: uart_rts_gpios = 77.
[    3.334696] [BT_RFKILL]: bluetooth_platdata_parse_dt: get property: BT,reset_gpio = 80.
[    3.334706] [BT_RFKILL]: bluetooth_platdata_parse_dt: get property: BT,wake_gpio = 81.
[    3.334716] [BT_RFKILL]: bluetooth_platdata_parse_dt: get property: BT,wake_host_irq = 79.
[    4.139925] usb 5-1.2: Product: Bluetooth 5.3 Radio
[   10.445222] systemd[1]: /lib/systemd/system/rk3399-bluetooth.service:11: Support for option SysVStartPriority= has been removed and it is ignored
[   13.724299] Bluetooth: hci0: command 0x1001 tx timeout
[   21.834080] Bluetooth: hci0: RTL: HCI_OP_READ_LOCAL_VERSION failed (-110)
`

I've found other such cases online for other distros, and it seems like it's an issue when using an older kernel?
https://forum.libreelec.tv/thread/23452-how-to-install-mpow-bluetooth-dongle-driver/
https://gist.github.com/rometsch/dfd24fb09c85c1ad2f25223dc1481aaa

CMahk, 2024-07-28:
I recognize that bluetooth is supposedly supported onboard with the CB2, but the latest release at this moment (v1.0.5) does not seem to recognize as such.

I built Debian 12 from your guys' Armbian fork via the CB2 branch and things seem to be better, albeit without the extra Klipper stuff, so I'll just install it all via KIAUH and live with it for now.

cin3m, 2024-10-30:
> I recognize that bluetooth is supposedly supported onboard with the CB2, but the latest release at this moment (v1.0.5) does not seem to recognize as such.
>
> I built Debian 12 from your guys' Armbian [...]

Hey,

I tried to compile Debain 12 myself but I'm unable to. Would you mind sharing the instructions or the Image with us?

Kind Regards

CMahk, 2024-12-06:
You will need to use KIAUH to install everything after compiling CB2 image

First, pull BTT's fork: https://github.com/bigtreetech/build and check out to CB2 branch
Follow Armbian's instructions on how to build the target. Things should be good to go by default.
Let it build, it'll take a good while. Write the built image to a new MicroSD and boot up your printer, then open a SSH shell into it.

* Comment out CSI.sh in /boot/scripts/btt_init.sh since if you don't use CSI/DSI cameras

* Canbus info is not provided inherently (if you need it)
PUT THIS IN `/etc/network/interfaces.d/can0`:
```
allow-hotplug can0
iface can0 can static
    bitrate 1000000
    up ifconfig $IFACE txqueuelen 1024

```

* KlipperScreen may not load if it can't find the theme

* If you're using an HDMI display, run the following command:
> export DISPLAY=:0

* xrandr may not set resolution to HDMI screen (800x480), so touchscreen gets buggy
TO FIX, PUT THIS IN `/usr/share/X11/xorg.conf.d/90-monitor.conf`:
```
Section "Monitor"
    Identifier "HDMI-1"
    Option "Rotate" "normal"
    Option "PreferredMode" "800x480"
EndSection

Section "Screen"
    Identifier "Screen0"
    Subsection "Display"
        Modes "800x480"
    EndSubSection
EndSection
```

Everything should be good to go after this. Good luck!