bigtreetech/CB2 #31: The 40-pin GPIO header is missing important markings.

JohnTheCoolingFan, 2024-08-10:
As there is no uart adapter on the type-c connector, a different way to connect and debug the board is needed in case it's not possible to connect to the network or when developing OS images. The pin roles match up somewhat with BTT Pi, notably UART pins are the same: `GPIO0_D0` is RX and `GPIO0_D1` is TX, which are labeled UART2 pins in the rk3566 datasheet I found. The UART connection used 1.5M baud speed.

Please add GPIO pin roles into the manual, just like for BTT Pi and CB1

BTTJUSTIN, 2024-08-15:
Hi, Thanks for your information.
which one you refer to ? 
please check the following picture if meet your need.
1.
[image]

2.
[image]

JohnTheCoolingFan, 2024-08-15:
The Picture №1 is from the Pi 2 and CB2 manual, and it only has gpio numbers, but not roles such as uart rx/tx and others.

BTTJUSTIN, 2024-08-21:
Hi,The pins of CB2 and CB1 are the same, RX, TX  the pins are in the same position, but with different serial numbers.
Please refer to the pin  of CB1 
[image]

JohnTheCoolingFan, 2024-08-21:
It would be better if this was explicitly marked in the manual for Pi 2 and CB2, instead of being implied through CB1 manual.

Leka1B, 2024-11-05:
I just spent a couple hours trying to figure out the location of the spi pins without any success, because several reviews of the board explicitly stated that the pins of the CB2 are different from CB1 while showing Picture №2 in their videos. (Do not remember the exact videos). Them being the same probably needs to be stated in the manual

bigtreetech, 2024-11-05:
@BTTJUSTIN I agree with the user comments. We need to update the CB2 and Pi2 manuals to indicate the pin functions. Please make sure that this point is raised at the next PM meeting.

BTTJUSTIN, 2024-11-08:
OK，Got it : )    I already talk to PM last week. I am working on it .

tpruvot, 2024-12-27:
Indeed, i can confirm it works with console=serial in /boot/armbianEnv.txt. Could have saved my day, i had to do a full reflash...

[image]

exerkaoss, 2025-02-16:
> Indeed, i can confirm it works with console=serial in /boot/armbianEnv.txt. Could have saved my day, i had to do a full reflash...
> 
> [image]

I'm trying to attach an old tft touch screen from btt to those uart pins, what is the overlay that should be enabled in the config, the UART0?
i'm goin crazy because my device should show up as ttyAMA0 but is not.