bigtreetech/CB2 #33: UART PI2

Brenddpaula, 2024-08-16:
Hello there. I'm having trouble to connect my PI2 to the SKR Mini E3 V3. It does work fine with the USB cable but not over UART.

I've updated the armbianEnv.txt to console=serial, also activated UART0 to 5 (at different times) and had not been successful. Also updated the printer.cfg accordingly, still no success. Updated the firmware to use uart at both 115000 and 250000 baudrate.

Cable is working fine, checked continuity and had no issues.

What I do find weird is that the printer does shows up over Kaiuah detect MCU feature, but I'm not able to connect it to klipper at all.

Has someone been able to do it? I haven't found any manuals for this...

ejhoness, 2024-08-18:
I tried to use UART, but my ignorance, research and documentation didn't help, please more information about gpios and other ports currently without image in dsi only white screen. 

trying to help, for me ttyS3 at I don't know which pins I think they would be
GPIO4_B2(gpiochip4/gpio10)
GPIO4_B3(gpiochip4/gpio11)

include in /boot/armbianEnv.txt
overlays=dsi, uart0, uart1, uart3

it would be something like this 
[mcu name]
serial: /dev/ttyS3

summary, I still haven't been able to connect over UART
I couldn't do loopback
and to install the tools need include repository 
/etc/apt/sources.list
"deb http://deb.debian.org/debian bullseye main"

BTTJUSTIN, 2024-08-21:
Thanks for reaching out. Our support team does not operate using the GitHub
platform but we would be happy to help you via email based ticketing system. 
Please visit https://biqu.equipment/pages/technical-support 
and submit a ticket and we will be in touch with you soon

closed 2024-08-21 by BTTJUSTIN

djObsidian, 2024-09-28:
Hey, i figured out why. Support told me: 

> Hello sir, I am sorry, PI2 does not support UART mode. Please be informed. I am very sorry.

Lol! Actually, you just need to make some tweaks. Device tree is configured wrong in several ways.
TL;DR:
armbianEnv.txt DOESN'T WORKS LIKE THAT.  You cant't write 'overlays' multiple times like that. If you want to use multiple overlays, you have to write them like this, space separated. For first case only last overlays defenition will be used.
So, wrong config:
```
######uart0 config######
overlays=uart0
######uart1 config######
overlays=uart1 
```

Valid config:
```
######uart0 uart1 config######
overlays=uart0 uart1
```

Secondy, uart1 section in /boot/dtb/rockchip/rk3566-bigtreetech-pi2.dtb was missing config for uart1m1. And guess what, rk3566-uart1.dtbo used exactly uart1m1, not uart1m0! I tried to copypaste some configuration from orangepi3b dts, it did made `failed on fdt_overlay_apply(): FDT_ERR_NOTFOUND` error go away but uart1 still isn't working. 
So to make at least uart0 work you need to edit your `armbianEnv.txt` file leaving only one overlays variable with `overlays=uart0`. Then you should have /dev/ttyS0 avaliable after reboot

djObsidian, 2024-09-28:
And the pins for UART0, from klipper discord #big_tree_tech channel
[image]

Pneumanifest, 2024-09-29:
> > Hello sir, I am sorry, PI2 does not support UART mode. Ple
> Lol! Actually, you just need to make some tweaks. Device tree is configured wrong in several ways. TL;DR: armbianEnv.txt DOESN'T WORKS LIKE THAT. You cant't write 'overlays' multiple times like that. If you want to use multiple overlays, you have to write them like this, space separated. For first case only last overlays defenition will be used. So, wrong config:
> 
> ```
> ######uart0 config######
> overlays=uart0
> ######uart1 config######
> overlays=uart1 
> ```
> 
> Valid config:
> 
> ```
> ######uart0 uart1 config######
> overlays=uart0 uart1
> ```
>

The way to write overlays is detailed in the user guide.
Nice to see you got UART working though.
[image]

bigtreetech, 2024-09-29:
> Hey, i figured out why. Support told me:
>
>
> Lol! Actually, you just need to make some tweaks. Device tree is configured wrong in several ways. TL;DR: armbianEnv.txt DOESN'T WORKS LIKE THAT. You [...]

@BTTJUSTIN Please take note that the support team gave the user the wrong advice. They Pi2 does indeed support UART mode. Please ensure that the support team are updated.