bigtreetech/CB2 #34: CB2 GPIO Default Configuration

irebane, 2024-09-10:
@bigtreetech Is there any particular reason why we don't have any FREE output GPIO pins that are pulled LOW?

I am trying to use one of the GPIO pins with **SN74AHCT125N** and drive Addressable LEDs to solve main Manta series boards LED issue and all free pins are mostly input and pulled HIGH so there is no way for me out of the box to use any of the GPIO pins to get LED signal working.

Thank you.

Pneumanifest, 2024-09-11:
Not sure that it pertains to the reason, but in general with motion control a low or open signal is what is used to designate a trigger or failure. As long as there's a high signal then things are ok or in operational state. 
Depending on which pins you are wanting to use, you should be able to reassign and set pullup or pull down in the firmware. If the circuit is made for a specific purpose then there is only so much you can do to change its function.

irebane, 2024-09-11:
> you should be able to reassign and set pullup or pull down in the firmware

Could you please share exact way to do it? I figured there should be a way to configure GPIO chip for pins to behave specific way but per my research there is no easy way to do it without making custom **.dts** overlay file and I have 0 experience with it, is anyone have done it before?

Pneumanifest, 2024-09-11:
I'm assuming klipper since you mentioned the Manta board. You could set it in the printer.cfg for example 
```

[output_pin caselight]

pin: !GPIO17  # put the gpio for the pin you are using starting with an !
value: 0

```
 the use SET_PIN command to turn it off (0) and on (1).
```SET_PIN PIN=GPIO17 VALUE=1```

irebane, 2024-09-11:
> I'm assuming klipper since you mentioned the Manta board. You could set it in the printer.cfg for example
> 
> ```
> 
> [output_pin caselight]
> 
> pin: host:!GPIO17  # put the gpio for the pin you are using starting with an !
> value: 0
> ```
> 
> the use SET_PIN command to turn it off (0) and on (1). `SET_PIN PIN=host:GPIO17 VALUE=1`
> 
> Or if you are controlling neopixels use the neopixel function.

Yes, We talk about Klipper.
We are setting up **Addressable LED** it requires pin pulled down so it be logical 0 by default.

Also offered solution won't work: 
[image]

irebane, 2024-09-11:
> My mistake. The ! goes in front of the whole pin description like this `pin: !cb1:GPIO26` I fixed that in my previous post.

I have tried this way as well with all sorts of escape characters:

[image]

Pneumanifest, 2024-09-11:
keep in mind that the ! is to invert logic so you may not need that. for hardware pullup is ^ and pulldown is ~ so for example 
` pin: ~cb1:GPIO26 ` should set the pulldown if it is an actual option.

Pneumanifest, 2024-09-11:
> I have tried this way as well with all sorts of escape characters: [...]

do you have the cb1 setup as a host mcu?
https://www.klipper3d.org/RPi_microcontroller.html

irebane, 2024-09-11:
cb1 is an alias for host.
Running BTT CB2 with current setup. 
I haven't changed alias since migrating over to CB2 that is why it's cb1

> keep in mind that the ! is to invert logic so you may not need that. for hardware pullup is ^ and pulldown is ~ so for example `pin: ~cb1:GPIO26` should set the pulldown if it is an actual option.

Thank you. I consider myself very experienced Klipper user and do understand the logic behind those characters pretty well

Pneumanifest, 2024-09-11:
> cb1 is an alias for host. Running BTT CB2 with current setup. I haven't changed alias since migrating over to CB2 that is why it's cb1

so you have set up the the linux process on the cb2, and it is up to date?

irebane, 2024-09-11:
> > cb1 is an alias for host. Running BTT CB2 with current setup. I haven't changed alias since migrating over to CB2 that is why it's cb1
> 
> so you have set up the the linux process on the cb2, and it is up to date?

Yes sir.

Pneumanifest, 2024-09-11:
> > > cb1 is an alias for host. Running BTT CB2 with current setup. I haven't changed alias since migrating over to CB2 that is why it's cb1
> > 
> > 
> > so you have set up the the linux process on the cb2, and it is up to date?
> 
> Yes sir. I was able to get signal from that pin but as it's pulled up LEDs just blinking constantly as pin has 3v constantly on it.

I can play around with mine later tonight to see the exact config. I just use the onboard rgb port on the manta boards, so there might be something I'm forgetting from when I last did it with gpio on the host. 
Are you setting it with the [neopixel my_LEDs] function or one of the equivalent?

irebane, 2024-09-11:
> I can play around with mine later tonight to see the exact config. I just use the onboard rgb port on the manta boards, so there might be something I'm forgetting from when I last did it with gpio on the host. Are you setting it with the [neopixel my_LEDs] function or one of the equivalent?

Yes using **[neopixel]**
Original LED port on Manta board has flaw where LEDs won't behave properly while printer is working: 
https://github.com/bigtreetech/Manta-M8P/issues/81#event-14183613743

I wanted to use gpio pin and as it's 3.3v signal boosting it with **SN74AHCT125N**  to 5v needed for LED strip.
Was very surprised to see all gpio pins on CB2 that are configured as outputs are pulled up.

I have seen in Klipper documentation we have **[output_pin]** haven't tried using it so far.

Thanks for trying.

Pneumanifest, 2024-09-11:
There's no flaw there. The user is just hitting the maximum amount of bits that the MCU can handle. How many LEDs are you trying to control? I'm only using 9 at the moment so no risk of maxing things out for my use case.

looxonline, 2024-09-11:
> > I can play around with mine later tonight to see the exact config. I just use the onboard rgb port on the manta boards, so there might be something I'm forgetting from when I last did it with gpio on the host. Are you setting it with the [neopixel my_LEDs] function or one of the equivalent?
> 
> Yes using **[neopixel]** Original LED port on Manta board has flaw where LEDs won't behave properly while printer is working: [bigtreetech/Manta-M8P#81 (comment)](https://github.com/bigtreetech/Manta-M8P/issues/81#event-14183613743)
> 
> I wanted to use gpio pin and as it's 3.3v signal boosting it with **SN74AHCT125N** to 5v needed for LED strip. Was very surprised to see all gpio pins on CB2 that are configured as outputs are pulled up.
> 
> I have seen in Klipper documentation we have **[output_pin]** haven't tried using it so far.
> 
> Thanks for trying.

This is not a hardware problem. It is simply a limit of the MCU combined with the method used to drive the LEDs in firmware. It is as much of a hardware problem as dropping frames when trying to run a game at 4k on a GPU that can't handle it is.

looxonline, 2024-09-11:
> > There's no flaw there. The user is just hitting the maximum amount of bits that the MCU can handle. How many LEDs are you trying to control? I'm only using 9 at the moment so no risk of maxing things out for my use case.
> 
> My setup is running **42** x **SK6812** LEDs and while printer is idle it's works perfect and for me personally it's an inconvenience I just like when things work. That is why I decided to take that route with GPIO which looks like it's also the path of pain and is fine it's hobby and I like to learn.
> 
> > This is not a hardware problem. It is simply a limit of the MCU combined with the method used to drive the LEDs in firmware. It is as much of a hardware problem as dropping frames when trying to run a game at 4k on a GPU that can't handle it is.
> 
> Totally got the reason and I look at the question touch simpler ether feature works or not. **In this case it is doesn't work or very limited it's that simple.**
> 
> I DO NOT blame company for making product with an issue **it's normal production process to have bugs and issues**, nothing is perfect. But there are plenty ways to solve this problem especially considering the fact we have full size GPIO right next to LED header and so far company haven't provided any work around solution. That is why this topic was created I just hope people will get together to figure out simple fix with minimum investment cus looks like I am not smart enough to figure how to preset GPIO pins on CB1/CB2 boards.

I believe that the LEDs need a comms signal of 5V but the SBC will only output 3V3 so I don't think that will work.

irebane, 2024-09-11:
> I believe that the LEDs need a comms signal of 5V but the SBC will only output 3V3 so I don't think that will work.

That is why I mentioned I am using **SN74AHCT125N**

irebane, 2024-09-11:
> ![image](https://private-user-images.githubusercontent.com/19784349/366495242-96e15a08-df3c-4d29-a3af-602abdd1ebcf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYwNjQ3NzksIm5iZiI6MTcyNjA2NDQ3OSwicGF0aCI6Ii8xOTc4NDM0OS8zNjY0OTUyNDItOTZlMTVhMDgtZGYzYy00ZDI5LWEzYWYtNjAyYWJkZDFlYmNmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTExVDE0MjExOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWU3MjA1NDQwMmY4M2Q4MmZmZDk3YWI3YjczZmYzMzVjMTBhNjMxOWVkM2RmNTlmMTg1OWMxZDA4NjJhYTIwZDUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.g-ceMhKz1WHjknSyVgFdsNCO9Xu5E6JgaxNH_D6aGfU)

Ohh... I missed that portion of the documentation, thanks for pointing to it.

JuanR3D, 2025-03-10:
Hey guys. I need to use the GPIO pins in order to read some alarms in my device but I have seen that some of them remain always high even defining them as input.  How can i set them low?