bigtreetech/CB2 #35: please make the BTT RPI-CAN HAT V1.0 module able to Communicate

shwors, 2024-09-17:
hi biqu/bigtreetech and dev-users hereoff

i have a hermitcrab v1 that connects to a btt pi2 through the BTT RPI-CAN HAT V1.0 CAN module.
but i can't get a connection trough.

i tryed it because i saw that all the pins on bbt pi2 ARE mechanicaly connected that btt pi1/cb1 has'nt,
but nothing works. i even tryed to manipulate the .dtb/dtbo files but doesnt seem that i have the possibility to change anything
this could be simple but isnt.

i want the spi controller to be the same but i want the interrupt controller to use gpio20 and the chipselect on gpio2
as if it was a raspberry pi. yeah you are using the raspberry pi/broadcom ecosystem but without descriptions or guides. i mean both btt hermitcrab v1 and btt pi2 are your products so it shouldent be hard for experts to code.

im not an expert though but i love your products
best regards

looxonline, 2024-09-18:
Hey. Thanks for writing. The HAT is only compatible with the BTT Pi 1 and our recommendation for the BTT Pi 2 is a simple U2C board via USB. We found that the HAT was not as popular as the U2C boards for direct SBC to CAN connections and therefore we standardised on a U2C connection for future boards.

closed 2024-09-18 by looxonline

bigtreetech, 2024-09-18:
You can make use of the CAN bypass jumpers to connect CAN-H and CAN-L directly to the U2C https://github.com/bigtreetech/HermitCrab/blob/master/HermitCrab1/CanbusKlipper/Hardware/BTT%20CAN-Tool%20Board%20V1.0/BTT%20CAN-Tool%20Board%20V1.0%20Wiring%20Diagram.png

shwors, 2024-09-18:
> You can just cut the rj11 plug off one end. Its the 2 center wires that
> are in use for can high and can low, typically red and green. Those will go
> to the the can high and low output of the u2c.

But cant i just use something like io2can instead? My understanding of this
board is that i can only control one node and thats perfect for me as i
only have hermitcrab v1 on can. Then i dont need to cut any wires and i get
rid of all the boards that came whit the hermitcrab v1 (tool board, can hat)