bigtreetech/CB2 #37: Raspberry Pi Camera v2.1 not working

awoodcroft95, 2024-10-09:
I have a Raspberry Pi Camera v2.1 that I have confirmed works on a Raspberry Pi 4. When I connect it to my BTT Pi 2 however it does not work.

Here are the details of my setup:
Installation image: 1.0.5
OS Version: BTT-CB2_1.0.5_Armbian 23.05.0-trunk Bullseye with Linux 4.19.232-bigtree-cb2
Camera Sensor: Sony Exmor IMX219

`i2cdetect -y 2` result:
[image]

`v4l2-ctl --list-devices` result:
[image]

`v4l2-ctl -d /dev/video0 --set-fmt-video=width=640,height=480,pixelformat=NV12 --stream-mmap=3 --stream-skip=3 --stream-to=/usr/nv12.bin --stream-count=1 --stream-poll` result:
[image]

checking `ls /dev/v4l/by-id/` and `ls /dev/v4l/by-path/` gives the following:
[image]

Is there anything else I can test to see what might be causing issues with the camera?
Thanks in advance.

BTTJUSTIN, 2024-10-11:
Thanks for reaching out.
This is because the system was upgraded, causing the script to be overwritten.please restart to it ，check if it work.

awoodcroft95, 2024-10-16:
Hi Justin, thanks for taking the time to look at my issue.

How would I go about turning off the CSI camera in the script?
Which script are we talking about?

Hilpas, 2024-10-20:
I also have trouble getting my pi camera v2.1 to work. Altough crownsnest seems to detect it. I dont get a Signal. 

[10/20/24 17:43:56] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[10/20/24 17:43:56] crowsnest: Version: v4.1.9-1-gd75a3ae
[10/20/24 17:43:56] crowsnest: Prepare Startup ...
[10/20/24 17:43:56] crowsnest: INFO: Host information:
[10/20/24 17:43:56] crowsnest: Host Info: Distribution: Armbian 24.8.4 bullseye
Armbian 24.8.4 bullseye
[10/20/24 17:43:56] crowsnest: Host Info: Kernel: Linux 4.19.232-bigtree-cb2 aarch64
[10/20/24 17:43:56] crowsnest: Host Info: Available CPU Cores: 4
[10/20/24 17:43:56] crowsnest: Host Info: Available Memory: 2029724 kB
[10/20/24 17:43:56] crowsnest: Host Info: Diskspace (avail. / total): 24G / 29G
[10/20/24 17:43:56] crowsnest: INFO: Checking Dependencies
[10/20/24 17:43:56] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[10/20/24 17:43:56] crowsnest: Dependency: 'find' found in /usr/bin/find.
[10/20/24 17:43:57] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[10/20/24 17:43:57] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/src/ustreamer.bin.
[10/20/24 17:43:57] crowsnest: Version Control: ustreamer is up to date. (v6.10)
[10/20/24 17:43:57] crowsnest: INFO: Print Configfile: '/home/biqu/printer_data/config/crowsnest.conf'
[10/20/24 17:43:57] crowsnest:		[crowsnest]
[10/20/24 17:43:57] crowsnest:		log_path: /home/biqu/printer_data/logs/crowsnest.log
[10/20/24 17:43:57] crowsnest:		log_level: verbose
[10/20/24 17:43:57] crowsnest:		delete_log: false
[10/20/24 17:43:57] crowsnest:		no_proxy: false
[10/20/24 17:43:57] crowsnest:		
[10/20/24 17:43:57] crowsnest:		[cam 1]
[10/20/24 17:43:57] crowsnest:		mode: ustreamer
[10/20/24 17:43:57] crowsnest:		
[10/20/24 17:43:57] crowsnest:		enable_rtsp: false
[10/20/24 17:43:57] crowsnest:		rtsp_port: 8554
[10/20/24 17:43:57] crowsnest:		port: 8080
[10/20/24 17:43:57] crowsnest:		device: /dev/video0
[10/20/24 17:43:57] crowsnest:		resolution: 640x480
[10/20/24 17:43:57] crowsnest:		max_fps: 15
[10/20/24 17:43:57] crowsnest: INFO: Detect available Devices
[10/20/24 17:43:57] crowsnest: INFO: Found 1 total available Device(s)
[10/20/24 17:43:57] crowsnest: INFO: Found 1 available v4l2 (UVC) camera(s)
[10/20/24 17:43:57] crowsnest: /dev/v4l/by-id/platform-rkisp-vir0-video-index0 -> /dev/video0
[10/20/24 17:43:57] crowsnest: Supported Formats:
[10/20/24 17:43:57] crowsnest:		[0]: 'UYVY' (UYVY 4:2:2)
[10/20/24 17:43:57] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/20/24 17:43:57] crowsnest:		[1]: '422P' (Planar YUV 4:2:2)
[10/20/24 17:43:57] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/20/24 17:43:57] crowsnest:		[2]: 'NV16' (Y/CbCr 4:2:2)
[10/20/24 17:43:57] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/20/24 17:43:57] crowsnest:		[3]: 'NV61' (Y/CrCb 4:2:2)
[10/20/24 17:43:57] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/20/24 17:43:57] crowsnest:		[4]: 'YM16' (Planar YUV 4:2:2 (N-C))
[10/20/24 17:43:57] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/20/24 17:43:57] crowsnest:		[5]: 'NV21' (Y/CrCb 4:2:0)
[10/20/24 17:43:57] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/20/24 17:43:57] crowsnest:		[6]: 'NV12' (Y/CbCr 4:2:0)
[10/20/24 17:43:57] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/20/24 17:43:57] crowsnest:		[7]: 'NM21' (Y/CrCb 4:2:0 (N-C))
[10/20/24 17:43:57] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/20/24 17:43:57] crowsnest:		[8]: 'NM12' (Y/CbCr 4:2:0 (N-C))
[10/20/24 17:43:57] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/20/24 17:43:57] crowsnest:		[9]: 'YU12' (Planar YUV 4:2:0)
[10/20/24 17:43:57] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/20/24 17:43:57] crowsnest:		[10]: 'YM24' (Planar YUV 4:4:4 (N-C))
[10/20/24 17:43:57] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/20/24 17:43:57] crowsnest:		[11]: 'RGGB' (8-bit Bayer RGRG/GBGB)
[10/20/24 17:43:57] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/20/24 17:43:57] crowsnest:		[12]: 'GRBG' (8-bit Bayer GRGR/BGBG)
[10/20/24 17:43:57] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/20/24 17:43:57] crowsnest:		[13]: 'GBRG' (8-bit Bayer GBGB/RGRG)
[10/20/24 17:43:57] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/20/24 17:43:57] crowsnest:		[14]: 'BA81' (8-bit Bayer BGBG/GRGR)
[10/20/24 17:43:57] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/20/24 17:43:57] crowsnest:		[15]: 'RG10' (10-bit Bayer RGRG/GBGB)
[10/20/24 17:43:57] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/20/24 17:43:57] crowsnest:		[16]: 'BA10' (10-bit Bayer GRGR/BGBG)
[10/20/24 17:43:57] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/20/24 17:43:57] crowsnest:		[17]: 'GB10' (10-bit Bayer GBGB/RGRG)
[10/20/24 17:43:57] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/20/24 17:43:57] crowsnest:		[18]: 'BG10' (10-bit Bayer BGBG/GRGR)
[10/20/24 17:43:57] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/20/24 17:43:57] crowsnest:		[19]: 'RG12' (12-bit Bayer RGRG/GBGB)
[10/20/24 17:43:57] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/20/24 17:43:57] crowsnest:		[20]: 'BA12' (12-bit Bayer GRGR/BGBG)
[10/20/24 17:43:57] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[... cut ...]
[10/20/24 17:43:57] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/20/24 17:43:57] crowsnest: Supported Controls:
[10/20/24 17:43:57] crowsnest: Try to start configured Cams / Services...
[10/20/24 17:43:59] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue ...
[10/20/24 17:44:00] crowsnest: V4L2 Control: No parameters set for [cam 1]. Skipped.
[10/20/24 17:44:03] crowsnest: Starting ustreamer with Device /dev/video0 ...
[10/20/24 17:44:04] crowsnest:  ... Done!

BTTJUSTIN, 2024-10-23:
> Hi Justin, thanks for taking the time to look at my issue.
> 
> How would I go about turning off the CSI camera in the script? Which script are we talking about?

Hi， Please turn off the CSI camera in the script
[image]

zhangyumingsh, 2024-10-23:
After the system update, add ENV {ID-PATH}=="? *", KERNEIL=="video * | vbi *", SYMLINK+="v4l/by path/$env" in the 60-persistent-v4l.rules directory of/lib/udev/rules {ID_PATH}-video-index $attr{index}"
[image]

zhangyumingsh, 2024-10-23:
> I also have trouble getting my pi camera v2.1 to work. Altough crownsnest seems to detect it. I dont get a Signal.
>
> [10/20/24 17:43:56] crowsnest: crowsnest - A webcam Service for multiple Cams and [...]

After the system update, add ENV {ID-PATH}=="? *", KERNEIL=="video * | vbi *", SYMLINK+="v4l/by path/$env" in the 60-persistent-v4l.rules directory of/lib/udev/rules {ID_PATH}-video-index $attr{index}"

awoodcroft95, 2024-10-23:
> After the system update, add
> 
> ENV {ID-PATH}=="? *", KERNEL=="video * | vbi *", SYMLINK+="v4l/by-path/$env{ID_PATH}-video-index$attr{index}"
> 
> in the 60-persistent-v4l.rules directory of/lib/udev/rules.d ![csi_camera](https://private-user-images.githubusercontent.com/26036538/379231152-db998ed0-ca5d-4f1b-aa04-4f113956ad5e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk2ODE2MjMsIm5iZiI6MTcyOTY4MTMyMywicGF0aCI6Ii8yNjAzNjUzOC8zNzkyMzExNTItZGI5OThlZDAtY2E1ZC00ZjFiLWFhMDQtNGYxMTM5NTZhZDVlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDIzVDExMDIwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJhZTc2NDMzYzMwMTZmZDJiYTYzODhkOWQ3ODY4NjY3MGU1NzU0Yjc0ZWQxODQ3YTEyNzhjZGMwNjk4Mzc3Y2QmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.r_muKcHNRqcX-tbjxhdEs21ZJK-M7oqriQ-CxmdnT0s)

I already have that in that file: [image]

bigtreetech, 2024-10-23:
> > After the system update, add
> > ENV {ID-PATH}=="? *", KERNEL=="video * | vbi *", SYMLINK+="v4l/by-path/$env{ID_PATH}-video-index$attr{index}"
> > in the 60-persistent-v4l.rules directory of/lib/udev/rules.d ![csi_camera](https://private-user-images.githubusercontent.com/26036538/379231152-db998ed0-ca5d-4f1b-aa04-4f113956ad5e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk2ODE2MjMsIm5iZiI6MTcyOTY4MTMyMywicGF0aCI6Ii8yNjAzNjUzOC8zNzkyMzExNTItZGI5OThlZDAtY2E1ZC00ZjFiLWFhMDQtNGYxMTM5NTZhZDVlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDIzVDExMDIwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTJhZTc2NDMzYzMwMTZmZDJiYTYzODhkOWQ3ODY4NjY3MGU1NzU0Yjc0ZWQxODQ3YTEyNzhjZGMwNjk4Mzc3Y2QmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.r_muKcHNRqcX-tbjxhdEs21ZJK-M7oqriQ-CxmdnT0s)
> 
> I already have that in that file: ![{00C5B40F-4E91-4CC9-ADDF-E124D5329506}](https://private-user-images.githubusercontent.com/15670952/379237044-47c3a41c-c3a1-4767-b169-3947963bb8cb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk2ODIzOTcsIm5iZiI6MTcyOTY4MjA5NywicGF0aCI6Ii8xNTY3MDk1Mi8zNzkyMzcwNDQtNDdjM2E0MWMtYzNhMS00NzY3LWIxNjktMzk0Nzk2M2JiOGNiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDIzVDExMTQ1N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc5MDU3MTMxMDQ2YTRmNjVhY2ZjMjc2MGY3MzA3MjNkOGQ4OGQzZTIwMzQzOWU2MmMwZTI4Y2ZiZDdiN2VmYTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ep1wq-t5Uf0MjKJKxDbxqrME6V-YDirnQyy4_VxVLZ0)

Sorry, there was a mistake. I am updating the comments. Please check in 1 minute.

awoodcroft95, 2024-10-23:
> > Hi Justin, thanks for taking the time to look at my issue.
> > How would I go about turning off the CSI camera in the script? Which script are we talking about?
> 
> Hi， Please turn off the CSI camera in the script ![71c7adaa32e1950098e27b877e53246e](https://private-user-images.githubusercontent.com/169997478/379194054-35fe32b2-9211-4f8e-9790-3c7da015e03c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk2ODE2MjMsIm5iZiI6MTcyOTY4MTMyMywicGF0aCI6Ii8xNjk5OTc0NzgvMzc5MTk0MDU0LTM1ZmUzMmIyLTkyMTEtNGY4ZS05NzkwLTNjN2RhMDE1ZTAzYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAyM1QxMTAyMDNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02OTc5YzRhYWFmMjZlNDRmNGQyYTQ5NjdhYjQ3MDE5ZmM2M2Y5ZWMyZDc2NmM0MWMzNDk4NWU5NDdkZTI1Njg0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Z6Ex90bt0dbNT6askKn-H2qY6mNfk00uv2KBENELcKc)

Hi, I do not have crowsnest installed as I saw in another issue thread that someone else had an issue with crowsnest not allowing the camera to be available. 
[image]

wigglebrows, 2024-10-24:
I followed what Justin said and uncommented "custom_flags: --format=UYVY". The camera started streaming just fine. I didn't make any other changes (apart from adding the device path from the log) from what comes straight off the BTT debian kernel with klipper preinstalled.

Hilpas, 2024-10-25:
No luck for me so far. Even tried a fresh Install onto a Micro SD Card.

bigtreetech, 2024-10-25:
> No luck for me so far. Even tried a fresh Install onto a Micro SD Card.

What camera are you running? Seems like crowsnest is detecting 800x600 res and you have it set to 640x480.

Hilpas, 2024-10-25:
> > No luck for me so far. Even tried a fresh Install onto a Micro SD Card.
> 
> What camera are you running? Seems like crowsnest is detecting 800x600 res and you have it set to 640x480.

I should be a genuine v2.1 pi camera. The detected available resolution seems to low then though. But it should at least work, because it did before on my pi4. Tried every Variation of the config 800x600 too.

bigtreetech, 2024-10-25:
Try the commands listed in the first comment here and then post your results including the generated bin file.

Hilpas, 2024-10-26:
> Try the commands listed in the first comment here and then post your results including the generated bin file.

It seems to only detect a 0x10 when i unplug and plug back in the camera. 
[image]
Otherwise its only 0x64
[image]
Taking a .bin doesnt work either. 
[image]

[image]

[image]

[image]

`[10/26/24 07:38:27] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services.
[10/26/24 07:38:27] crowsnest: Version: v4.1.9-1-gd75a3ae
[10/26/24 07:38:27] crowsnest: Prepare Startup ...
[10/26/24 07:38:27] crowsnest: INFO: Host information:
[10/26/24 07:38:27] crowsnest: Host Info: Distribution: BTT-CB2_1.0.5.1_Armbian 23.05.0-trunk Bullseye
[10/26/24 07:38:27] crowsnest: Host Info: Kernel: Linux 4.19.232-bigtree-cb2 aarch64
[10/26/24 07:38:27] crowsnest: Host Info: Available CPU Cores: 4
[10/26/24 07:38:27] crowsnest: Host Info: Available Memory: 2029724 kB
[10/26/24 07:38:27] crowsnest: Host Info: Diskspace (avail. / total): 10G / 15G
[10/26/24 07:38:27] crowsnest: INFO: Checking Dependencies
[10/26/24 07:38:27] crowsnest: Dependency: 'crudini' found in /usr/bin/crudini.
[10/26/24 07:38:27] crowsnest: Dependency: 'find' found in /usr/bin/find.
[10/26/24 07:38:27] crowsnest: Dependency: 'xargs' found in /usr/bin/xargs.
[10/26/24 07:38:27] crowsnest: Dependency: 'ustreamer' found in bin/ustreamer/src/ustreamer.bin.
[10/26/24 07:38:27] crowsnest: Version Control: ustreamer is up to date. (v6.10)
[10/26/24 07:38:27] crowsnest: INFO: Print Configfile: '/home/biqu/printer_data/config/crowsnest.conf'
[10/26/24 07:38:27] crowsnest:		[crowsnest]
[10/26/24 07:38:27] crowsnest:		log_path: /home/biqu/printer_data/logs/crowsnest.log
[10/26/24 07:38:27] crowsnest:		log_level: verbose
[10/26/24 07:38:27] crowsnest:		delete_log: false
[10/26/24 07:38:27] crowsnest:		no_proxy: false
[10/26/24 07:38:27] crowsnest:		
[10/26/24 07:38:27] crowsnest:		[cam 1]
[10/26/24 07:38:27] crowsnest:		mode: ustreamer
[10/26/24 07:38:27] crowsnest:		
[10/26/24 07:38:27] crowsnest:		enable_rtsp: false
[10/26/24 07:38:27] crowsnest:		rtsp_port: 8554
[10/26/24 07:38:27] crowsnest:		port: 8080
[10/26/24 07:38:27] crowsnest:		device: /dev/video0
[10/26/24 07:38:27] crowsnest:		resolution: 640x480
[10/26/24 07:38:27] crowsnest:		max_fps: 15
[10/26/24 07:38:27] crowsnest: INFO: Detect available Devices
[10/26/24 07:38:27] crowsnest: INFO: Found 1 total available Device(s)
[10/26/24 07:38:27] crowsnest: INFO: Found 1 available v4l2 (UVC) camera(s)
[10/26/24 07:38:27] crowsnest: /dev/v4l/by-id/platform-rkisp-vir0-video-index0 -> /dev/video0
[10/26/24 07:38:27] crowsnest: Supported Formats:
[10/26/24 07:38:27] crowsnest:		[0]: 'UYVY' (UYVY 4:2:2)
[10/26/24 07:38:27] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/26/24 07:38:27] crowsnest:		[1]: '422P' (Planar YUV 4:2:2)
[10/26/24 07:38:27] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/26/24 07:38:27] crowsnest:		[2]: 'NV16' (Y/CbCr 4:2:2)
[10/26/24 07:38:27] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/26/24 07:38:27] crowsnest:		[3]: 'NV61' (Y/CrCb 4:2:2)
[10/26/24 07:38:27] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/26/24 07:38:27] crowsnest:		[4]: 'YM16' (Planar YUV 4:2:2 (N-C))
[10/26/24 07:38:27] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/26/24 07:38:27] crowsnest:		[5]: 'NV21' (Y/CrCb 4:2:0)
[10/26/24 07:38:27] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/26/24 07:38:27] crowsnest:		[6]: 'NV12' (Y/CbCr 4:2:0)
[10/26/24 07:38:27] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/26/24 07:38:27] crowsnest:		[7]: 'NM21' (Y/CrCb 4:2:0 (N-C))
[10/26/24 07:38:27] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/26/24 07:38:27] crowsnest:		[8]: 'NM12' (Y/CbCr 4:2:0 (N-C))
[10/26/24 07:38:27] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/26/24 07:38:27] crowsnest:		[9]: 'YU12' (Planar YUV 4:2:0)
[... 8 lines cut ...]
[10/26/24 07:38:27] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/26/24 07:38:27] crowsnest:		[14]: 'BA81' (8-bit Bayer BGBG/GRGR)
[10/26/24 07:38:27] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/26/24 07:38:27] crowsnest:		[15]: 'RG10' (10-bit Bayer RGRG/GBGB)
[10/26/24 07:38:27] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/26/24 07:38:27] crowsnest:		[16]: 'BA10' (10-bit Bayer GRGR/BGBG)
[10/26/24 07:38:27] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/26/24 07:38:27] crowsnest:		[17]: 'GB10' (10-bit Bayer GBGB/RGRG)
[10/26/24 07:38:27] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/26/24 07:38:27] crowsnest:		[18]: 'BG10' (10-bit Bayer BGBG/GRGR)
[10/26/24 07:38:27] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/26/24 07:38:27] crowsnest:		[19]: 'RG12' (12-bit Bayer RGRG/GBGB)
[10/26/24 07:38:27] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/26/24 07:38:27] crowsnest:		[20]: 'BA12' (12-bit Bayer GRGR/BGBG)
[10/26/24 07:38:27] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/26/24 07:38:27] crowsnest:		[21]: 'GB12' (12-bit Bayer GBGB/RGRG)
[10/26/24 07:38:27] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/26/24 07:38:27] crowsnest:		[22]: 'BG12' (12-bit Bayer BGBG/GRGR)
[10/26/24 07:38:27] crowsnest:		Size: Stepwise 32x16 - 800x600 with step 8/8
[10/26/24 07:38:27] crowsnest: Supported Controls:
[10/26/24 07:38:27] crowsnest: Try to start configured Cams / Services...
[10/26/24 07:38:30] crowsnest: INFO: Configuration of Section [cam 1] looks good. Continue ...
[10/26/24 07:38:31] crowsnest: V4L2 Control: No parameters set for [cam 1]. Skipped.
[10/26/24 07:38:33] crowsnest: Starting ustreamer with Device /dev/video0 ...
[10/26/24 07:38:34] crowsnest:  ... Done!`

awoodcroft95, 2024-10-28:
> /lib/udev/rules.d

I have updated the file and it now looks like this:
[image]

Running `v4l2-ctl -d /dev/video0 --set-fmt-video=width=640,height=480,pixelformat=NV12 --stream-mmap=3 --stream-skip=3 --stream-to=/usr/nv12.bin --stream-count=1 --stream-poll` gives the follwing:
[image]

`i2cdetect -y 2 no longer shows the camera at address 0x10
[image]

awoodcroft95, 2024-10-28:
@bigtreetech 
after updating the file to look like this:
[image]

this is the result of the testing commands:
[image]

awoodcroft95, 2024-11-09:
@bigtreetech any update on this?

Do you know if linux compatible webcams are working okay with klipper on the CB2? I was thinking of just picking up UGREEN's 1080p 30fps webcam, or a Logitech C270.

Have either of those been tested?

bigtreetech, 2024-11-11:
@zhangyumingsh please assist @awoodcroft95

zhangyumingsh, 2024-11-11:
> > Try the commands listed in the first comment here and then post your results including the generated bin file.
> 
> It seems to only detect a 0x10 when i unplug and plug back in the camera. ![i2cdetect](https://private-user-images.githubusercontent.com/44614165/380373718-4981f02d-ed0e-4258-9922-7b764e5b9e1c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEzMTA1MDAsIm5iZiI6MTczMTMxMDIwMCwicGF0aCI6Ii80NDYxNDE2NS8zODAzNzM3MTgtNDk4MWYwMmQtZWQwZS00MjU4LTk5MjItN2I3NjRlNWI5ZTFjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTExVDA3MzAwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ0MDc4YmM0YjVkYmVjN2I1MDIzMzI2ZjM4ODVhNDdkMTE0ZGZjZGJhNWVkMzA1ZmYwOTQzMDkzNmNkNTY0N2EmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.8KrIdryCWMKspFSCzbQYyAE1BNSD2THHeJMz2zSusQI) Otherwise its only 0x64 ![i2cdetectnosensor](https://private-user-images.githubusercontent.com/44614165/380373722-05c7a8b2-7ab5-493e-a69f-2194199d2d67.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEzMTA1MDAsIm5iZiI6MTczMTMxMDIwMCwicGF0aCI6Ii80NDYxNDE2NS8zODAzNzM3MjItMDVjN2E4YjItN2FiNS00OTNlLWE2OWYtMjE5NDE5OWQyZDY3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTExVDA3MzAwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQxNTQwNmNiOGM3YjEzYTI2YmFjZDI2OTViZjcyOWQwZjg3NTVjNzhkOGFlYWVhZGJiYzBmNzc5ZDk2ZDAwODAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.bQHBgMXunUwNTyN_9YgmtYOjMPA7y1yUy0PMzZ3HgZc) Taking a .bin doesnt work either. ![v4l2photo](https://private-user-images.githubusercontent.com/44614165/380373785-f64e0037-3343-414e-bf07-ab957a592483.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEzMTA1MDAsIm5iZiI6MTczMTMxMDIwMCwicGF0aCI6Ii80NDYxNDE2NS8zODAzNzM3ODUtZjY0ZTAwMzctMzM0My00MTRlLWJmMDctYWI5NTdhNTkyNDgzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTExVDA3MzAwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJiZDg5OGM1Y2QyYWNlZDUxOGE4NWQxYjQ5Njk0MzkzYmE2NGM1M2U5ZGNmZDg4Njk4YWU1ZjRlODk1MWViODEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.qEJwgn_61dOhutT54Bl83WoltBRg7P3ZJqvJH_vvt-I)
> 
> ![v4l2devices](https://private-user-images.githubusercontent.com/44614165/380373847-01931575-f7df-4fee-aab4-aae80b0be64a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEzMTA1MDAsIm5iZiI6MTczMTMxMDIwMCwicGF0aCI6Ii80NDYxNDE2NS8zODAzNzM4NDctMDE5MzE1NzUtZjdkZi00ZmVlLWFhYjQtYWFlODBiMGJlNjRhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTExVDA3MzAwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY2YWQyMTFiNzUyMWIzYzczZmYxZDllY2ExMWRmMjExYTdiM2Q0NjdhNGI4NjFhYTVkMmFiMDJjNjZmZDE5MWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.H55VMEaLI6o0sBB6G-2oZCF-F0X7J5h_QBQ1AHcW7MU)
> 
> ![v4l2byid](https://private-user-images.githubusercontent.com/44614165/380373853-25b9651f-1644-4d82-9ca7-db4800834db3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEzMTA1MDAsIm5iZiI6MTczMTMxMDIwMCwicGF0aCI6Ii80NDYxNDE2NS8zODAzNzM4NTMtMjViOTY1MWYtMTY0NC00ZDgyLTljYTctZGI0ODAwODM0ZGIzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTExVDA3MzAwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc4Zjk5NzYzMzYyYjc5ZDRjNmU5MTM2Yzk2Zjg1ZjUxMDIxMzJmZmM4NDkzNDY0NjQ1MDYxNzQ5NjMwNjJmOTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.FF7tY0t-Zg2LlBBKNHvTHiufmkU1EnZUARqmQj5ujDA)
> 
> ![v4l2bypath](https://private-user-images.githubusercontent.com/44614165/380373855-71ac4cef-bd39-41e2-87a3-a61304f5151a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEzMTA1MDAsIm5iZiI6MTczMTMxMDIwMCwicGF0aCI6Ii80NDYxNDE2NS8zODAzNzM4NTUtNzFhYzRjZWYtYmQzOS00MWUyLTg3YTMtYTYxMzA0ZjUxNTFhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTExVDA3MzAwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTEyNzdjYzVlMTExOTE3MGQ3MDczMmJjNmUwNzU2ZDRkNDBmNTdlZTllYjBmZmQxMjkyODQ0NWQ0ODc1NTQ0MGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.KdjCABw0HWAMmaEcs6saxMEL2uvwLG0qANSilfnb7GM)
> 
> `[10/26/24 07:38:27] crowsnest: crowsnest - A webcam Service for multiple Cams and Stream Services. [10/26/24 07:38:27] crowsnest: Version: v4.1.9-1-gd75a3ae [10/26/24 07:38:27] crowsnest: Prepare Startup ... [10/26/24 07:38:27] crowsnest: INFO: Host information: [10/26/24 07:38:27] crowsnest: Host Info: Distribution: BTT-CB2_1.0.5.1_Armbian 23.05.0-trunk Bullseye [10/26/24 07:38:27] crowsnest: Host Info: Kernel: Linux 4.19.232-bigtree-cb2 aarch64 [10/26/2 [... cut ...] Starting ustreamer with Device /dev/video0 ... [10/26/24 07:38:34] crowsnest: ... Done!`

76/10000
 As long as the hardware of these two camera models is normal, the ov5647 and imx219 of btt-cb2/pi2 do not need to be configured. Has the system made any changes since the driver was not loaded? Translation: The ov5647 and imx219 of btt-cb2/pi2 do not require configuration, as long as the hardware of these two camera models is normal. Has the system made any changes since the driver was not loaded?

zhangyumingsh, 2024-11-11:
> @bigtreetech any update on this?
> 
> Do you know if linux compatible webcams are working okay with klipper on the CB2? I was thinking of just picking up UGREEN's 1080p 30fps webcam, or a Logitech C270.
> 
> Have either of those been tested?

None of them have been tested. What specific model of USB network camera is UGREEN? As long as there is a corresponding driver in the image, it can work normally in the Klipper image of cb2. Can you send me the models of USB cameras that need to be supported? Thank you

Hilpas, 2024-11-11:
> 76/10000 As long as the hardware of these two camera models is normal, the ov5647 and imx219 of btt-cb2/pi2 do not need to be configured. Has the system made any changes since the driver was not [...]

I also tried a freshly installed Image with no luck. Thats why i now bought a OV5648 USB which is working.

smbenson0712, 2025-02-20:
Has anyone got this issue resolved?  I have tried the arducam v2.3 (imx219) and the picam v2 (imx219) on both the CB2 and BTT Pi2 with no luck.  I have tried with Deb10, 11, and the beta 12 versions both minimal and full klipper.  I get the same outputs with the commands as above.  In Crowsnest log it shows that the device is found, and the stream starts, but in mainsail it will not show any output.  In Crowsnest log, device it only shows a serial device, no video device.  When I use that serial device id in Crowsnest config, it attaches to that device but shows no image.  Any help would be appreciated.

bderancourt, 2025-02-21:
I have an imx219 cam from AliExpress and it's working with the fix from @zhangyumingsh :
https://github.com/bigtreetech/CB2/issues/37#issuecomment-2431718679
But by default I have a near black screen as there is no automatic gain due to the lack of ISP currently. So working yes, but nearly unusable now.

smbenson0712, 2025-02-21:
I checked the file mentioned and it has both the "by-id" and "by-path" lines.  The only difference I see the listed fix does not include an * after the ?, an * after video, and an * after vbi, so the lines are different.  I'm not a linux expert so I don't know if that change will make any difference.  First line is from your referenced comment and the second is from my pi.
ENV{ID_PATH}=="?", KERNEL=="video|vbi", SYMLINK+="v4l/by-id/$env{ID_PATH}-video-index$attr{index}"
ENV{ID_PATH}=="?*", KERNEL=="video*|vbi*", SYMLINK+="v4l/by-path/$env{ID_PATH}-video-index$attr{index}"

Karl-Barx, 2025-03-18:
this is ridiculous for something that claims to be plug and play. nothing on here has worked and this isnt a new issue

Karl-Barx, 2025-03-18:
> Has anyone got this issue resolved? I have tried the arducam v2.3 (imx219) and the picam v2 (imx219) on both the CB2 and BTT Pi2 with no luck. I have tried with Deb10, 11, and the beta 12 versions [...]

did you ever figure this out?

zhangyumingsh, 2025-03-18:
> this is ridiculous for something that claims to be plug and play. nothing on here has worked and this isnt a new issue

are u serious？and try it？

Karl-Barx, 2025-03-18:
> [@bigtreetech](https://github.com/bigtreetech) any update on this?
> 
> Do you know if linux compatible webcams are working okay with klipper on the CB2? I was thinking of just picking up UGREEN's 1080p 30fps webcam, or a Logitech C270.
> 
> Have either of those been tested?

those are working fine in my experience, i have a c270 on my voron it works fine, the second cam im trying to use is a rsp cam and once again BTT just pushed out products with no support or real world testing AGAIN.

zhangyumingsh, 2025-03-18:
The Image supports the UVC protocol, and our engineers have tested it. Just follow the document settings. The ISP processing of the camera with CSI interface has been resolved in the upcoming image release.

> those are working fine in my experience, i have a c270 on my voron it works fine, the second cam im trying to use is a rsp cam and once again BTT just pushed out products with no support or real [...]

smbenson0712, 2025-03-18:
I have not got it working - kind of gave up and ordered a CM4 for my manta - known to work as it is a pi.  @zhangyumingsh, so what you are saying is that there is an issue with ISP processing for csi cameras?  Is there a procedure to add configs/setting to make it work without reimaging the piv2?

Karl-Barx, 2025-03-19:
> The Image supports the UVC protocol, and our engineers have tested it. Just follow the document settings. The ISP processing of the camera with CSI interface has been resolved in the upcoming image [...]

> The Image supports the UVC protocol, and our engineers have tested it. Just follow the document settings. The ISP processing of the camera with CSI interface has been resolved in the upcoming image [...]

When is the upcoming release? Will the beta work?

bigtreetech, 2025-03-24:
> When is the upcoming release? Will the beta work?

Released as the second beta. ISP should be fully functional. Give it a try.

smbenson0712, 2025-03-25:
Tried the new beta 3 and can confirm that both the pi cam v2 and the arducam v2 both work with the new images, both minimal and full version.  Thanks for fixing this issue.

bigtreetech, 2025-03-25:
> Tried the new beta 3 and can confirm that both the pi cam v2 and the arducam v2 both work with the new images, both minimal and full version. Thanks for fixing this issue.

Good to hear!