bigtreetech/CB2 #40: Balena Etcher and CB2 Klipper Firmware failing

txmedictom, 2024-11-06:
I have tried multiple SD cards and downloaded the CB2 Klipper image several times with no luck. I am having to use the minimal image then install Klipper. Is there a problem with the file?

txmedictom, 2024-11-06:
It fails verification and will not write a usable image to the SD card. I have to use the minimal the KIAUH to get everything else current.

Pneumanifest, 2024-11-06:
The failure of verification does not always mean that there was an actual failure. It's quite often based off of hash numbers and when you split up partitions hash numbers can get mismatches.  Every image that I have had a failure to verify has always worked as expected.   Did you try to use the what was burned even though the error was given?

Pneumanifest, 2024-11-06:
That's odd that nothing is written. It writes before it verifies. Have you unzipped the image before writing the image?

txmedictom, 2024-11-06:
No, I am not unzipping anything. I just use it to write the image. The same way that I wrote the image for the minimal that works.

WEBOWNIA, 2024-11-08:
Same here. I can't flash any of the available images. I'm experiencing the same issue as my colleague.

zhangyumingsh, 2024-11-08:
> Same here. I can't flash any of the available images. I'm experiencing the same issue as my colleague.

I downloaded it from the internet and tested it.

> I have tried multiple SD cards and downloaded the CB2 Klipper image several times with no luck. I am having to use the minimal image then install Klipper. Is there a problem with the file?

What version is the minimum image?I will download it from the git and test it.

zhangyumingsh, 2024-11-08:
> I have tried multiple SD cards and downloaded the CB2 Klipper image several times with no luck. I am having to use the minimal image then install Klipper. Is there a problem with the file?

The Klipper image we released was tested by a testing engineer before being released. Generally, the system runs normally without any problems.

txmedictom, 2024-11-08:
Got it working.

I used Raspberry Pi Imager and got it working. Balena evidently doesn't
like the BTT image but RPi Imager did it just fine.

Pneumanifest, 2024-11-08:
> No, I am not unzipping anything. I just use it to write the image. The same way that I wrote the image for the minimal that works.

Sometimes you must unzip the compressed image file.

Pneumanifest, 2024-11-08:
> Got it working. I used Raspberry Pi Imager and got it working. Balena evidently doesn't like the BTT image but RPi Imager did it just fine.
Yeah raspberry pi imager fully unzips if it is compressed before it attempts to flash. As a rule of thumb ita good practice to unzip a compressed.img file before attempting to flash.

zhangyumingsh, 2024-11-08:
> Same here. I can't flash any of the available images. I'm experiencing the same issue as my colleague.

Please refer to the user manual for instructions

zhangyumingsh, 2024-11-08:
> Got it working. I used Raspberry Pi Imager and got it working. Balena evidently doesn't like the BTT image but RPi Imager did it just fine.
> […](#)

Thank you for your feedback. We used BalenaEtcher for testing, I'll test it again

Pneumanifest, 2024-11-08:
BalenaEtcher is my main flashing tool too. It has worked every time. Although it does occasionally throw a failed to verify error at the end. Even with the verify error it has always worked for me. I have just started ignoring that error.

Pneumanifest, 2024-11-08:
Further digging shows the etcher team is somewhat aware of the issue and have an open ticket on it. It mostly looks like what I suspect, it a match hash error. Not an actual flashing error.

WEBOWNIA, 2024-11-08:
[image]

Pneumanifest, 2024-11-08:
If it is the .img.xz unzip the file first.

WEBOWNIA, 2024-11-08:
Yes, it’s unpacked before flashing.

Pneumanifest, 2024-11-08:
> Yes, it’s unpacked before flashing.

you might need to download that again. That looks like a file error.

WEBOWNIA, 2024-11-08:
I tried downloading all three versions, and after unpacking, flashing results in the same error. Ultimately, it doesn’t matter whether the image is unpacked or not.

However, Raspberry Pi Imager doesn’t return an error and creates the disk, but then I face another issue—the bootloader doesn’t start on the Manta E3EZ V1.0. I’ve already reported this separately on https://github.com/bigtreetech/Manta-E3EZ/issues/45

txmedictom, 2024-11-08:
This is the issue I had. Balena errors and would not write anything. RPi Imager goes right through and doesn't care. I would strongly recommend using the Ed's 3d Printer Tutorials method of using RPi Imager. My E3EZ + CB2 are not working. Now I am trying to get a solution to getting the EBB36 since there seems to be nothing out there to clarify the differences in pins between a Manta 5 and an E3EZ. They used an almost illegible yellow on the PDF for the board. BTT seems to have almost no informati9on about the CAN specific details for the E3EZ. I have been referred to Esoterical's we site by someone on Discord. I plan to work on that today. You would think that a board that works so well in Tridents, Switchwires, etc would have details about full CAN setup.

bigtreetech, 2024-11-08:
> This is the issue I had. Balena errors and would not write anything. RPi Imager goes right through and doesn't care. I would strongly recommend using the Ed's 3d Printer Tutorials method of using [...]

https://github.com/bigtreetech/Manta-E3EZ/blob/master/Hardware/BIGTREETECH%20Manta%20E3EZ%20V1.0-PIN.png

The can pins are identified here. There is not much more to it. The rest is all described in the standard klipper documentation on setting up a USB to CAN bridge.

WEBOWNIA, 2024-11-08:
> This is the issue I had. Balena errors and would not write anything. RPi Imager goes right through and doesn't care. I would strongly recommend using the Ed's 3d Printer Tutorials method of using RPi Imager. My E3EZ + CB2 are not working.

:+1: We need to join forces… But in the videos, they show it working, although I find it curious that exactly at this moment https://www.youtube.com/watch?v=XfZppjHU0ak&t=531s, the guy removes another card from the frame, as if he already had one set up next to it.

txmedictom, 2024-11-08:
> > This is the issue I had. Balena errors and would not write anything. RPi Imager goes right through and doesn't care. I would strongly recommend using the Ed's 3d Printer Tutorials method of using RPi Imager. My E3EZ + CB2 are not working.
> 
> 👍 We need to join forces… But in the videos, they show it working, although I find it curious that exactly at this moment https://www.youtube.com/watch?v=XfZppjHU0ak&t=531s, the guy removes another card from the frame, as if he already had one set up next to it.

He uses RPi Imager and it does the CB2 image just fine. My issue is setting up the EBB now. I can geet the CB2 Klipper image and all of that done. BTT just sent me the firmware config image for what pins to select, etc on Katapult. I am working on that now. The problem I have is the whole order of when you connect to PC, when you connect to USB on the Manta, and when you connect to the CAN pins. All of the documentation and process that I have seen is confusing. The best one I have seen so far is from https://www.youtube.com/watch?v=EA-oBfenxAE&t=1992s and it is still a little confusing but now I have the right pins to assign in Katapult and Klipper for the Manta E3EZ so I will give it another shot later today.

WEBOWNIA, 2024-11-09:
@bigtreetech Can you put checksums on to other versions?

bigtreetech, 2024-11-11:
> @bigtreetech Can you put checksums on to other versions (https://github.com/bigtreetech/CB2/releases)?

Do you mean a manually pasted hash? The checksum that balena uses is calculated by the tool itsef.

WEBOWNIA, 2024-11-14:
OK, I understand. Anyway, after a few attempts, I managed to flash it successfully, so everything worked out for me in the end.

serbiansky, 2024-12-09:
Is it required to use SD if we flash directly to eMMC? As I am not able to have any of the option running, tried both images directly to eMMC and over SD card but it is not booting from any of them. Maybe my laptop USB is not powerful enough.

looxonline, 2024-12-09:
> Is it required to use SD if we flash directly to eMMC? As I am not able to have any of the option running, tried both images directly to eMMC and over SD card but it is not booting from any of them. Maybe my laptop USB is not powerful enough.

It may be your laptop power. You could try to power from 24V and then simply plug it in via USB and flash to eMMC. Otherwise you can use the SD method with it powered via 24V.

serbiansky, 2024-12-09:
Thank you, I am successfully flashing the eMMC but when I reconnect USB nothing is happening on LAN. In the instructions we need to return only USBOTG to 0 what about other one for pi?
I need to cross check my wiring before I switch to 24v do not want to burn something.

looxonline, 2024-12-09:
> Thank you, I am successfully flashing the eMMC but when I reconnect USB nothing is happening on LAN. In the instructions we need to return only USBOTG to 0 what about other one for pi? I need to cross check my wiring before I switch to 24v do not want to burn something.

Make sure there is no SD card inserted. Keep the eMMC switch set to the setting that you used to program the eMMC. You will need to plug the ethernet into a router that has DHCP and then search for the IP address on the network using a tool like angry IP scanner.

serbiansky, 2024-12-09:
No SD card but I moved back otg switch, will try without that, was just following the instructions. IP search is not an issue I am monitoring my router for lan and wifi.

serbiansky, 2024-12-10:
No SD card but I moved back otg switch, will try without that, was just following the instructions. IP search is not an issue I am monitoring my router for lan and wifi. 
No luck, I tried all opinions but boot from eMMC is not working.