bigtreetech/CB2 #42: 32 Bit OS Image

Kiiiddd, 2024-11-12:
Is there any plans to offer a 32 Bit image the CB2 or any of your SBC's? 32 bit OS works better with canbus

zhangyumingsh, 2024-11-26:
> Is there any plans to offer a 32 Bit image the CB2 or any of your SBC's? 32 bit OS works better with canbus

Try setting up a 32-bit runtime environment：
sudo dpkg --add-architecture i386
sudo apt-get update
sudo apt install libc6:i386 libstdc++6:i386
sudo apt-get update
sudo apt install libncurses5-dev lib32z1

looxonline, 2024-11-26:
Not at the moment. Why would you want 32bit when there is 64bit available? There are major updates to the 64bit images coming in the next month.

viniavila, 2024-11-28:
@looxonline he said already. Because canbus support is buggy in 64bit armbian. And, in systems with less than 4GB of RAM, like the BTT Pi2, using 64bit memory addressing is overkill. 32bit is far enough (and saves memory).