bigtreetech/CB2 #43: RPI camera module 3 not working

ploucandco, 2024-11-17:
Hello, I cannot get the RPI camera module 3 working with a PI2 V1.0.1 and klipper kernel V1.0.5. I got an USB webcam working without issues. Here are the logs. What should I do?
[crowsnest (17).log](https://github.com/user-attachments/files/17790590/crowsnest.17.log)

ploucandco, 2024-11-22:
Any news on this one?

BTTJUSTIN, 2024-11-29:
Hi, thanks for your information, the current system does not support imx708

Hooba97, 2025-01-14:
Hello,
Any luck getting it to work?
I wonder if BTT is planning to add support to pi cam 3

looxonline, 2025-01-15:
> Hello, Any luck getting it to work? I wonder if BTT is planning to add support to pi cam 3

We are working on a new image running on a later linux kernel. No guarantees about the IMX with this new kernel but I will speak to the firmware engineers about it.

Hooba97, 2025-04-13:
New image released 

Can anyone confirm if Pi cam 3 driver is included / test if it works

ploucandco, 2025-04-13:
RPI camera module 3 (IMX708) is not working with the Beta2. 
The chip is not even correclty recognised. I get a resolution of 800x600 what is clearly wrong.

Anyway, the ISP with the IMX719 is working well with this Beta2 release.

I just saw that the V3.0.1 is out. I will install that but see nothing mentioned about the camera in the release notes. Also, not only you need the right kernel driver for the IMX708, you also need a totally different SW stack (like libcamera and camera-streamer on RPI).

Hooba97, 2025-04-17:
i just installed the new image, and still no support for RPI camera module 3 (IMX708) out of the box