bigtreetech/CB2 #46: Can't access eMMC to install OS

viniavila, 2024-11-28:
I booted from SD Card and tried to run `sudo nand-sata-install` to install ARMBIAN into eMMC but there is no option "Boot from eMMC - system on eMMC" (figure below)

[image]

When I run `dmesg | grep mmc1` I get this output:

[image]

It appears the OS either can't recognize the eMMC, or there is a hardware issue, as I tried to flash eMMC from PC, by USB-OTG, and it fails (flashing bootloader works). I checked the EN-EMMC switch position and there is no mistake.

Keurus, 2024-11-29:
I have the same problem when booting from the SD card. If I use RKDevtool, this is the picture:
[image]

Escrich, 2024-12-21:
[image]
[image]

In my case, the much I get are the first error, and then the second screen, never in English, and never recognizing anything more.

Drivers are correctly installed:

[image]

closed 2025-01-25 by BTTJUSTIN