bigtreetech/CB2 #47: Please Update Kernel

crazybanane, 2024-12-01:
Since latest Update Klipper.bin can't get flashed on mcu. Kernel is too old.
Please Update Image with Newer Kernel.

zhangyumingsh, 2024-12-06:
> Since latest Update Klipper.bin can't get flashed on mcu. Kernel is too old. Please Update Image with Newer Kernel.

We will update the kernel in a few weeks

tpruvot, 2024-12-27:
Had no issue to update Klipper here : 
[image]

mast3rmindx, 2025-01-25:
@tpruvot Do you have HDMI working on the Armbian build?

bderancourt, 2025-01-25:
> > Since latest Update Klipper.bin can't get flashed on mcu. Kernel is too old. Please Update Image with Newer Kernel.
> 
> We will update the kernel in a few weeks

We are few weeks later. Do you have new ETA for the kernel update ?

crazybanane, 2025-01-25:
> [@tpruvot](https://github.com/tpruvot) Do you have HDMI working on the Armbian build?

Hey, i dont use HDMI it needs some time to get a adapter cable to test it.

tpruvot, 2025-01-25:
> [@tpruvot](https://github.com/tpruvot) Do you have HDMI working on the Armbian build?

yes on the first/left plug, but not on all monitors and doesn't seem stable... Maybe 2GB of memory is not enough, unsure. If you do tests, provide a real power input to the manta board, not only the USB C. Also tested the DSI TFT touch screen but i dont really like it, the angle of view is very limited.

mast3rmindx, 2025-01-25:
@tpruvot Did you have to enable HDMI via ArmbianENV.txt? I’m facing issue where xstart can’t find the display

tpruvot, 2025-01-25:
i enabled hdmi and dsi yes to try it, but ive no real use of a hdmi display without a touchscreen.

mast3rmindx, 2025-01-25:
Could you tell me what command you use to enable HDMI? So the touchscreen isn’t being recognized?

tpruvot, 2025-01-25:
no, i meant hdmi screens dont have any touchscreen, the one on the picture is the DSI one which is made for printers and touch. I just enabled the 2 overlays in the env text file

zhangyumingsh, 2025-01-26:
> no, i meant hdmi screens dont have any touchscreen, the one on the picture is the DSI one which is made for printers and touch. I just enabled the 2 overlays in the env text file

The work on the kernel is nearing completion, and the kernel still needs to be organized. The image can only be released after the Chinese Lunar New Year.

zhangyumingsh, 2025-01-26:
The work on kernel6.1 is nearing completion, and the kernel still needs to be organized. The image can only be released after the Chinese Lunar New Year.

mast3rmindx, 2025-01-26:
@zhangyumingsh You also said kernel update was coming in a few weeks in Oct 2024.

zhangyumingsh, 2025-01-26:
> [@zhangyumingsh](https://github.com/zhangyumingsh) You also said kernel update was coming in a few weeks in Oct 2024.

Yes, the 6.1 kernel was completed in stages by mid July 2024. But adding some other features, such as ISP improving pixel quality, took some time. So it has been postponed until now and has not been released yet

closed 2025-02-27 by BTTJUSTIN