bigtreetech/CB2 #49: Power_off GPIO

Shaun87, 2024-12-04:
Good day.

I want to use GPIO0_C3 to switch a relay when the CB2 has shut down. On a Raspberry I would have used:
dtoverlay=gpio-poweroff,,gpio_pin=26.
Can someone help me or point me in the right direction?

Kind regards

Pneumanifest, 2024-12-09:
I may be wrong but I believe that the GPIO are set high as default. Since you want to switch a relay at shutdown, the pin should be low when shutdown so set the relay in normally closed if you want to high trigger something when the cb2 is off.

aftansin, 2025-01-04:
the same question. Did you find solution?

bigtreetech, 2025-01-04:
This has always been possible with the BTT Pi. Look in the config.txt for all of the possible overlays. Read page 11 and 12 of this manual for more info about shutdown pins.

https://github.com/bigtreetech/SKSM/blob/master/BIGTREETECH%20SKSM%20V1.0%20User%20Manual.pdf

closed 2025-01-04 by bigtreetech

Shaun87, 2025-01-06:
Thank you very much for your assistance. The SKSM looks like a great addition to my setup.
Just to be sure, will **param_gpio_poweroff_pin=GPIO0_C3** work to send a power off signal when the CB2 has turned off?

bigtreetech, 2025-01-06:
> Thank you very much for your assistance. The SKSM looks like a great addition to my setup. Just to be sure, will **param_gpio_poweroff_pin=GPIO0_C3** work to send a power off signal when the CB2 has turned off?

Yes, the intention is to let the SKSM know that the power down is complete. This gives it an understanding of the state of the SBC which is important in the following scenario:

- User turns machine power off.
- SBC is told to shut down by SKSM.
- SBC initiates the shutdown sequence.
- SBC indicates to SKSM that shutdown sequence is complete using that pin (in reality, it is not 100% complete since the SoC still needs some control over the pin in order to change the state but it is 98% complete).
- User turns machine power back on before the SKSM has run out of reserve energy.
- SKSM is aware that the SBC is in a shutdown state so it sends the "kick start" signal. Without this signal the SBC would just remain asleep even with the power applied.