bigtreetech/CB2 #52: Manta M8P v2.0 with CB2 not booting from sd Card

Alterminifahrer1986, 2024-12-16:
I had an issue with mainsail and I erase everything with rk dev tool as described.
I push back the DiP switches 3-4 to off.
Also I make a new image on the sd card and flash it fresh on the sd card and insert it in the M8P v2.

The system is not flashing or the green light is not lightning. Also there is no chance to find the board over dfu or WLAN

BTTJUSTIN, 2024-12-20:
Thanks for reaching out. Our support team does not operate using the GitHub platform but we would be happy to help you via
email based ticketing system. Please visit https://biqu.equipment/pages/technical-support and submit a ticket and we will be in touch with you.

KD-4-Lyfe, 2025-01-13:
Got same sort of issue after trying to setup ebb36 and eddy duo, I reverted back to my cb1 and everything is working

looxonline, 2025-01-14:
As long as the eMMC is truly empty, the SD has a bootable partition on it (don't use balena etcher as there is a bug in the latest version) and the dip switch is set to boot from SD then it should be OK. If there are issues after that then please reach out to the support team using the ticket tool mentioned above and they will offer personalised help.

closed 2025-01-14 by looxonline

KD-4-Lyfe, 2025-01-14:
I just found a fix for my issue.

First off belena etcher does work but you need to extract the image you download to the PC first then use that otherwise use raspberry pi imager that will extract it as part of the process,

My issue was the wifi antenna, i noticed in a particular position it would flash green, I then put on the antenna from my CB1 and it worked.

hope this helps someone later.