bigtreetech/CB2 #53: Time out error

KA009onPC, 2024-12-24:
I keep getting this error, sometimes 4 hours into a print.
[image]
[image]
 It's been getting frustrating

bigtreetech, 2024-12-25:
Make sure that you have no processes running on your CB2 that are starting up and consuming a lot of CPU. Also, this error is often masking wiring issues. If I were to guess, it would likely be a wiring issue.

KA009onPC, 2024-12-25:
Goodmorning, the printer will run for hours or even days and then randomly during a 4 hour print it'll throw this error, cpu load is around 10-20 percent while printing, it's never high.

KA009onPC, 2024-12-25:
Which wiring could it be?

pgradeckas, 2024-12-29:
I've been fighting this issue for months on my two Vorons when upgraded them to CanBus. First of all on a RPi3b+ using 64bit OS, then on a BTT Pi2 (obviously 64bit) and then finally switched back to RPI3b+ with 32bit OS and all the issues are gone on both machines. BTT Pi2/CB2 cannot assure reliably working canbus setup and it does not look like BTT is able to get this addressed. 
BTW - my third voron that runs manta + cb1 and SB2209 canboard never had this issue.

igeighty, 2024-12-31:
https://docs.siboor.com/siboor-trident-june/faq-oct-17/new-cb2-timer-too-close i changed MCY.py as described. this has allowed me to avoid MCU error for a day or more, used to occur hourly

godbnc, 2025-01-04:
wait to release version to cb2 because cb1 no have problem ...

godbnc, 2025-01-04:
try this ~/moonraker-env/bin/pip install -r ~/moonraker/scripts/moonraker-speedups.txt
# Move all logs from SDCard to RAM
# 2 commands, copy and paste as is:

GID=$(id -g)

echo "tmpfs $HOME/printer_data/logs/  tmpfs  
defaults,noatime,nosuid,nodev,noexec,mode=0755,uid=$UID,gid=$GID,size=200M 0 0" | sudo tee -a /etc/fstab

# Special note if you mapped pi to biqu user
# while it totally works as it it would be best 
# to move the newly created line from this command 
# below the other entry that bind mounts pi to biqu home folder

also create a bind mount for pi to biqu logs if you are sshing with pi (see compat below and also crownests is hard coded)

echo "/home/biqu/printer_data/logs/ /home/pi/printer_data/logs/ none defaults,bind 0 0" | sudo tee -a /etc/fstab

# connect to your printer via ssh
# run these lines one after each other

sudo apt-get install cpufrequtils

# Single line command, copy and paste as is:
echo '%sudo ALL=(ALL:ALL) NOPASSWD: /usr/bin/cpufreq-set' | sudo tee -a /etc/sudoers.d/cpufreq-set

sudo chmod 440 /etc/sudoers.d/cpufreq-set

godbnc, 2025-01-04:
use this https://www.printables.com/model/736220-btt-cb1-customization for timer to close look to down page