bigtreetech/CB2 #55: No NPU driver

Sukhoi-air, 2025-01-16:
[image]
I used V1.0.5.1 image and didnt find NPU driver,please add NPU driver to the kernel.
Thanks

zhangyumingsh, 2025-01-21:
> [image] I used V1.0.5.1 image and didnt find NPU driver,please add NPU driver to the kernel. Thanks

The driver has been compiled into the kernel. Please replace the dtb file and try again

> [image] I used V1.0.5.1 image and didnt find NPU driver,please add NPU driver to the kernel. Thanks

Please provide your email so that I can send it directly to your inbox

Sukhoi-air, 2025-01-22:
> The driver has been compiled into the kernel. Please replace the dtb file and try again
>
>
> Please provide your email so that I can send it directly to your inbox

thanks, 595664276@qq.com

isilcala, 2025-04-16:
or at least share the file to the public

closed 2025-08-27 by Sukhoi-air