bigtreetech/CB2 #58: BTT what's wrong with you? New hardware with old & buggy software?

MarkusNiewoehner, 2025-01-25:
When will you finally be able to release a new and working version? It's really starting to be a joke...

looxonline, 2025-01-26:
New image is due for release just after the Chinese New year. We were just finishing up the image processing part before release.

bderancourt, 2025-01-26:
Good news !

Will it resolve issue #10 ?

TrevTrev84, 2025-01-26:
I bought a PI2 AI when first released... it can't even run a 920C camera... I'm ordering an Orange PI... Higher Clocks, more memory and storage...

krzysiekslimak, 2025-01-30:
Happy Chinese New Year! So we are just after, how Your plans for update looks?

looxonline, 2025-01-30:
> Happy Chinese New Year! So we are just after, how Your plans for update looks?

https://www.shipbob.com/blog/chinese-new-year-shutdown/

It is a two week shutdown.

krzysiekslimak, 2025-01-30:
Chinese New Year’s Day: January 29, 2025
The Lantern Festival: February 12, 2025
 
So not "just after Chinese New Year" but "just after Lantern Festival"?  That's a big difference, don't you think?

looxonline, 2025-01-30:
> Chinese New Year’s Day: January 29, 2025 The Lantern Festival: February 12, 2025
> 
> So not "just after Chinese New Year" but "just after Lantern Festival"? That's a big difference, don't you think?

No, they are completely different things. The new year shutdown is a two week shutdown separate from the lantern festival. It is the equivalent of the western summer shutdown. The colloquial term when referring to the shutdown is simply "Chinese new year". You will find accurate shutdown times in the link provided.

krzysiekslimak, 2025-01-30:
As customer, when You wrote "just after Chinese new year" which was on 29 January 2025, I understand it as 30 January 2025. If now there is 2 weeks off why You just don't wrote to us, that You will push this update in the second half of February? Be more precise, we pay for BTT stuff and You treat us like Idiots.

looxonline, 2025-01-30:
> As customer, when You wrote "just after Chinese new year" which was on 29 January 2025, I understand it as 30 January 2025. If now there is 2 weeks off why You just don't wrote to us, that You will [...]

As I mentioned already, the accurate shutdown times are in the link provided. I.e. Until Feb 4/5th. I also mentioned that the colloquial term for the shutdown period is simply "Chinese new year". This encompases the whole shutdown period.

Didier-T, 2025-02-01:
> En tant que client, lorsque vous avez écrit « juste après le nouvel an chinois », soit le 29 janvier 2025, je comprends que c'était le 30 janvier 2025. S'il y a maintenant 2 semaines de congé, pourquoi ne nous avez-vous pas simplement écrit que vous lancerez cette mise à jour dans la deuxième quinzaine de février ? Pour être plus précis, nous payons pour des trucs BTT et vous nous traitez comme des idiots.

Je crois que le réel souci est qu'il ne nous a pas pris pour des idiots et ne nous a donc pas donné toutes les informations qui a ses yeux semblaient être évidentes.

Je suis moi aussi impatient de pouvoir profiter de cette mise a jour

krzysiekslimak, 2025-02-06:
Hi @looxonline, any news for us?

looxonline, 2025-02-06:
> Hi [@looxonline](https://github.com/looxonline), any news for us?

About to release public beta. Hang tight a few more hours. Just waiting for the files before I can upload them to github releases.

bigtreetech, 2025-02-06:
https://github.com/bigtreetech/CB2/releases/tag/V2.0.0

bderancourt, 2025-02-11:
Is it possible to provide a bullseye release with a recent kernel and a working ISP ? Thank you.

bigtreetech, 2025-02-12:
> Is it possible to provide a bullseye release with a recent kernel and a working ISP ? Thank you.

Check out the beta release notes. You will see that the ISP is coming next.

isilcala, 2025-02-14:
> > Is it possible to provide a bullseye release with a recent kernel and a working ISP ? Thank you.
> 
> Check out the beta release notes. You will see that the ISP is coming next.

does this image come with the NPU driver?

MamedovArtur, 2025-02-24:
Does tft35 work only in text mode?
Xorg does not starts 
`

(==) Assigned the driver to the xf86ConfigLayout[   312.986] (II) LoadModule: "modesetting"
[   312.987] (II) Loading /usr/lib/xorg/modules/drivers/modesetting_drv.so[   312.987] (II) Module modesetting: vendor="X.Org Foundation"
[   312.987]    compiled for 1.21.1.7, module version = 1.21.1[   312.987]    Module class: X.Org Video Driver
[   312.987]    ABI class: X.Org Video Driver, version 25.2[   312.988] (II) LoadModule: "fbdev"
[   312.988] (II) Loading /usr/lib/xorg/modules/drivers/fbdev_drv.so[   312.988] (II) Module fbdev: vendor="X.Org Foundation"
[   312.988]    compiled for 1.21.1.3, module version = 0.5.0[   312.988]    Module class: X.Org Video Driver
[   312.988]    ABI class: X.Org Video Driver, version 25.2[   312.988] (II) modesetting: Driver for Modesetting Kernel Drivers: kms
[   312.988] (II) FBDEV: driver for framebuffer: fbdev[   312.990] (II) modeset(0): using drv /dev/dri/card0
[   312.990] (WW) Falling back to old probe method for fbdev[   312.990] (II) Loading sub module "fbdevhw"
[   312.990] (II) LoadModule: "fbdevhw"[   312.990] (II) Loading /usr/lib/xorg/modules/libfbdevhw.so
[   312.991] (II) Module fbdevhw: vendor="X.Org Foundation"[   312.991]    compiled for 1.21.1.7, module version = 0.0.2
[   312.991]    ABI class: X.Org Video Driver, version 25.2[   312.991] (WW) VGA arbiter: cannot open kernel arbiter, no multi-card support
[   312.991] (II) modeset(0): Creating default Display subsection in Screen section    "Default Screen Section" for depth/fbbpp 24/32
[   312.991] (==) modeset(0): Depth 24, (==) framebuffer bpp 32[   312.991] (==) modeset(0): RGB weight 888
[   312.991] (==) modeset(0): Default visual is TrueColor[   312.991] (II) Loading sub module "glamoregl"
[   312.991] (II) LoadModule: "glamoregl"[   312.992] (II) Loading /usr/lib/xorg/modules/libglamoregl.so
[   313.010] (II) Module glamoregl: vendor="X.Org Foundation"[   313.010]    compiled for 1.21.1.7, module version = 1.0.1
[   313.011]    ABI class: X.Org ANSI C Emulation, version 0.4[   314.096] (II) modeset(0): Refusing to try glamor on llvmpipe
[   314.098] (II) modeset(0): glamor initialization failed[   314.098] (II) modeset(0): ShadowFB: preferred NO, enabled NO
[   314.099] (WW) modeset(0): No outputs definitely connected, trying again...[   314.099] (WW) modeset(0): Unable to find connected outputs - setting 1024x768 initial framebuffer
[   314.099] (==) modeset(0): Using gamma correction (1.0, 1.0, 1.0)[   314.099] (EE) modeset(0): No modes.
[   314.099] (II) UnloadModule: "modesetting"[   314.099] (II) UnloadSubModule: "glamoregl"
[   314.099] (II) Unloading glamoregl[   314.100] (EE) Screen(s) found, but none have a usable configuration.
[   314.100] (EE) Fatal server error:
[   314.100] (EE) no screens found(EE) [   314.100] (EE) 
Please consult the The X.Org Foundation support      at http://wiki.x.org
 for help. [   314.100] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[   314.100] (EE) [   314.102] (EE) Server terminated with error (1). Closing log file.

`

bat86, 2025-03-04:
When are the updates for the beta version planned? The release notes mention that beta releases will be quickly released over the next few weeks until the final release. A month has already passed, and there haven't even been solutions to some fairly critical issues.

Karl-Barx, 2025-03-19:
Yeah, really tired of the run around with BTT honestly thinking of moving to an LDO when the leviathan is avaliable again. Zero update just "it's coming soon trust us, bro".

bat86, 2025-03-20:
@bigtreetech @looxonline  Any updates?