bigtreetech/CB2 #60: need to reflash after shutdown or reboot?

exerkaoss, 2025-02-07:
after updating the whole group of apps after fresh install (klipper/moonraker/mainsail/etc) the filesystem goes in read only mode and lets me connect only once until i reflash it. I tried different sd cards, different sizes but the problems appear to be the same ove r and over (reflashing now both my btt pi2 (i have 2 printers). before this one of the printers run off an acer aspire one netbook and never had a problem. 
Could it be an sd card issue? is getting flashed bad? I tried verbatim, samsung extreme pro, kingston and another brand from 8-16-32 and even 64 gb size but nothing changes.

frutabruta, 2025-02-07:
It could be the imager. I had a problem with Balena Etcher/Rapberry Pi imager (one of them worked), but it didn't boot at all.

exerkaoss, 2025-02-07:
> It could be the imager. I had a problem with Balena Etcher/Rapberry Pi imager (one of them worked), but it didn't boot at all.

i tried both but balena was giving me an ending error. pi imager doesn t and until i reboot is all ok. it's so weird :(

frutabruta, 2025-02-07:
> i tried both but balena was giving me an ending error. pi imager doesn t and until i reboot is all ok. it's so weird :(

Did you do sudo apt upgrade? If yes, it might have broken some packages and caused the error.

exerkaoss, 2025-02-07:
> > i tried both but balena was giving me an ending error. pi imager doesn t and until i reboot is all ok. it's so weird :(
> 
> Did you do sudo apt upgrade? If yes, it might have broken some packages and caused the error.

wait, sudo upgrade is a no-no? i didn t know that

exerkaoss, 2025-02-07:
mostly, i started to see this error since the new moonraker update. i once did it on the second host i was using and caused the same problem. can be that the new moonraker has some compatibility problem with upgrading the system image and does this?

frutabruta, 2025-02-07:
> wait, sudo apt upgrade is a no-no? i didn t know that

I don't know if this is true for the current image, but the early CB1 image was hacked together in a way that apt upgrade just broke it.
In a normal production system this definitely shouldn't happen and usually it's a safe command.

exerkaoss, 2025-02-07:
> > wait, sudo apt upgrade is a no-no? i didn t know that
> 
> I don't know if this is true for the current image, but the early CB1 image was hacked together in a way that apt upgrade just broke it. In a normal production system this definitely shouldn't happen and usually it's a safe command.

I agree it shouldn't. Is the first thing i do with update after a fresh install on every ubuntu or other debian i had. 
I even tried the new CB2 debian12 2.0.0 beta1, got kiauh, install all, but guess what happened after i tried to reboot :D

frutabruta, 2025-02-08:
Ok, I tried it now and everything works, even after reboots
What i did:
 - downloaded CB2 debian12 2.0.0 beta1
 - opened Raspberry Pi imager
 - selected the .xz file without unpacking it first
 - you mustn't use the settings/customisation in the imager
 - then I opened the file system.cfg from the BOOT partition in VS Code (don't use Notepad for this!) and set up my Wi-Fi credentials
 - boot the system with HDMI screen and keyboard
 - setup wizzard pops up in the console
     - I did set up a new root password
     - created a new user, which I use to log in and install everything later on
     - setup locale and the rest of the wizzard
 - reboot
 - the rest is done over SSH:
   - sudo apt update
   - sudo apt upgrade
   - reboot   
   - install KIAUH
      - install Klipper
      - install Moonraker
      - install Mainsail
      - install Klipperscreen
  - reboot
  - copy all config files from backup
  - reboot
  - everything works as exected

bigtreetech, 2025-02-08:
> after updating the whole group of apps after fresh install (klipper/moonraker/mainsail/etc) and rebooting/shutdown -h now, the filesystem goes in read only mode and lets me connect only once until i reflash it. I tried different sd cards, different sizes but the problems appear to be the same ove r and over (reflashing now both my btt pi2 (i have 2 printers). before this one of the printers run off an acer aspire one netbook and never had a problem. Could it be an sd card issue? is getting flashed bad? I tried verbatim, samsung extreme pro, kingston and another brand from 8-16-32 and even 64 gb size but nothing changes.

I'm thinking that this may be related to having data both on the SD card and on the eMMC. Did you ever try to program the eMMC at all? If not, then it may have been that your unit had an image on the eMMC during factory testing at some point. If there is a valid boot partition on both memories then the bootloader sometimes has race conditions where the eMMC will win and this will cause it to boot to whatever is on there even if the kernel is not valid. The best way to prevent this is to do two things:

1. Check your dip switches to ensure that they are set to boot from SD and not eMMC (the BTT P2/CB2 manuals have this info). 
2. Use RKDevTool or an ssh interface to completely erase the eMMC. Again, the instructions for doing this are in the manuals.

bigtreetech, 2025-02-08:
One last point, there is no risk at all in doing an update/upgrade via apt.

exerkaoss, 2025-02-08:
> Ok, I tried it now and everything works, even after reboots What i did:
>
> * downloaded CB2 debian12 2.0.0 beta1
> * opened Raspberry Pi imager
> * selected the .xz file without unpacking it first
> * you [...]
**_### >   * reboot_**
>   * install KIAUH (EDIT: I selected to use the experiental v6 after launch)
>     
>     * install Klipper
>     * install Moonraker
>     * install Mainsail
>     * install Klipperscreen
> * reboot
> * copy all config files from backup
> * reboot
> * everything works as exected

already 3rd time and at that highlighted reboot i have the problem. like btt suggested i moved the emmc pin to on in order to short it and not use it but to no avail. do i have to start to think i have a faulty unit?

exerkaoss, 2025-02-08:
> I'm thinking that this may be related to having data both on the SD card and on the eMMC. Did you ever try to program the eMMC at all? If not, then it may have been that your unit had an image on [...]

I had the switch on 1 so i put it on ON like the manual says, but after setting up the user and then apt update-apt upgrade, i have the same problem and i have to reflash, configure eddy again and so on. at this point i may just not apt upgrade.

bigtreetech, 2025-02-08:
> I had the switch on 1 so i put it on ON like the manual says, but after setting up the user and then apt update-apt upgrade, i have the same problem and i have to reflash, configure eddy again and [...]

Check if the problem happens *before* you do an update/upgrade but after you reboot. Also, even if you change the switch, it can still attempt to boot off eMMC. You must ensure that eMMC is completely erased.

tpruvot, 2025-02-09:
Also remove the sd card after the emmc is copied, because they can have the same UUID partitions. I noticed that on previous versions... not yet tested this one. check with "blkid"

exerkaoss, 2025-02-09:
lil update. deb 11 worked until this morning. managed to apt update, upgrade, update all apps for the printer. it did shutdown and turn on without problems. shut it down again in the night after a lil print.
now woke up to the same problem. I am getting frustrated because  I waslooking forward to move from my laptop hosts but not like this. I never had to fiddle and flash so much with an sbc like i did with this two pi2. disappointed

exerkaoss, 2025-02-09:
> Also remove the sd card after the emmc is copied, because they can have the same UUID partitions. I noticed that on previous versions... not yet tested this one. check with "blkid"

thx but doesn't work anyways, mmc on or off, card in or out. it will give the same problem always. Just starting to reflash now after the night.

bigtreetech, 2025-02-09:
> thx but doesn't work anyways, mmc on or off, card in or out. it will give the same problem always. Just starting to reflash now after the night.

Did you completely erase the eMMC as I previously mentioned?

exerkaoss, 2025-02-09:
> Did you completely erase the eMMC as I previously mentioned?

yes, i redid everything now following manual and decided to copy everything on mmc so i m going cardless. i could apt update and install all the plethora of printing stuff. reboot and shutdown work but when apt upgrade i receive this.

[image]

bigtreetech, 2025-02-09:
It seems like you did indeed have data on the eMMC and that was causing your previous boot issues. This is why I mentioned that you need to erase it completely if you want the SD to be reliable.

Regarding the download issues you are seeing, that could be due to the armbian repo currently being down or due to a connection issue. It seems that the repo is currently reporting incorrect file sizes. Be sure that you have run a sudo apt update just prior to attempting an apt upgrade. Also leave a ping to google.com running for a few minutes to see if there are any dropped packets as a way to test your connection.

exerkaoss, 2025-02-09:
> It seems like you did indeed have data on the eMMC and that was causing your previous boot issues. This is why I mentioned that you need to erase it completely if you want the SD to be reliable. [...]

i left the google ping going and i didn t have lost packets so i suppose network side should be ok.
Next I'm calibrating eddy again and trying a print. will update

bigtreetech, 2025-02-10:
Good stuff. Closing.

closed 2025-02-10 by bigtreetech