bigtreetech/CB2 #62: Only 2.4G wifi network awailable.

krzysiekslimak, 2025-02-08:
I just install new beta os, unfortunately only 2.4g wifi work. Is there any additional settings or drivers to run 5g?

bat86, 2025-02-10:
I can see 5Ghz, but i can't  connect to this, because of WPA3 support missing

frutabruta, 2025-02-11:
Maybe setting the Wi-Fi country? Not all 5GHz frequencies all available in all countries and I don't know, which one is set in CB2 by default. My 802.11n 5GHz network with WPA2 Personal on channel 57 works on CB2 without having to set the WiFi country.

krzysiekslimak, 2025-02-18:
Still didn't found solution. I can't change region for PL. 
`user@bigtreetech-cb2:/root$ sudo iw reg get
global
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (6, 20), (N/A)
        (2457 - 2482 @ 20), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (6, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (6, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (5250 - 5330 @ 80), (6, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (6, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (6, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

phy#0 (self-managed)
country 99: DFS-UNSET
        (2402 - 2482 @ 40), (6, 20), (N/A)
        (2474 - 2494 @ 20), (6, 20), (N/A)
        (5140 - 5360 @ 80), (6, 20), (N/A)
        (5460 - 5920 @ 80), (6, 20), (N/A)
        (5945 - 7065 @ 160), (6, 20), (N/A)
        (5925 - 7125 @ 20), (6, 20), (N/A)
        (5945 - 7105 @ 40), (6, 20), (N/A)
        (5945 - 7065 @ 80), (6, 20), (N/A)`

`user@bigtreetech-cb2:/root$ sudo iw reg set PL` - didn't work

wihtouc correct region I can't use 5ghz wifi

bat86, 2025-02-20:
I think it's not region issue. Try to  change wifi  security from WPA3 or WPA2/WPA3 to WPA2. For me this works

krzysiekslimak, 2025-02-20:
In my case it is 100% region problem, becuase lot of frequencies are disabled. I just change channel on my router in 5ghz for lower and now I can connect.