bigtreetech/CB2 #63: Temp error on cold boot

d0u8l3m, 2025-02-09:
Hello 

I am using the new 2.0 beta Armbian 12 release and have gotten Klipper installed and working.

When I power up my printer with a CB2 and M5P from a cold boot I get this error when it tries to first start. 

unable to open temperature file '/sys/class/thermal/thermal_zone0/temp'

Pressing FW reset resolves the issue but not sure why it's not ready by the time the OS is booted. 

I don't see anything in the klippy.log file but is that the correct location to be looking for startup errors? 

Thank you!

bigtreetech, 2025-02-09:
Thanks for reporting! We will get this looked into and patched in a subsequent beta. @hwisgold

d0u8l3m, 2025-02-09:
Awesome thank you, if I can provide some logs or something please let me know.

bat86, 2025-02-10:
> Hello
>
> I am using the new 2.0 beta Armbian 12 release and have gotten Klipper installed and working.
>
> When I power up my printer with a CB2 and M5P from a cold boot I get this error when it tries to [...]

Same problem