bigtreetech/CB2 #65: UART devices not showing

exerkaoss, 2025-02-16:
[image]
Hello,
i have connected a device to those pins but nothing is showing up.
here my armbian config and ls /dev

[image]

[image]

I'm new to the Pi2 and the SBCs in general and i can't understand if i'm doing something wrong.
The purpose of this is to use an old TFT 43 v3 as an interface for klipper as shown in https://github.com/oldhui-uk/tftbridge?tab=readme-ov-file

Thanks for your help!

Pneumanifest, 2025-02-17:
put all overlays on the same line separated with one space.

exerkaoss, 2025-02-17:
> put all overlays on the same line separated with one space. like this `overlays=hdmi uart0`

i did and activated only UART0 but /dev is looking the same and the device is not showing up.
would a uart device show as ttyAMA or ttyS? everytime i change uart overlay i see those ttys changing.

Pneumanifest, 2025-02-17:
Yes ttyAMA I believe. You have the firmware for uart on the MCU already right?

exerkaoss, 2025-02-17:
> Yes ttyAMA I believe. You have the firmware for uart on the MCU already right?

the touchscreen will communicate with the PI2, not the octopus, that is still connected usb to the PI

Pneumanifest, 2025-02-17:
Wait what are you trying to connect to UART? A touchscreen?

exerkaoss, 2025-02-17:
> > > put all overlays on the same line separated with one space. like this `overlays=hdmi uart0`
> > 
> > 
> > i did and activated only UART0 but /dev is looking the same and the device is not showing up. would a uart device show as ttyAMA or ttyS? everytime i change uart overlay i see those ttys changing.
> 
> Yes, it will be a tty device.

ok, but ttyAMA or ttyS? as of now i can see ttyS changing number when i change uart overlay, but no device

Pneumanifest, 2025-02-17:
Do you have the screen baud rate set to 250000? I think default is 115200

exerkaoss, 2025-02-17:
> Do you have the screen baud rate set to 250000 in the screen settings? I think default is 115200

yes the baudrate can be changed also in setting thru the UI. i've tried every baudrate since yesterday. :(

Pneumanifest, 2025-02-17:
No trying. All baud rates must match in all sections. 1 mismatch can mess up all communication. And you have TX going to RX right? Being that you are seeing the tty show up states that things are at least trying to work. Keep in mind that what you are trying to use is not officially klipper nor BTT code. There's always the possibility of something that has broke compatibility in an update. That repo you linked has not been maintained since last year.

exerkaoss, 2025-02-17:
> No trying. All baud rates must match in all sections. 1 mismatch can mess up all communication. And you have TX going to RX right? Being that you are seeing the tty show up states that things are at [...]

i see. yes all bauds are 250000 in all the places. yes i have tx to rx and rx to tx like shown in the tutorial's pic (pins are not split so is idiot-proof).
I understand what you mean but at this point i don't know if the problem is screenside, which would anyway work if i connect it to the mcu in marlin mode (bleh), or tftbridge side.

exerkaoss, 2025-02-17:
an addition to this:

[image] is it normal to have 2 consoles at line 3 and 8?

tpruvot, 2025-02-17:
no, its not, delete the 3rd line

exerkaoss, 2025-02-17:
> no, its not, delete the 3rd line

done. i've also tried a spare tft 35 v3 i had but is the same problem and i just reflashed it. why are the devices not coming up? :'(

tpruvot, 2025-02-17:
serial tty are /dev/ttySxx, but avoid uart0 on first 2 pins, because the boot is using it to display stuff (like a bios), better to keep it as default serial remote console (for putty)

exerkaoss, 2025-02-17:
> serial tty are /dev/ttySxx, but avoid uart0 on first 2 pins, because the boot is using it to display stuff (like a bios), better to keep it as default serial remote console (for putty)

so shall i split the pins and send tx and rx to other gpio pins?

tpruvot, 2025-02-17:
i see on the overlay list a lot of other possible uarts, but ive no idea yet which pins are used

exerkaoss, 2025-02-17:
> i see on the overlay list a lot of other possible uarts, but ive no idea yet which pins are used (1 3 5 7)

no clue as well. i have the PI2 as my first ever sbc since a month or so because i wanted to move host from a laptop.

exerkaoss, 2025-02-17:
IT WORKS!  used parts of this and connected to same pins and is uart5. IT FINALLY WORKS! BOTH SCREENS!

[image]

exerkaoss, 2025-02-17:
https://github.com/user-attachments/assets/2d42d5cb-d792-4245-acfd-b836f773ae78

Pneumanifest, 2025-02-17:
Nice looks like you got it going!
One note on the console lines. The first one (lines) was 2 in 1 ```console=serialoverlay_prefix=rk3566``` should be ```overlay_prefix=rk3566```

exerkaoss, 2025-02-17:
> Nice looks like you got it going! One note on the console lines. The first one (lines) was 2 in 1 `console=serialoverlay_prefix=rk3566` should be `overlay_prefix=rk3566` The other `console=serial ` on line 5 is good where it is.

thank you so much for your precious time and the support you and others gave me on this post. I had this screens off since june 2024 and despite i was always using mobileraker on a tablet or mainsailweb or another device, is nice to have something to interact directly with the printer. <3

tpruvot, 2025-02-17:
overlay_prefix=rk3566 line is already present a few lines under too

Pneumanifest, 2025-02-17:
> overlay_prefix=rk3566 line is already present a few lines under too

Ah, yes it is. I didn't see that. 

> thank you so much for your precious time and the support you and others gave me on this post. I had this screens off since june 2024 and despite i was always using mobileraker on a tablet or [...]

It's cool seeing those screens getting used with klipper. Last i saw (quite a while ago) Kevin (Klipper dev) had no interest in implanting support for them.

exerkaoss, 2025-02-17:
> > overlay_prefix=rk3566 line is already present a few lines under too
> 
> Ah, yes it is. I didn't see that.
> 
_for the moment is commented out but working after reboot, i will modify with your suggestions and by the time i finish this print (morrow)_
> It's cool seeing those screens getting used with klipper. Last i saw (quite a while ago) Kevin (Klipper dev) had no interest in implanting support for them.

_I'm glad the solution despite not updated since a year, still works. the python script looks simple and i hope is gonna work for a while_

exerkaoss, 2025-02-17:
> overlay_prefix=rk3566 line is already present a few lines under too

oh i read this later, so nothing to modify. i closed the issue.
Thanks for your time and helping me!

closed 2025-02-17 by exerkaoss