bigtreetech/CB2 #66: WiFi Passwords with a "$" in them won't save correctly if entered into system.cfg preventing a wifi connection from being established.

jmart384, 2025-02-20:
My Wifi password has a "$" in it and for some reason this prevents it from automatically connecting to wifi if I enter it into the "System.cfg" as instructed in the manual. I had to google the baud rate and ssh into my CB2 over usb then use network manager ("nmtui" command) to see what was going on and noticed that the saved password for my wifi network didn't have the $ or any characters after it so I corrected this and it automatically connected but upon a restart it did it again so I removed the Wifi SSID and Password from System.cfg and corrected the password in network manager and now it succesfully reconnects upon a restart. I was ignoring this issue for months and just reconnecting to my wifi manually using klipperscreen on my btt HDMI 5 upon every restart but that was incredibly tedious and frustrating. A lot of this could have been sorted much more easily. Firstly I suggest including the usb ssh baud rate in the manual as it took me far too long to figure out that 1. having the wrong baud rate was the only reason I couldn't use putty and 2. what that value even was. I'd also suggest documenting more clearly how to set the wifi name and password in system.cfg if you are flashing the emmc with the rkdevtool rather than an sd card as that also took way too long to figure out as I hadn't really used Linux prior to buying the CB2. Finally I'd suggest fixing the issue with "$" in wifi passwords causing them to apply incorrectly.

bigtreetech, 2025-02-20:
All very good points. Thanks for the feedback. We will make some updates to the docs and patches in the next software image. @BTTJUSTIN

bigtreetech, 2025-02-25:
Thanks for the heads up. We will update the system.cfg files in the new images to point this out.

closed 2025-02-25 by bigtreetech