bigtreetech/CB2 #68: VI missing

jasonlshelton, 2025-02-20:
Just downloaded and installed v1.0.5.2 (20250220)

First thing I try to do is open a config file with vi, but it's not installed. I tried to use apt to install it, but it's not found. 

Where's vi? I'm not using that childish nano editor

bigtreetech, 2025-02-20:
You will use nano and you will be happy!

bigtreetech, 2025-02-20:
Although you do have a good point that VI is not found from the default sources because it most certainly should be.

jasonlshelton, 2025-02-20:
I've been using VI since probably before you were born. I will burn your OS to the ground before I use nano

jasonlshelton, 2025-02-20:
I was finally able to install VI by using that child's toy nano to edit the /etc/apt/source.list and uncommenting the lines for the binaries

Not including VI is a major faux pas and needs to be fixed

bigtreetech, 2025-02-20:
There is a kernel 6.1 image on the way where we will add VI for those who enjoy pressing too many keys ;)

jasonlshelton, 2025-02-20:
wow, you are kindof an asshole. VI has been standard on all flavors of unix for years, every admin should be able to log on to any unix system anywhere and have some for of vi for text editing.

bigtreetech, 2025-02-21:
Hi, please try
```
sudo apt update
sudo apt install vim
```

In the future, `vi` will be built-in in released image

bigtreetech, 2025-02-21:
>..... VI has been standard on all flavors of unix for years, every admin should be able to log on to any unix system anywhere and have some for of vi for text editing.

Mate, if it wasn't already clear from the winking face and earlier joking text I'll say the quiet part out loud. I was playfully having a bit of friendly banter.

It's common knowledge that there is playful banter between vim and nano users.

feldgendler, 2025-02-21:
I remember the holy war being Vi vs. Emacs long before the first line of nano was written.

jasonlshelton, 2025-02-21:
Yeah, I gotta apologize, I was having a bad day and running into this problem only made it worse. I'm sorry for being the A**hole in this situation. Thanks for the work you do, I should learn to be more appreciative. (In case anyone is wondering, no, there is no sarcasm or smiley face here, I'm just admitting I was wrong)