bigtreetech/CB2 #70: BigTreeTech Pi 2 Documentation Error. "UPS POWER" Specifications for connector is wrong.

EdenNelson, 2025-03-16:
The documentation under  "UPS POWER". the fallowing is stated:
"Specifications: HX5P-2.54MM horizontal type, compatible with our company's SKSM emergency power supply board."

The pads on the bottom of the Pi 2 for UPS Power is NOT 2.54mm it's 2.0mm. The SKSM has the 2.54MM connector, not the Pi 2.

[image]

zhangyumingsh, 2025-03-19:
> The documentation under "UPS POWER". the fallowing is stated: "Specifications: HX5P-2.54MM horizontal type, compatible with our company's SKSM emergency power supply board."
> 
> The pads on the bottom of the Pi 2 for UPS Power are NOT 2.54mm; they're 2.0mm. The SKSM has the 2.54MM connector, not the Pi 2.
> 
> [image]
> 
> See the picture below of the HX5P-2.54MM connector with Pi 2 UPS Power Pads.
> 
> [image]

What is the reason for asking customer service

EdenNelson, 2025-03-19:
I’m not asking for customer service. The documentation has errors and needs to be corrected.
I'll close the issue since the subject and description are quite clear and you are asking me:
> What is the reason for asking customer service

Pneumanifest, 2025-03-25:
Don't mark it completed unless you have confirmation that it has been fixed.

bigtreetech, 2025-03-26:
Corrected in 40b55050c5a45f7b29795f9aeab4498cf3875f90, thank you for the reminder

closed 2026-09-09 by EdenNelson