bigtreetech/CB2 #73: V3.0.0 Beta 2 CB2 on Manta M8P v2 no ethernet

nakaori, 2025-03-29:
Hi,

on a fresh installation I cannot get ethernet to work. When I turn on the Board both LEDs on the M8Ps NIC are turned on steadily regardless if a cable is plugged in or not. 

When I SSH into the board via wifi, I can see that the interface is down and no cable is being detected:

`2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000`

I already tried different cables to no success. According to the manual it should work out of the box, but it's not. 

Thanks for your time!

nakaori, 2025-03-31:
> aqui normal com a manta m8p 2.0, so a camera que so fica em 5 fps

I don't really get what this means. Are you saying that your board is working fine with wifi?

bigtreetech, 2025-03-31:
Thanks for reporting. We will look into this for the next beta. We may need to ask you for some additional info but we will be in touch if so.

bigtreetech, 2025-03-31:
> Hi,
>
> on a fresh installation I cannot get ethernet to work. When I turn on the Board both LEDs on the M8Ps NIC are turned on steadily regardless if a cable is plugged in or not.
>
> When I SSH into the [...]

Could you try V1.0.5 and see if the same thing happens with your ethernet? It will help us to narrow down where the issue may be.

klipper-vp, 2025-03-31:
> > aqui normal com a manta m8p 2.0, so a camera que so fica em 5 fps
> 
> I don't really get what this means. Are you saying that your board is working fine with wifi?

Sim, instalei sem problemas, rede funcionando, wifi funcionando, ja imprimi tudo prefeito..

nakaori, 2025-04-01:
> Could you try V1.0.5 and see if the same thing happens with your ethernet? It will help us to narrow down where the issue may be.

Hi,

thanks for your reply. I just installed v1.0.5 but sadly it's indeed the same thing. 

`2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000`

If I try "ifconfig eth0 up" nothing happens. State stays the same. If I give it an IP address like "ifconfig eth0 xxx.xxx.xxx.xxx/yy up" the whole networking stack locks up and the machine is unreachable until I reboot. CAN bus still works tho. 

The LEDs on the Manta M8P NIC are active and steady, but those on the switch the board is connected to are off. The switch does not see any MAC address on the port.

At this point I guess I have a broken board... It's sad because this is the second Manta M8P v2 I bought...