bigtreetech/CB2 #76: CB2 not booting & not detected by Windows (Manta m4p board)

hitman01560, 2025-03-31:
hello!
I can't get my cb2 working with my manta m4p using the v3 beta minimal image.

Flashing to sd then sudo nand-sata-install works. but after installing Kiauh, Klipper etc and updating the system it won't boot anymore.

tried erasing the emmc with the rkdevtool following the wiki, putting pins 3&4 on, my computer
doesn't detect the cb2 

what can I do? need some help please

bigtreetech, 2025-04-01:
Hello, please pay attention to the `Known issues with beta 2` section in the note of V3.0.0 beta2. It uses the mainline uboot, and there was an issue with the `recovery` button entering dfu mode in the V3.0.0 beta2 system (we have now resolved it, but there is no new system image yet),

So please refer to section `4.4.2` of the manual,
Insert the SD card that has already been written to the system (in this version of the system, if both eMMC and SD exist, boot from the SD card first), and then execute the `sudo mkfs /dev/mmcblk1` command to erase the contents of eMMC

eltoozero, 2025-04-01:
**> updating the system it won't boot anymore.**

Related: https://github.com/bigtreetech/CB2/issues/77