bigtreetech/CB2 #77: CB2 Beta2 does not boot after armbian-upgrade or apt-get upgrade

eltoozero, 2025-04-01:
```
Processing triggers for initramfs-tools (0.142+deb12u1) ...
ln: failed to create hard link '/boot/initrd.img-6.1.99-vendor-rk35xx.dpkg-bak' => '/boot/initrd.img-6.1.99-vendor-rk35xx': Operation not permitted
update-initramfs: Generating /boot/initrd.img-6.1.99-vendor-rk35xx
update-initramfs: Armbian: Converting to u-boot format: /boot/uInitrd-6.1.99-vendor-rk35xx
Image Name:   uInitrd
Created:      Tue Apr  1 12:22:24 2025
Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
Data Size:    17089518 Bytes = 16688.98 KiB = 16.30 MiB
Load Address: 00000000
Entry Point:  00000000
update-initramfs: Armbian: Symlinking /boot/uInitrd-6.1.99-vendor-rk35xx to /boot/uInitrd
ln: failed to create symbolic link '/boot/uInitrd': Operation not permitted
update-initramfs: Symlink failed, moving /boot/uInitrd-6.1.99-vendor-rk35xx to /boot/uInitrd
renamed '/boot/uInitrd-6.1.99-vendor-rk35xx' -> '/boot/uInitrd'
update-initramfs: Armbian: done.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

After issuing reboot the system does not boot, black screen only; I must re-image SD to restore function.

eltoozero, 2025-04-01:
I am able to work around the problem by manually holding the following packages:

```
apt-mark hold linux-image-vendor-rk35xx
apt-mark hold linux-dtb-vendor-rk35xx  
apt-mark hold armbian-firmware
```

From this point apt upgrade and armbian-upgrade can proceed normally and the system will reboot properly.

bigtreetech, 2025-04-02:
Got it. Thank you very much for your feedback.

The official image source of armbian includes .deb installation packages such as `linux-image-vendor-rk35xx`, `linux-dtb-vendor-rk35xx `,  `linux-image-legacy-rk35xx` and `linux-dtb-legacy-rk35xx `. Once the keywords(`-vendor-rk35xx`, or `legacy-rk35xx`) are matched, this .deb becomes upgradable. But in reality, we cannot update these installation packages as they will replace the drivers of the Linux system kernel.

In the next image system, we will replace `vendor` with a non-existent `btt`, and then `apt upgrade` will not be able to match the corresponding one `.deb`, so it won't update automatically

feldgendler, 2025-04-05:
What about `armbian-firmware`, are you going to stop it from upgrading somehow, too?

bigtreetech, 2025-04-07:
> What about `armbian-firmware`, are you going to stop it from upgrading somehow, too?

`armbian-firmware` should not be a problem.  In my testing, it did not cause any problems. If you have any abnormalities, please describe the situation to me and I will try to reproduce and solve it

raizer88, 2025-04-12:
Is this problem fixed in the new 3.0.1 ? @bigtreetech thanks.

OriginaldCrypt, 2025-04-13:
Moreover, how CB2 v3.0.0 image shall be upgraded to v3.0.1? @bigtreetech thanks.

OriginaldCrypt, 2025-04-15:
Thanks @raizer88 , can you be more precise? Can you detail the process a bit? 

Thanks!

raizer88, 2025-04-15:
I am speaking about the bug in the issue, not about updating from 3.x to 3.0.1.
I did a clean install from image and I could update/upgrade/reboot without problem.

OriginaldCrypt, 2025-05-03:
> Moreover, how CB2 v3.0.0 image shall be upgraded to v3.0.1? [@bigtreetech](https://github.com/bigtreetech) thanks.

How, @bigtreetech ? You told here https://github.com/bigtreetech/CB2/issues/77#issuecomment-2771785888 that the upgrade can't be done anymore after your changes but, how one with v3.0.0 can upgrade to the latest image keeping the configuration?

Thanks!