bigtreetech/CB2 #81: USB camera consumes a lot of CPU in version 3.0.1

jfml2020, 2025-04-13:
Causes random freezes camera

jfml2020, 2025-04-13:
> Causes random freezes when trying to view on TFT5 DSI screen I have a USB camera connected, and when I try to view the BTTPI TFT5 screen, there are random freezes, and the system indicates high CPU consumption. It only occurs when I activate the camera on the TFT5 DSI screen.

I have reinstalled the previous version V1.0.5 in eMMC and there are no problems with the CPU consumption when viewing the camera image on a DSI TFT5 connected to a BTT Pi 2....
With version 3.0.1, the CPU consumption increases to 60, 80,,,, % when viewing images on a DSI TFT5.

bigtreetech, 2025-04-14:
Confirmed, the problem has been reproduced on my end. Identifying the cause now

bigtreetech, 2025-04-23:
Hi, This may be due to the GPU not running properly.

We can download the `linux-image-btt-rk35xx_3.0.1-25.05.0-gpu.deb` file in https://github.com/bigtreetech/CB2/releases/tag/V3.0.1 to the system and run:
  ```
  sudo dpkg -i  linux-image-btt-rk35xx_3.0.1-25.05.0-gpu.deb 
  ```
  to install kernel updates. After installation, we can delete this `.deb` file, and the GPU is available after system restart.

jfml2020, 2025-04-25:
> Hi, This may be due to the GPU not running properly.
> 
> We can download the `linux-image-btt-rk35xx_3.0.1-25.05.0-gpu.deb` file in https://github.com/bigtreetech/CB2/releases/tag/V3.0.1 to the system and run:
> 
> ```
> sudo dpkg -i  linux-image-btt-rk35xx_3.0.1-25.05.0-gpu.deb 
> ```
> 
> to install kernel updates. After installation, we can delete this `.deb` file, and the GPU is available after system restart.
> 
> Looking forward to your feedback. If this issue is resolved, we will soon release a new version of the system image to solve it
It seems that now with the deb file it works. Thanks

jfml2020, 2025-04-25:
> Hi, This may be due to the GPU not running properly.
>
> We can download the `linux-image-btt-rk35xx_3.0.1-25.05.0-gpu.deb` file in https://github.com/bigtreetech/CB2/releases/tag/V3.0.1 to the system [...]

What I saw when installing the .deb is not important. I left a screenshot.

[image]

swollen-jid, 2025-04-29:
In my case, this update did not help, however I use a bit different DSI, which is Fysetc 5" DSI. 
Also, I had huge issues with canbus/eddy duo - timeouts all the time.
I decided to come back to the old stable version of the system.

knive87, 2025-04-29:
> In my case, this update did not help, however I use a bit different DSI, which is Fysetc 5" DSI. I had huge issues with canbus/eddy duo - timeouts all the time. ided to come back to the old stable version of the system.

Support it. The problem with Fysetc 5" DSI didn't solved.

jfml2020, 2025-04-29:
In my case, if the update works, the CPU stops consuming excessively, but with the insolation of the .deb and doing updates, etc., the USB bed randomly starts to become unavailable. 
I reinstalled the new version 3.0.1 and the .deb several times and after updating the camera it suggests not appearing or being available at every startup.

bigtreetech, 2025-04-29:
> In my case, if the update works, the CPU stops consuming excessively, but with the insolation of the .deb and doing updates, etc., the USB bed randomly starts to become unavailable. I reinstalled [...]

What is a USB bed?

swollen-jid, 2025-04-30:
> > > In my case, if the update works, the CPU stops consuming excessively, but with the insolation of the .deb and doing updates, etc., the USB bed randomly starts to become unavailable. I reinstalled the new version 3.0.1 and the .deb several times and after updating the camera it suggests not appearing or being available at every startup.
> > 
> > 
> > What is a USB bed?
> 
> Yes .... With the stable version there is no problem with the BTT Pi TFT50 v2.1 screen and no problem with the USB camera either.

But you said:
> > the USB bed randomly starts to become unavailable.

I wonder what you mean by that...

bigtreetech, 2025-04-30:
So do I. I have no idea how to help.

jfml2020, 2025-04-30:
English is not my language. Sorry.I'll try to explain.
When you install firmware 3.0.1 the USB bed is detected also when installing the fix .deb it also works the camera is detected but when doing update upgra or updating mainsail (I don't know exactly when) The USB camera is not detected when restarting randomly and sometimes it detects it, other times it doesn't (I know it sounds strange)  If I have time I will try to document or record the problem and download the log.