bigtreetech/CB2 #82: wlan0 is missing on CB2

yzheka, 2025-04-14:
I can't conenct to wifi because wlan is missing in the system. I have successfully flashed the latest os image but Wifi si not working
```
Last login: Thu Apr 10 07:40:22 UTC 2025 from 192.168.0.43 on pts/2
biqu@bigtreetech-cb2:~$ ifconfig
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 56:15:31:b8:2b:aa  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 100

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 907  bytes 98450 (96.1 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 907  bytes 98450 (96.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```
and is not listed among otehr interfaces

[image]

iamLellep, 2025-04-16:
Same problem. I cant use wlan0 to connect. In this Moment i downgrade to old version 1.0.5

yzheka, 2025-04-17:
I have located the issue. I had ipex connector fall off and why soldering it short circuit nearby capacitor. After cleaning all things wlan0 have appeared in the system

closed 2025-04-17 by yzheka