bigtreetech/CB2 #86: Please have the NPU driver built-in in the new OS image

isilcala, 2025-04-22:
The point that I bough this board is hoping to make use of the NPU.

bigtreetech, 2025-04-23:
Hi, Please download the `linux-image-btt-rk35xx_3.0.1-25.05.0-gpu.deb` file in https://github.com/bigtreetech/CB2/releases/tag/V3.0.1 to the system and run:
  ```
  sudo dpkg -i  linux-image-btt-rk35xx_3.0.1-25.05.0-gpu.deb 
  ```
  to install kernel updates. After installation, we can delete this `.deb` file, and the NPU driver should be available after system restart.

Looking forward to your feedback. If this issue is resolved, we will soon release a new version of the system image to solve it

isilcala, 2025-04-28:
> Hi, Please download the `linux-image-btt-rk35xx_3.0.1-25.05.0-gpu.deb` file in https://github.com/bigtreetech/CB2/releases/tag/V3.0.1 to the system and run:
>
> ```
> sudo dpkg -i [...]

NPU driver loaded with some errors:

`[    8.054088] RKNPU fde40000.npu: Adding to iommu group 0
[    8.055284] RKNPU fde40000.npu: RKNPU: rknpu iommu is enabled, using iommu mode
[    8.055385] RKNPU fde40000.npu: Looking up rknpu-supply from device tree
[    8.055720] RKNPU fde40000.npu: Looking up mem-supply from device tree
[    8.055900] RKNPU fde40000.npu: can't request region for resource [mem 0xfde40000-0xfde4ffff]
[    8.057201] [drm] Initialized rknpu 0.9.8 20240828 for fde40000.npu on minor 1
[    8.058451] RKNPU fde40000.npu: bin=0
[    8.058543] RKNPU fde40000.npu: leakage=4
[    8.058587] RKNPU fde40000.npu: pvtm = 88820, from nvmem
[    8.058607] RKNPU fde40000.npu: pvtm-volt-sel=2
[    8.058620] RKNPU fde40000.npu: soc version=0, speed=2
[    8.058738] RKNPU fde40000.npu: Looking up rknpu-supply from device tree
[    8.058833] RKNPU fde40000.npu: Looking up mem-supply from device tree
[    8.058913] RKNPU fde40000.npu: opp_parse_supplies: Invalid number of elements in opp-microvolt property (3) with supplies (2)
[    8.058928] RKNPU fde40000.npu: _of_add_opp_table_v2: Failed to add OPP, -22
[    8.058941] RKNPU fde40000.npu: failed to add opp table
[    8.059046] RKNPU fde40000.npu: RKNPU: failed to init_opp_table`