bigtreetech/CB2 #88: Communication timeout with canbus

thouve, 2025-04-24:
With version 1.05, there were canbus communication problems "Error during homing probe: Communication timeout during homing"
When I upgraded to version 2.0.0 beta 1, this problem was solved.
I've just upgraded to the latest version, 3.0.1, and the problem is back.
Impossible to make a QGL

Liimp, 2025-04-25:
> With version 1.05, there were canbus communication problems "Error during homing probe: Communication timeout during homing" When I upgraded to version 2.0.0 beta 1, this problem was solved. I've just upgraded to the latest version, 3.0.1, and the problem is back. Impossible to make a QGL

What is your CPU load during QGL? I have the same problem but I thought it was the probe.

bigtreetech, 2025-04-25:
This will be fixed up in the next release. The GPU is not being used properly in the current release which is placing a high load on the CPU. There is a deb patch that can be applied that is mentioned in another ticket but we will integrate that patch in the next release that is coming in a few days.

cjackson234, 2025-05-05:
when you say next release, do you mean the one that is out now - because it's not fixed for me

Johnnytk36, 2025-05-06:
> say

I was having the same issue using 3.0.2 with the PI2 after upgrading from the PI1. I tested the older versions of the OS also and had some better luck with them, but I need/want NVME support for data storage. So I wanted to be on the latest 3.0.2. 

What i ended up doing is installing the OS using the minimal version on a SD card and booting from it. Then i set up the armbian environment, used kiuah to install klipper, while manually installing everything else i wanted. After doing that i transferred the boot loader to the EMMC using the NVME for the data storage option.

I have a test macro that does all my start gcode stuff like head the hotend to 150 and heat the bed, but also loops my 7x7 bedmesh over and over 100 times. I could always get a canbus timeout doing at least doing that once on their "klipper version" of 3.0.2.  I am going on me third run right now, and no timeout errors yet.

I have no data for this, but i think my idle CPU percentage is lower this way also. I don't know if there is anything about their klipper version causing the issue, but it seems like it to me. 

Note: I had also removed all the Obico stuff from their klipper version and never installed it when i did my setup from the minimal version. 

Hope this helps you.

cjackson234, 2025-05-07:
I'll open a new issue since its all USB and not CAN.

bigtreetech, 2025-05-07:
> > when you say next release, do you mean the one that is out now - because it's not fixed for me
> 
> [@cjackson234](https://github.com/cjackson234)
> 
> I was having the same issue using 3.0.2 with the PI2 after upgrading from the PI1. I tested the older versions of the OS also and had some better luck with them, but I need/want NVME support for data storage. So I wanted to be on the latest 3.0.2.
> 
> What i ended up doing is installing the OS using the minimal version on a SD card and booting from it. Then i set up the armbian environment, used kiuah to install klipper, while manually installing everything else i wanted. After doing that i transferred the boot loader to the EMMC using the NVME for the data storage option.
> 
> I have a test macro that does all my start gcode stuff like head the hotend to 150 and heat the bed, but also loops my 7x7 bedmesh over and over 100 times. I could always get a canbus timeout doing at least doing that once on their "klipper version" of 3.0.2. I am going on me third run right now, and no timeout errors yet.
> 
> I have no data for this, but i think my idle CPU percentage is lower this way also. I don't know if there is anything about their klipper version causing the issue, but it seems like it to me.
> 
> Note: I had also removed all the Obico stuff from their klipper version and never installed it when i did my setup from the minimal version.
> 
> Hope this helps you.

The klipper version on the full image is directly from mainline. If it were not, the git hash would indicate so. There is nothing that we can think of that would result in different performance between the full image and the minimal image. Please let us know how the minimal image works for you because they should both have the same performance.

Johnnytk36, 2025-05-07:
> The klipper version on the full image is directly from mainline. If it were not, the git hash would indicate so. There is nothing that we can think of that would result in different performance [...]

Sorry, I don't just mean klipper, i meant all the addons, plugins, everything else with kilpper combined that is on the "klipper version" of your download. I didn't think yall installed a different version of klipper.

It just seems to me that doing it the way I did fixed my issues with the canbus errors during homing. 

I did no hardware changes between all of this software testing. 

I have no data for it, but it just seems like my CCPU idles are lower and doing things like clicking the refresh updates inside of mansail uses less CPU. This is all just my memory, as i didnt think to do any testing beforehand. I also do not want to go back, since it is working.

hellEVIL-13, 2025-05-10:
I have the same problem, only this helps sudo systemctl stop KlipperScreen.service

Oneseat1, 2025-05-12:
I am also still having this problem it mostly occurs when my extruder it will usually get through most of my start gcode then fail on communication timeout during the QGL.

Oneseat1, 2025-05-13:
this set up works with my CB1, i'm going to switch back to that until the comunication timeout issue gets fixed

bigtreetech, 2025-05-13:
> I am also still having this problem it mostly occurs when my extruder it will usually get through most of my start gcode then fail on communication timeout during the QGL. I didnt change any of the hardware or configuration and it was working fine with my CB1 i'm going to switch back to it until this issue is resolved.

Could you try to stop the moonmaker-obico service? You can use the setting in the upper right corner of mainsail or run: sudo systemctl stop moonraker-obico.

hellEVIL-13, 2025-05-27:
> Could you try to stop the moonmaker-obico service? You can use the setting in the upper right corner of mainsail or run: sudo systemctl stop moonraker-obico.

yes it helped, now I shouldn't use obico?
or is there some normal solution to the problem?

bigtreetech, 2025-05-27:
> yes it helped, now I shouldn't use obico? or is there some normal solution to the problem?

It may be that the latest version of Obico has a bug. We would need to report this to the developers of Obico and get them to investigate. For now, I would recommend keeping it off until they can look into it.

Oneseat1, 2025-06-01:
This is a poor excuse because the firmware on your cb1 works fine with obico running

bigtreetech, 2025-06-02:
> This is a poor excuse because the firmware on your cb1 works fine with obico running

Thanks for your input. Could you elaborate please on the underlying hardware differences between the CB1 and CB2 with specific focus on GPU acceleration kernel drivers, how they integrate with FFMPEG and how Obico takes advantage of that integration? 

The key point here is that the hardware is completely different and therefore the way that Obico uses that hardware is also completely different. Just because an app works on one windows machine does not mean that it will work on another. The app has different pieces of code to handle different usage environments.

Oneseat1, 2025-06-02:
My point is your solution was hope obico updates. In the meantime we can't use that service.

bigtreetech, 2025-06-02:
This is the only image that does not have obico installed. All others (with klipper) will have obico installed and may be causing the errors.

[image]

https://github.com/bigtreetech/CB2/releases/download/V3.0.2/CB2_Debian12_Klipper_kernel6.1_20250529_without_obico.img.xz

Oneseat1, 2025-06-03:
ok i was mistaken i did have obico installed I shut it down it and i had the same error. I then had to shut down octoapp and it seems to to be working now. I assume you are going to say its their fault and i have to wait for an update on their end to use it.

Oneseat1, 2025-06-05:
I was mistaken I did have obico Installed. I shut it off and still had a communication error, I had to shut down octoapp for it to work. I'm going to do a clean install of the lite os and reinstall klipper via kiauh. Hope for a different result octoapp is a non starter for me.

Liimp, 2025-12-17:
> This is the only image that does not have obico installed. All others (with klipper) will have obico installed and may be causing the errors.
> 
> [image]
> 
> https://github.com/bigtreetech/CB2/releases/download/V3.0.2/CB2_Debian12_Klipper_kernel6.1_20250529_without_obico.img.xz

If I uninstall (not stop the service, but delete) Obiсo, will that solve the problem?