bigtreetech/CB2 #90: Boot With NVME

albertogx34, 2025-04-25:
is possible to boot directly with and NVME connected ?

i flashed directly with balena an image to a NVME with EMMC clean and not BOOT 

if start with and SD and try with nand-sata-install not see the option to NVME, if see in lsblk the NVME

bigtreetech, 2025-04-28:
Hi, We have verified and will refine this step in the next version of the system/documentation.

Now we can use the following temporary method steps:
RK3566 won't boot directly from NVMe, so we need to write u-boot to the SD Card or eMMC first. After u-boot start up, u-boot can boot the Linux from NVMe. Please follow the steps below:

1. Start the system from the SD card first.
2. Download [cb2-u-boot-with-nvme.bin](https://github.com/bigtreetech/CB2/releases/download/V3.0.1/cb2-u-boot-with-nvme.bin) (the u-boot with nvme startup) to the system
3. Erasing the entire eMMC to prevent the system from booting from eMMC, as eMMC has a higher startup priority than NVMe
    `sudo mkfs /dev/mmcblk1`
5. Only write u-boot to eMMC
    `sudo dd if=cb2-u-boot-with-nvme.bin of=/dev/mmcblk1 seek=64`
6. If your computer can access NVMe, you can write the system image into NVMe using your computer. If your computer cannot access NVMe, you can also use PI2 to write the system into NVMeby following these steps:
    * Decompression the system image file from .xz to .img file
    * Upload the .img file to the PI2 system
    * Write the system to NVMeusing the `dd` command on the PI2 command line
        `sudo dd if=xxxx.img of=/dev/nvme0n1 status=progress`
    * Wait for the write to complete, then we can shut down and remove the SD card
 7. After powering on again, the system should be able to start normally from NVMe
 8. After startup, we will find that the available system space of NVMeis only 8GB ( It's the .img file size). We can expand the system space to the actual space of NVMe by using the following command:
    `sudo /usr/lib/armbian/armbian-resize-filesystem start`

albertogx34, 2025-04-29:
Perfect Now i have everything in the NVME and boot perfect 

Thanks

closed 2025-04-29 by albertogx34

jcheger, 2025-11-05:
I did try your solution. I also did try the one from https://github.com/bigtreetech/CB2/releases/tag/V3.0.2. But the BTT Pi2 did never boot on NVMe. It's just not booting at all (screen stays black).